ros-navigation/navigation2 #2177: Resizing Costmap

HosseinSheikhi, 2021-02-11:
## Bug report

**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 

I am going to add a costmap layer that gives ceiling images as input, and after segmenting images, generates a costmap. This ceiling_costmap has a bigger size than other layers, so the (minimal) layers are as follow:

[image]

#### Steps to reproduce the first issue
For the sake of efficiency, I want just to resize the `master_grid` to have the same size as `ceiling_costmap`. So I started with implementing a costmap plugin that just resizes the `master_grid`. 
Here is the code:
in CeilingCostmap::UpdateCosts(...)

```
if ( (master_grid.getSizeInCellsX() < size_x_desired) || (master_grid.getSizeInCellsY() <size_y_desired)) {
    master_grid.resizeMap(size_x_desired, size_y_desired, master_grid.getResolution(), master_grid.getOriginX(), master_grid.getOriginY());
    RCLCPP_INFO(node_->get_logger(), "Gradient layer after resizing: %d, %d, %f, %f, %f",
                master_grid.getSizeInCellsX(),
                master_grid.getSizeInCellsY(),
                resolution_,
                origin_x_,
                origin_y_
    );
  }
```
Worth to mention I used minimal plugins for global costmap: (plugins: ["static_layer", "ceiling_layer"]):

#### Steps to reproduce the second issue
In `static_layer.cpp`, in `processMap()`, tried to resize the `layered_costmap_` double size of `static_map`:

#### Expected behavior
I expect the final costmap to be something like this by extra unknown cells:
[image]

 ```
if (!layered_costmap_->isRolling() && (master->getSizeInCellsX() < size_x ||
    master->getSizeInCellsY() < size_y ||
    master->getResolution() != new_map.info.resolution ||
    master->getOriginX() != new_map.info.origin.position.x ||
    master->getOriginY() != new_map.info.origin.position.y))
  {
    // Update the size of the layered costmap (and all layers, including this one)
    RCLCPP_INFO(
      node_->get_logger(),
      "StaticLayer: Resizing costmap to %d X %d at %f m/pix", size_x, size_y,
      new_map.info.resolution);
    layered_costmap_->resizeMap(
      size_x*2, size_y*2, new_map.info.resolution,
      new_map.info.origin.position.x,
      new_map.info.origin.position.y,
      true);
```
#### Actual behavior
For the first issue:
[image]

For the second issue:
[image]
and after moving the robot to update the map:
[image]

SteveMacenski, 2021-02-11:
Resize the layered costmap the same way static layer does https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/plugins/static_layer.cpp#L194 - the `master_grid` is the internal costmap to your own layer itself. Your costmap layer has a costmap it deals with, then there's the layered costmap this is combined with at the end of updates. Be careful that you either don't modify or you modify in all places the updated origin information. 

Please don't add 2+ problems into the same ticket, I can't dissect them or start to help if I can't deal with one problem at a time.

HosseinSheikhi, 2021-02-11:
Sorry for submitting two issues.

I added the following lines in `CeilingLayer::UpdateBounds`:
```
Costmap2D * master = layered_costmap_->getCostmap();
  if ( (master->getSizeInCellsX() < size_x_desired) || (master->getSizeInCellsY() <size_y_desired)) {

    layered_costmap_->resizeMap(size_x_desired, size_y_desired, master->getResolution(), master->getOriginX(), master->getOriginY());
    RCLCPP_INFO(node_->get_logger(), "Gradient layer after resizing: %d, %d, %f, %f, %f",
                master->getSizeInCellsX(),
                master->getSizeInCellsY(),
                resolution_,
                origin_x_,
                origin_y_
    );
  }
```
This is the output:

Sensor origin at (-2.01, -0.50) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [INFO] [1613075874.120697365] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 111 X 102 at 0.050000 m/pix
[planner_server-2] [INFO] [1613075874.803046111] [global_costmap.global_costmap]: CeilingLayer: Resizing costmap to 180 X 140 at 0.050000 m/pix
[planner_server-2] [INFO] [1613075879.119516498] [global_costmap.global_costmap]: StaticLayer: Resizing static layer to 111 X 102 at 0.050000 m/pix

and here is the final costmap (`plugins: ["static_layer", "obstacle_layer", "ceiling_layer" , "inflation_layer"]`):
[image]
Same things happen if I do not let the StaticLayer to resize it self after CeilingLayer resized the costmap

SteveMacenski, 2021-02-12:
I would not mess with anything else but the ceiling layer to start off and get what you're looking for there, then slowly add more to it. Trying to solve everything at once is just going to create stress. Solve one problem at a time
- get your ceiling layer to do its thing
- add in the static layer and deal with any integration issues there
- add in the sensor data processing layers (also unlikely to cause an issue at this point) 
- add inflation (which shouldn't cause any issue ever)

SteveMacenski, 2021-02-12:
this is starting to hit the point of a ROS answers-type thing. While I typically try to help people in with their integration problems, this feels like we've hit the boundaries of what I can help you with unless this is going to be open-source in Nav2/ROS2. I can't unfortunately help all users through the development process of their custom proprietary / research features or I'd get nothing else done :wink: 

If you think this is a good point, I'd recommend closing the ticket unless there's something tangible I can do in the stack / issues to be fixed somewhere

HosseinSheikhi, 2021-02-13:
I quite understand what you mean, and I believe I messed things by pointing to two issues in one ticket, sorry for that. let us forget about the ceiling layer. I edited my question. It is just about doubling the cost map size. Does the edited ticket seem to be a bug?

HosseinSheikhi, 2021-02-16:
Have figured it out, the initialization block ([here](https://github.com/ros-planning/navigation2/blob/4e52bad8d7b8add98291472c114ea0ec054bcd1b/nav2_costmap_2d/plugins/static_layer.cpp#L219))
must be written as:
`  // initialize the costmap with static data
  for (unsigned int i = 0; i < size_y_; ++i) {
    for (unsigned int j = 0; j < size_x_; ++j) {
      if(i>=new_map.info.height || j>=new_map.info.width){
        costmap_[master->getIndex(j,i)] = FREE_SPACE;
      }
      else{
        unsigned char value = new_map.data[index];
        costmap_[master->getIndex(j,i)] =interpretValue(value);
        ++index;
      }
    }
  }`

HosseinSheikhi, 2021-02-16:
@SteveMacenski one more thing, when a plugin (except `static_layer`) resizes the 'layered_costmap_', will lead to a back and forth of changing `layered_costmap_`'s size because `static_layer` tries to resize it to the `new_map`'s size [here](https://github.com/ros-planning/navigation2/blob/4e52bad8d7b8add98291472c114ea0ec054bcd1b/nav2_costmap_2d/plugins/static_layer.cpp#L182).

I have solved this back and forth for my case, and I know it's less likely that people want to have a costmap bigger than the static map. So, if you think we have to figure out this "maybe bug" let me know, otherwise feel free to close this issue.

SteveMacenski, 2021-02-16:
Would changing the checks to `<` rather than `!=` solve your issue? If something else resizes it later, then it wouldn't resize if what you have set otherwise is later. 

I would be concerned about the changes in origin of a new map update on how that impacts your ceiling layer, but that's an easier thing to test to see if they drift apart than try to guess.

HosseinSheikhi, 2021-02-16:
> Would changing the checks to `<` rather than `!=` solve your issue? If something else resizes it later, then it wouldn't resize if what you have set otherwise is later.

Yes, this would be a straightforward solution. I also suggest [`resizeMap`](https://github.com/ros-planning/navigation2/blob/4e52bad8d7b8add98291472c114ea0ec054bcd1b/nav2_costmap_2d/include/nav2_costmap_2d/layered_costmap.hpp#L81) function get one extra `bool clear` arguments, if its `true` the functionality will be same as current implementation but it is `false`, before resizing keeps a copy of current size and data, and initialize that part with the previous data and extra part by `NO_INFORMATION`.

> I would be concerned about the changes in origin of a new map update on how that impacts your ceiling layer, but that's an easier thing to test to see if they drift apart than try to guess.

Am working on this part.

HosseinSheikhi, 2021-02-16:
[Here](https://github.com/ros-planning/navigation2/blob/4e52bad8d7b8add98291472c114ea0ec054bcd1b/nav2_costmap_2d/plugins/static_layer.cpp#L391) static layer assumes if not rolling, the layered costmap (master_grid) has the same coordinates as static layer, and the update functions like [this one ](https://github.com/ros-planning/navigation2/blob/4e52bad8d7b8add98291472c114ea0ec054bcd1b/nav2_costmap_2d/src/costmap_layer.cpp#L109) are implemented based on this assumption. 
The contradiction arises [here](https://github.com/ros-planning/navigation2/blob/4e52bad8d7b8add98291472c114ea0ec054bcd1b/nav2_costmap_2d/plugins/static_layer.cpp#L199), when the static layer tries to resize JUST itself independent of rolling condition.

SteveMacenski, 2021-02-17:
I think that's how all costmap layers work. When you resize the master  grid, `matchSize` is called on all the layers that is responsible for keeping things consistent.

HosseinSheikhi, 2021-02-17:
1- Somewhere except static layer `layered_costmap_->resizeMap`
2- all the layers size will change to the same size
3- a new map is subscribed and the static layer will change its `costmap_` size
Now static the layer's `costmap_` size is not as same as master grid! So master will not update correctly. 

To me seems `nav2_costmap_2d` is implemented generic enough for the case static layer acts as a master and controls the dimension and origin. and if another layer wants to change those has to handle many exceptions.

@DLu @mikeferguson Would you please take a look at this thread, and correct me if am wrong.

SteveMacenski, 2021-02-17:
https://github.com/ros-planning/navigation2/blob/dcc1d4dafba6f6c7588cb5f7f6ca918b2a6c48ef/nav2_costmap_2d/src/layered_costmap.cpp#L87 Look at the function that its calling - it matches size for all of the plugins

There's not a situation as far as I know where the layered costmap and the costmap layers are different sizes in non-rolling situations. Else, we'd see serious issues with just the standard sensor processing layers.

HosseinSheikhi, 2021-02-17:
That's absolutely right, but am talking about the situation when someone wants to change the layered cost map size. It is about a generic implementation of costmap.

SteveMacenski, 2021-02-17:
I've lost insight into what this discussion is about. Is there a specific  bug or issue you want resolved? We talked about the `!=` to `<` but this seems like we're getting deep into specific issues to your application that I cannot help with. If there's a specific suggestion, we can work with that, but like I mentioned before, I cannot spend my time helping debug a proprietary system

HosseinSheikhi, 2021-02-18:
Right, let me close this one, will get back with a consistent suggestion to improve the `nav2_costmap_2d`.

closed 2021-02-18 by HosseinSheikhi