ros-navigation/navigation2 #2187: Segmentation fault when running AMCL with simulated LaserScan

abylikhsanov, 2021-02-15:
## Bug report

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

1. Download the current release which is 0.4.5 and compile the navigation package with` --cmake-args -DCMAKE_BUILD_TYPE=Release`
2. Use these AMCL parameters:
```
amcl:
  ros__parameters:
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: base_footprint
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: map
    initial_pose.x: 16.5
    initial_pose.y: 16.5
    initial_pose.z: 0.0
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: likelihood_field
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: odom
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: omni
    save_pose_rate: 0.5
    scan_topic: scan
    set_initial_pose: true
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    use_sim_time: true
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
amcl_map_client:
  ros__parameters:
    use_sim_time: true
amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: true

```
3. Run the Gazebo simulation with the simulated LaserScan and move the robot until you receive the seg fault.

[... 61 lines cut ...]
#12 0x00007ffff7d75560 in std::function<void (message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&)>::operator()(message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&) const (__args#0=..., this=0x555555b30758)
    at /usr/include/c++/9/bits/std_function.h:683
#13 message_filters::CallbackHelper1T<message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&, sensor_msgs::msg::LaserScan_<std::allocator<void> > >::call (this=0x555555b30750, event=..., nonconst_force_copy=<optimized out>)
    at /opt/ros/foxy/include/message_filters/signal1.h:74
#14 0x00007ffff7d75308 in message_filters::Signal1<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::call (event=..., this=0x555555927708) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#15 message_filters::SimpleFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::signalMessage (event=..., this=0x555555927708) at /opt/ros/foxy/include/message_filters/simple_filter.h:133
#16 message_filters::Subscriber<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::cb (e=..., this=0x555555927700) at /opt/ros/foxy/include/message_filters/subscriber.h:235
#17 message_filters::Subscriber<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::subscribe(rclcpp::Node*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rmw_qos_profile_t)::{lambda(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>)#1}::operator()(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>) const (this=<optimized out>, msg=...) at /opt/ros/foxy/include/message_filters/subscriber.h:174
#18 std::_Function_handler<void (std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>), message_filters::Subscriber<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::subscribe(rclcpp::Node*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rmw_qos_profile_t)::{lambda(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>)#1}>::_M_invoke(std::_Any_data const&, std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>&&) (__functor=..., __args#0=...) at /usr/include/c++/9/bits/std_function.h:300
#19 0x00007ffff7d65dee in std::function<void (std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>)>::operator()(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>) const (
    __args#0=std::shared_ptr<const sensor_msgs::msg::LaserScan_<std::allocator<void> >> (empty) = {...}, this=0x555555b3dce0) at /usr/include/c++/9/bits/std_function.h:683
#20 rclcpp::AnySubscriptionCallback<sensor_msgs::msg::LaserScan_<std::allocator<void> >, std::allocator<void> >::dispatch (this=0x555555b3dca0, message=std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> >> (use count 10, weak count 0) = {...}, message_info=...)
    at /opt/ros/foxy/include/rclcpp/any_subscription_callback.hpp:167
#21 0x00007ffff7d6a3a8 in rclcpp::Subscription<sensor_msgs::msg::LaserScan_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<sensor_msgs::msg::LaserScan_<std::allocator<void> >, std::allocator<void> > >::handle_message (this=0x555555b3dbc0, message=..., 

    message_info=...) at /usr/include/c++/9/ext/atomicity.h:96
#22 0x00007ffff7ecf35c in ?? () from /opt/ros/foxy/lib/librclcpp.so
#23 0x00007ffff7ecfc1b in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>) () from /opt/ros/foxy/lib/librclcpp.so
#24 0x00007ffff7ed03da in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) () from /opt/ros/foxy/lib/librclcpp.so
#25 0x00007ffff7ed4bdc in rclcpp::executors::SingleThreadedExecutor::spin() () from /opt/ros/foxy/lib/librclcpp.so
#26 0x00007ffff75e619d in std::thread::_State_impl<std::thread::_Invoker<std::tuple<nav2_util::NodeThread::NodeThread(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>)::{lambda()#1}> > >::_M_run() () from /home/abyl/genius_ros2/ros_ws/install/nav2_util/lib/libnav2_util_core.so
#27 0x00007ffff7b4ad84 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#28 0x00007ffff784d609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#29 0x00007ffff7989293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

SteveMacenski, 2021-02-15:
This looks like a lower level issue than nav2 (at first glance). did you try with cyclone DDS? Any changes in message_filters look like this could have been caused? Nothing here I can point to as a Nav2 issue, the traceback is going into message filters / binding / shared pointers

abylikhsanov, 2021-02-16:
@SteveMacenski I never fully understood the debug message and I also thought that this is a low level error at some point but when I have changed:

`robot_model_type: omni`

to 

`robot_model_type: differential`

I stopped seeing this crash. I am not sure whether I should close this issue.

closed 2021-02-16 by abylikhsanov

ste93, 2022-01-12:
@abylikhsanov  I have the same problem, how did you solved it?