ros-navigation/navigation2 #2191: nav2_behavior_tree fails for foxy-devel branch: cannot find behaviortree_cpp_v3 cmake config file

berndpfrommer, 2021-02-18:
Running on Nvidia Xavier (Ubuntu 18.04).
ROS version: foxy (compiled from source since there are no foxy packages available for Ubuntu 18.04 on ARM)

Here is the compile command I'm using (all packages succeed except for nav2_behavior_tree):

colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo --packages-select nav2_behavior_tree

Here is how it fails:

```
Starting >>> nav2_behavior_tree
--- stderr: nav2_behavior_tree
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findbehaviortree_cpp_v3.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "behaviortree_cpp_v3", but CMake did not find one.

  Could not find a package configuration file provided by
  "behaviortree_cpp_v3" with any of the following names:

    behaviortree_cpp_v3Config.cmake
    behaviortree_cpp_v3-config.cmake

  Add the installation prefix of "behaviortree_cpp_v3" to CMAKE_PREFIX_PATH
  or set "behaviortree_cpp_v3_DIR" to a directory containing one of the above
  files.  If "behaviortree_cpp_v3" provides a separate development package or
  SDK, be sure it has been installed.

```
I did compile and install "BehaviorTree.CPP" (master branch, but also tried 3.5.1 without luck). A config file is installed in the right place:
```
install/behaviortree_cpp_v3/lib/BehaviorTreeV3/cmake/BehaviorTreeV3Config.cmake
```
But it has the wrong name so it is not found by the nav2_behavior_tree package.

Any help is appreciated!

SteveMacenski, 2021-02-18:
Did you follow our [build and install](https://navigation.ros.org/build_instructions/index.html) instructions?

berndpfrommer, 2021-02-18:
well, admittedly not quite.
The xavier is not a speed demon when it comes to compiling so I was very reluctant to run commands that I don't know what they are doing like "vcs import" nad rosdep install because if they mess up my workspace it takes 1h+ to recompile.

I followed the instructions up to this point:
```
wget https://raw.githubusercontent.com/ros-planning/navigation2/main/tools/underlay.repos
vcs import src < underlay.repos
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro <ros2-distro>
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```
Instead of the vcs import I manually git cloned the repos listed in underlay.repos, making sure I checked out the same version as specified. All packages compiled nicely, except for the problems mentioned above.

SteveMacenski, 2021-02-18:
and did you source your underlay workspace before compiling your main one? If you've tried compiling the main one before, nuke your install/build/log and retry, might be a caching issue. 

But what those instructions call out, I do locally every few weeks and haven't had an issue (and essentially what CI does every time we trigger a PR https://github.com/ros-planning/navigation2/blob/main/.circleci/config.yml)

berndpfrommer, 2021-02-18:
I really appreciate your fast responses!
Unfortunately going by the book following the [ROS2 installation instructions here](https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/) falls flat on its face
already when trying to install the basic tools. This is because I also have the stock ROS1 melodic installed on the same machine because lots of components don't have ROS2 versions yet. Dreadful sounding messages like the one below are then the result:
```
sudo apt install python3-rospkg
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libapr1-dev libaprutil1-dev libassuan-dev libcharls1 libgdcm2.8 libgeographic-dev libgpg-error-dev libgpgme-dev liblog4cxx-dev liblog4cxx10v5 liblz4-dev libopencv-calib3d3.2
  libopencv-core3.2 libopencv-features2d3.2 libopencv-flann3.2 libopencv-highgui3.2 libopencv-imgcodecs3.2 libopencv-imgproc3.2 libopencv-ml3.2 libopencv-videoio3.2 libpoco-dev
  libpococrypto50 libpocodata50 libpocodatamysql50 libpocodataodbc50 libpocodatasqlite50 libpocofoundation50 libpocojson50 libpocomongodb50 libpoconet50 libpoconetssl50 libpocoredis50
  libpocoutil50 libpocoxml50 libpocozip50 libsctp-dev libsctp1 libsocket++1 libwxbase3.0-0v5 libwxgtk3.0-gtk3-0v5 libyaml-cpp-dev libyaml-cpp0.5v5 python-defusedxml python-gnupg
  python-netifaces python-pycryptodome python-wxgtk3.0 python-wxtools python-wxversion ros-melodic-actionlib-msgs ros-melodic-bond ros-melodic-bondcpp
  ros-melodic-camera-calibration-parsers ros-melodic-class-loader ros-melodic-cmake-modules ros-melodic-common-msgs ros-melodic-cv-bridge ros-melodic-diagnostic-msgs
  ros-melodic-diagnostic-updater ros-melodic-eigen-conversions ros-melodic-gencpp ros-melodic-geneus ros-melodic-genlisp ros-melodic-gennodejs ros-melodic-image-geometry
  ros-melodic-kdl-conversions ros-melodic-libmavconn ros-melodic-mavlink ros-melodic-message-filters ros-melodic-message-generation ros-melodic-nav-msgs ros-melodic-orocos-kdl
  ros-melodic-python-orocos-kdl ros-melodic-ros-environment ros-melodic-rosbag-migration-rule ros-melodic-rosboost-cfg ros-melodic-rosbuild ros-melodic-rosconsole
  ros-melodic-rosconsole-bridge ros-melodic-roscpp ros-melodic-roscpp-core ros-melodic-rosgraph-msgs ros-melodic-roslang ros-melodic-roslz4 ros-melodic-rosout ros-melodic-shape-msgs
  ros-melodic-smclib ros-melodic-std-srvs ros-melodic-stereo-msgs ros-melodic-tf2 ros-melodic-tf2-eigen ros-melodic-tf2-msgs ros-melodic-topic-tools ros-melodic-trajectory-msgs
  ros-melodic-visualization-msgs ros-melodic-xmlrpcpp sbcl
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  python-rospkg-modules python3-rospkg-modules
The following packages will be REMOVED:
  python-rosdep python-rosdep-modules python-rosinstall-generator python-rospkg ros-melodic-actionlib ros-melodic-bond-core ros-melodic-bondpy ros-melodic-camera-info-manager
  ros-melodic-dynamic-reconfigure ros-melodic-image-proc ros-melodic-image-transport ros-melodic-mavros ros-melodic-mk ros-melodic-nodelet ros-melodic-nodelet-core
  ros-melodic-nodelet-topic-tools ros-melodic-pluginlib ros-melodic-ros ros-melodic-ros-base ros-melodic-ros-comm ros-melodic-ros-core ros-melodic-rosbag ros-melodic-rosbag-storage
  ros-melodic-rosbash ros-melodic-rosclean ros-melodic-roscreate ros-melodic-rosgraph ros-melodic-roslaunch ros-melodic-roslib ros-melodic-roslisp ros-melodic-rosmake
  ros-melodic-rosmaster ros-melodic-rosmsg ros-melodic-rosnode ros-melodic-rospack ros-melodic-rosparam ros-melodic-rospy ros-melodic-rosservice ros-melodic-rostest ros-melodic-rostopic
  ros-melodic-rosunit ros-melodic-roswtf ros-melodic-tf ros-melodic-tf-conversions ros-melodic-tf2-geometry-msgs ros-melodic-tf2-py ros-melodic-tf2-ros
The following NEW packages will be installed:
  python3-rospkg
```
I will start from scratch again and try to work around any adverse ROS1/ROS2 interaction the best I can. Will post update.

SteveMacenski, 2021-02-18:
Uh, oh, umm, yeah. You're going to have a really, really bad time if you want to install Melodic next to Foxy. 

Recall that Foxy uses python3 (with Noetic) on 20.04 and Melodic is on 18.04 with python2. You're going to have an excessively bad time trying to get all this to work correctly. You really should either update to 20.04, use a ROS2 that's suitable for 18.04, or - really the best answer - use **Docker**. That would largely solve all of your problems.

berndpfrommer, 2021-02-18:
Documenting more snags that I hit:
1) when setting up the nav2_depend_ws I got the following error:
```
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro foxy                                                                        
ERROR: the following packages/stacks could not have their rosdep keys resolved                                                                                                             
to system dependencies:                                                                                                                                                                    
gazebo_dev: No definition of [libgazebo11-dev] for OS version [bionic]                                                                                                                     
Continuing to install resolvable dependencies...                                                                                                                                           
executing command [sudo -H apt-get install -y -qq python3-opencv]                                                                                                                          
[sudo] password for pfrommer:                                                                                                                                                              
debconf: delaying package configuration, since apt-utils is not installed                                                                                                                  
Selecting previously unselected package libgl2ps1.4.                      
```
This is because Ubuntu 18.04 ships with libgazebo9, not libgazebo 11.

2) The bond_cpp package build fails with this error:
```
/ssd/opt/ros/nav2_depend_ws/src/ros/bond_core/bondcpp/src/bond.cpp: In member function ‘void bond::Bond::connectTimerReset()’:
/ssd/opt/ros/nav2_depend_ws/src/ros/bond_core/bondcpp/src/bond.cpp:192:43: error: ‘from_nanoseconds’ is not a member of ‘rclcpp::Duration’
   rclcpp::Duration dur1(rclcpp::Duration::from_nanoseconds(connect_timeout_));
                                           ^~~~~~~~~~~~~~~~

```
I just checked: foxy's rclcpp duration does not have the from_nanoseconds() method on their duration class, but it is in the master branch.
I reverted bondcpp to 2.0.0 and at least it compiled.
The only package that didn't compile in nav2_depend_ws was gazebo_ros (no surprise there).
3) when running the rosdep install on the nav2_ws I get this error:
```
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro foxy
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_bringup: No definition of [slam_toolbox] for OS version [bionic]
nav2_system_tests: No definition of [gazebo_ros_pkgs] for OS version [bionic]
nav2_smac_planner: No definition of [ompl] for OS version [bionic]
Continuing to install resolvable dependencies...
```
A bit surprised about that ompl complaint because that one actually was in the nav2_depend_ws's install directory (I checked). The slam_toolbox though is missing. I guess another issue with trying to run foxy on bionic.
During build, I got a bunch of errors again when using the "main" branch (nav2_lifecycle_manager looking for from_nanoseconds() missing), so I checked out the foxy-devel branch. That one I was able to compile with the exception of nav2_system_tests, which failed because of missing gazebo_ros_pkgs.

Strangely enough, the error I was hitting before of the behaviortree_cpp_v3 cmake config file not being found has resolved itself. Now the cmake file shows up in the install directory. In both workspaces I have an identical BehaviorTree.CPP source tree, identical commit. Weird.

The reason I am banging my head against this is because NVidia is not yet supporting Jetpack on the Xavier AGX for Ubuntu 20.04, so I would go unsupported there if I upgraded the OS from 18.04 to 20.04. I have used docker in the past but I have the feeling that not all will be smooth sailing there with respect to CUDA, camera drivers, USB device access etc. So far I got enough of foxy to build that I can make recent progress - with the exception of navigation2.

Anyways, thank you for your help. I'm closing this issue as the original problem has mysteriously disappeared and I got most of the package to compile (not tested if it works properly).

closed 2021-02-18 by berndpfrommer

SteveMacenski, 2021-02-18:
Happy it all worked out (in > 1 day) :-)

3wnbr1, 2021-03-22:
@SteveMacenski, @berndpfrommer, I have an update regarding this issue

I also encountered it while cross building ros2 foxy + navigation2 for Debian aarch64 using [ros-tooling/cross_compile](https://github.com/ros-tooling/cross_compile). The root cause seems to lie between `BehaviorTree.CPP` and `ros_cross_compile`

BehaviorTree.CPP uses conditions in it's `package.xml` based on the `ROS_VERSION` environment variable which is expected to be set to `2`. This is documented by navigation2 in the docs.

```xml
<buildtool_depend condition="$ROS_VERSION == 1">catkin</buildtool_depend>
<depend condition="$ROS_VERSION == 1">roslib</depend>

<buildtool_depend condition="$ROS_VERSION == 2">ament_cmake</buildtool_depend>
<depend condition="$ROS_VERSION == 2">rclcpp</depend>
```

```xml
<export>
    <build_type condition="$ROS_VERSION == 2">ament_cmake</build_type>
</export>
```

But if you don't have `ROS_VERSION` set, cmake won't install it as a ros package. For my case, building through `ros_cross_compile`, it builds within a docker and sets `ROS_VERSION` to `ros2` which causes this exact issue.

In that case, even f you were able to copy the cmake config files at the right place, a linker issue would await you while linking with `nav2_bt_navigator`.

Therefore an `export ROS_VERSION=2` and a BehaviorTree.CPP clean build fixes the issue

*PS : Could that one be fixed for foxy*
https://github.com/ros-planning/navigation2/blob/3befe9aa1c2546f2178c7ced6fe55fc3ba56642a/tools/ros2_dependencies.repos#L22-L25