ros-navigation/navigation2 #2196: Provide slam_toolbox parameter file argument through nav2_bringup

v-lopez, 2021-02-19:
## Feature request

#### Feature description
Currently nav2_bringup/bringup_launch.py is perfect for launching the navigation of a robot. With minimal modifications to the parameter file I can run [on our PMB2 robot](https://github.com/pal-robotics/pmb2_navigation/blob/foxy-devel/pmb2_2dnav/params/pmb2_params.yaml). 

But when setting `slam:=True`, `bringup_launch.py` includes `slam_launch.py` which includes `slam_toolbox/online_sync_launch.py` without passing a parameter file. It always loads the one included on the package.

To change such params, I have to create a separate online_sync_launch.py file, and therefore modify bringup_launch.py and slam_launch.py.

SteveMacenski, 2021-02-19:
I'd be happy to consider a specific way of supporting your requested behavior in `nav2_bringup` directly, if it works for all else for your robots. Do you have a specific suggestion of what to change in Nav2/SLAM Toolbox? 

Keep in mind that is an example bringup package, its expected that a specific robot or specific company creates their own `<my-robot>_bringup` or `<my-robot>_nav` package containing your launch / configuration files no differently from ROS1. I'm happy that our provided one so closely mirrors your needs so I'd like to see if we could make it work general-purpose - but that's kind of what I was thinking in general.

v-lopez, 2021-02-19:
Short answer, I believe that it might be interesting to be able to specify the slam_toolbox parameters in the params file where the rest of parameters are.  I have not had the time to look at what all the nodes do, and whether it makes sense that they are all in the same file or not.

Regarding the `<my-robot>_bringup`, one of my issues with ROS2 is how complex launch files have become. 
For years we've been eager to have python launch files because the ROS1 XML where limiting us in many ways, but the current implementation can be difficult to work with. This is not navigation2's fault, it's just the way it is now.

Right now the basic launch files in nav2_bringup amount to 500 lines of python code. We have to mantain navigation for at least 5 different robots, hence why I tried to avoid copying them and using directly nav2_bringup_launch.py with just a different param file. 

Eventually I'll guess I'll have to split and maybe only reuse `slam_launch.py` and rewrite some of the others. But every reusable bit reduces entropy.

SteveMacenski, 2021-02-23:
> Short answer, I believe that it might be interesting to be able to specify the slam_toolbox parameters in the params file where the rest of parameters are. I have not had the time to look at what all the nodes do, and whether it makes sense that they are all in the same file or not.

That seems reasonable, but I don't want to own in general that requirement that Nav2 has **any** SLAM parameters at all (or any parameters of any packages outside of this stack). So we need to find a way to enable that but allow users to use profiles tuned and available in those packages if not in the param file (which should be possible through the python launch interface). Perhaps some quick python logic to see if the params file includes a `slam_toolbox:` namespace, then launches the node with that param file. Else, launches the slam toolbox `launch sync online` file. Python3 launch would allow us to do that (unlike XML)!

v-lopez, 2021-02-24:
Maybe having a `LaunchConfiguration` for a parameter file path on `online_sync_launch.py` that defaults to `mapper_params_online_sync.yaml`, and exposing that LaunchConfiguration to the callers of nav2_bringup would be enough. What do you think?

SteveMacenski, 2021-02-24:
So a separate launch configuration for a SLAM file vs the general navigation file? That would also work, though a bit less optimal in my opinion to have a user supply 2 files (vs 1 file, check if its there and use default otherwise). But I'd accept a PR to implement that behavior as well.

v-lopez, 2021-02-24:
Awesome, I'll see if I can find the time this Friday, otherwise next Tuesday.

SteveMacenski, 2021-03-03:
@v-lopez let me know if you have any launch questions I can help with

SteveMacenski, 2021-03-04:
Imminently to merge

closed 2021-03-04 by SteveMacenski