ros-navigation/navigation2 #2200: Test Updown path mismatch

Uladzslau, 2021-02-22:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- ROS2 Version: Foxy
- Version or commit hash: Docker rosplanning/navigation2:main.release
- DDS implementation: Default

#### Steps to reproduce issue
``` ros2 launch nav2_system_tests test_updown_launch.py``` 

#### Expected behavior
Running updown test
#### Actual behavior
```
[ERROR] [test_updown-14]: exception occurred while executing process:
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 737, in __execute_process
    transport, self._subprocess_protocol = await async_execute_process(
  File "/opt/ros/foxy/lib/python3.8/site-packages/osrf_pycommon/process_utils/async_execute_process_asyncio/impl.py", line 139, in async_execute_process
    transport, protocol = await _async_execute_process_nopty(
  File "/opt/ros/foxy/lib/python3.8/site-packages/osrf_pycommon/process_utils/async_execute_process_asyncio/impl.py", line 45, in _async_execute_process_nopty
    transport, protocol = await loop.subprocess_exec(
  File "/usr/lib/python3.8/asyncio/base_events.py", line 1630, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/usr/lib/python3.8/asyncio/unix_events.py", line 197, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
  File "/usr/lib/python3.8/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/usr/lib/python3.8/asyncio/unix_events.py", line 789, in _start
    self._proc = subprocess.Popen(
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/opt/overlay_ws/install/nav2_system_tests/lib/nav2_system_tests/test_updown'

```

#### Additional information
So the launch file tries to search in for test in `lib`, while the file is installed to `share`

Launch:
https://github.com/ros-planning/navigation2/blob/main/nav2_system_tests/src/updown/test_updown_launch.py#L70

Installation:
https://github.com/ros-planning/navigation2/blob/main/nav2_system_tests/src/updown/CMakeLists.txt#L9

Possible cause
https://github.com/ros-planning/navigation2/commit/29f97126fc8060d9403546cff6df0b739ae6c94f

SteveMacenski, 2021-02-23:
@Uladzslau a PR to resolve this would be appreciated, I believe you are correct

Uladzslau, 2021-02-26:
@SteveMacenski Will do, could you add my account as a collaborator?

closed 2021-02-26 by SteveMacenski