ros-navigation/navigation2 #2207: Smac Planner planning through high cost areas

maxlein, 2021-02-26:
It looks like smac planner is ignoring costs when obstacles are too small.
Navfn on the other side doesn't show this behavior.

I posted the question already on [ros answers.](https://answers.ros.org/question/372400/navfn-vs-smac-planner/?comment=372564#post-id-372564)

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy binaries with ported regulated pure pursuit
- DDS implementation:
  - Cyclone

#### Steps to reproduce issue

Load this grid map and the params mentioned down below.
[grid_map.zip](https://github.com/ros-planning/navigation2/files/6048489/grid_map.zip)

#### Additional information

Navfn:  

[image]   
Don't know why it's not showing, but clicking on it does...

Smac:  

[image]

Params:  

```yaml
planner_server:
  ros__parameters:
    planner_plugins: ["GridBased"]
    use_sim_time: True

    # smac_planner/SmacPlanner is replaced with nav2_navfn_planner/NavfnPlanner, params are from smac, but navfn works better right now
    GridBased:
      plugin: "nav2_navfn_planner/NavfnPlanner"
      tolerance: 0.5                    # tolerance for planning if unable to reach exact pose, in meters, for 2D node
      downsample_costmap: false         # whether or not to downsample the map
      downsampling_factor: 1            # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
      allow_unknown: false              # allow traveling in unknown space
      max_iterations: -1                # maximum total iterations to search for before failing
      max_on_approach_iterations: 1000  # maximum number of iterations to attempt to reach goal once in tolerance, 2D only
      max_planning_time: 2.0            # max time in s for planner to plan, smooth, and upsample. Will scale maximum smoothing and upsampling times based on remaining time after planning.
      smooth_path: false                # Whether to smooth searched path
      motion_model_for_search: "DUBIN"  # 2D Moore, Von Neumann; SE2 Dubin, Redds-Shepp
      angle_quantization_bins: 72       # For SE2 node: Number of angle bins for search, must be 1 for 2D node (no angle search)
      minimum_turning_radius: 0.20      # For SE2 node & smoother: minimum turning radius in m of path / vehicle
      reverse_penalty: 2.1              # For Reeds-Shepp model: penalty to apply if motion is reversing, must be => 1
      change_penalty: 0.20              # For SE2 node: penalty to apply if motion is changing directions, must be >= 0
      non_straight_penalty: 1.05        # For SE2 node: penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 1.3                 # For SE2 node: penalty to apply to higher cost zones

      smoother:
        smoother:
          w_curve: 30.0                 # weight to minimize curvature of path
          w_dist: 0.0                   # weight to bind path to original as optional replacement for cost weight
          w_smooth: 30000.0             # weight to maximize smoothness of path
          w_cost: 0.025                 # weight to steer robot away from collision and cost
          cost_scaling_factor: 10.0     # this should match the inflation layer's parameter

        # I do not recommend users mess with this unless they're doing production tuning
        optimizer:
          max_time: 0.10                # maximum compute time for smoother
          max_iterations: 500           # max iterations of smoother
          debug_optimizer: false        # print debug info
          gradient_tol: 1.0e-10
          fn_tol: 1.0e-20
          param_tol: 1.0e-15
          advanced:
            min_line_search_step_size: 1.0e-20
            max_num_line_search_step_size_iterations: 50
            line_search_sufficient_function_decrease: 1.0e-20
            max_num_line_search_direction_restarts: 10
            max_line_search_step_expansion: 50
```

SteveMacenski, 2021-02-27:
Please try with Nav2 master and let me know if you still see the issue. I suspect the issue you bring up has been patched.

maxlein, 2021-03-01:
Testing with master is very time consuming and I may not have time in the next weeks...  

I can only say that this was reproducible with the attached map.

Planner params are pretty much default.  
NavfnPlanner is activated here as I just changed to it, to test if the same happens with the other planner.

```yaml
planner_server:
  ros__parameters:
    planner_plugins: ["GridBased"]
    use_sim_time: True

    # smac_planner/SmacPlanner is replaced with nav2_navfn_planner/NavfnPlanner, params are from smac, but navfn works better right now
    GridBased:
      plugin: "nav2_navfn_planner/NavfnPlanner"
      tolerance: 0.5                    # tolerance for planning if unable to reach exact pose, in meters, for 2D node
      downsample_costmap: false         # whether or not to downsample the map
      downsampling_factor: 1            # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
      allow_unknown: false              # allow traveling in unknown space
      max_iterations: -1                # maximum total iterations to search for before failing
      max_on_approach_iterations: 1000  # maximum number of iterations to attempt to reach goal once in tolerance, 2D only
      max_planning_time: 2.0            # max time in s for planner to plan, smooth, and upsample. Will scale maximum smoothing and upsampling times based on remaining time after planning.
      smooth_path: false                # Whether to smooth searched path
      motion_model_for_search: "DUBIN"  # 2D Moore, Von Neumann; SE2 Dubin, Redds-Shepp
      angle_quantization_bins: 72       # For SE2 node: Number of angle bins for search, must be 1 for 2D node (no angle search)
      minimum_turning_radius: 0.20      # For SE2 node & smoother: minimum turning radius in m of path / vehicle
      reverse_penalty: 2.1              # For Reeds-Shepp model: penalty to apply if motion is reversing, must be => 1
      change_penalty: 0.20              # For SE2 node: penalty to apply if motion is changing directions, must be >= 0
      non_straight_penalty: 1.05        # For SE2 node: penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 1.3                 # For SE2 node: penalty to apply to higher cost zones

      smoother:
        smoother:
          w_curve: 30.0                 # weight to minimize curvature of path
          w_dist: 0.0                   # weight to bind path to original as optional replacement for cost weight
          w_smooth: 30000.0             # weight to maximize smoothness of path
          w_cost: 0.025                 # weight to steer robot away from collision and cost
          cost_scaling_factor: 10.0     # this should match the inflation layer's parameter

        # I do not recommend users mess with this unless they're doing production tuning
        optimizer:
          max_time: 0.10                # maximum compute time for smoother
          max_iterations: 500           # max iterations of smoother
          debug_optimizer: false        # print debug info
          gradient_tol: 1.0e-10
          fn_tol: 1.0e-20
          param_tol: 1.0e-15
          advanced:
            min_line_search_step_size: 1.0e-20
            max_num_line_search_step_size_iterations: 50
            line_search_sufficient_function_decrease: 1.0e-20
            max_num_line_search_direction_restarts: 10
            max_line_search_step_expansion: 50
```  

Important parts of global costmap params:  

```yaml
global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 1.0
      publish_frequency: 1.0
      robot_base_frame: base_link
      global_frame: slam_map
      use_sim_time: True
      plugins: ["static_layer", "obstacle_layer", "3d_obstacle_layer", "inflation_layer"]
      resolution: 0.1
      robot_radius: 0.7
      footprint: "[[0.0, 0.442],
                  [-0.465, 0.395],
                  [-0.495, 0.315],
                  [-0.545, 0.0],
                  [-0.495, -0.315],
                  [-0.465, -0.395],
                  [0.0, -0.442],
                  [0.53, -0.395],
                  [0.565, -0.34],
                  [0.565, 0.34],
                  [0.53, 0.395]]"
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        inflation_radius: 5.5
        cost_scaling_factor: 10.0
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: scan
        scan:
          topic: obstacle_scan
          expected_update_rate: 0.0
          max_obstacle_height: 2.0
          obstacle_range: 30.0
          raytrace_range: 31.0
          clearing: True
          marking: True
          data_type: "LaserScan"
```

SteveMacenski, 2021-03-01:
> Testing with master is very time consuming and I may not have time in the next weeks...

SteveMacenski, 2021-03-11:
@maxlein I grabbed your map file and testing with your inflation layer params with the obstacle layer removed (since I dont have your simulated environment, I'm simulating in a bogus environment so I dont want the impact of the "real" environment to mess with the "mapped" environment). Using the same SMAC planner params as well.

I see the issue - and removed all the params down the defaults just to verify it wasn't some odd combination of params in your example file.

I have not been able to reproduce this though on any other map/robot combo I have that is similar to this - I think its your footprint. When I remove it, the robot radius works fine. Though I'm not sure if that means my collision checking in SMAC is broken or your  formatting of the footprint is incorrect on what costmap_2d expects (as we leverage that in SMAC for footprint-based collision avoidance).

I tried a simpler footprint but no longer see this issue at all.
```
      footprint: "[[0.7, 0.7],
                  [0.7, -0.7],
                  [-0.7, -0.7],
                  [-0.7, 0.7]]"
```

This leads me to believe its probably an issue with your footprint point configuration. Though I have to admit, I don't yet understand why you're seeing what you're seeing so I'm resistant to thinking its just a configuration issue with the footprint (but you  can get around it for now using the radius parameter without the footprint if you like)

SteveMacenski, 2021-03-11:
https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/include/nav2_smac_planner/collision_checker.hpp#L80

This line is giving me pause though why I didn't set this to `INSCRIBED`. Either way, I think we have seen that some of the nodes are over literal collision cells so I'm not convinced it would be that simple. Though that would explain why we don't see this in radius mode...

SteveMacenski, 2021-03-11:
https://github.com/ros-planning/navigation2/pull/2232 please test this out, it should resolve your issue (though planning with footprints is computationally expensive, so I'd still recommend using radius mode unless you're needing to get very close to stuff, your robot is almost circular anyhow with a radius of 0.66m)

maxlein, 2021-03-11:
Thanks for looking into it!  
I tested your fix and it looks like it's doing it's job in that specific case.  
[image]
Although the planning took half a minute on 10cm grid size with footprint active:  
`[planner_server-2] It took 28295.9 milliseconds with 1432030 iterations.`  

With radius mode it's much faster:  
`[planner_server-2] It took 70.8374 milliseconds with 3265 iterations.`

But there is another issue:  
Both / footprint and radius mode sometimes ignore obstacles it seems:  

[image]

Here it's happening when driving back to start point.
[image]

Looks a bit like it's planning on an empty costmap after clearing the costmap.  
But I am not sure if this can be the problem, as these are synchronous actions:  

```
<RateController hz="1.0">
  <RecoveryNode number_of_retries="1" name="ComputePathToPose">
    <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
    <ClearEntireCostmap service_name="global_costmap/clear_entirely_global_costmap"/>
  </RecoveryNode>
</RateController>
```

maxlein, 2021-03-11:
Ok, costmap clearing is the problem...  

[image]

Without clearing it works:  
[image]
Is this maybe a foxy issue, that planning can occur on a not yet ready costmap?  
As it seems the costmap has not received any sensor update since clearing. ( Although my sensor is a static map )

SteveMacenski, 2021-03-11:
Full footprint collision checking is going to be heavier than a single point check, yes. That's why I don't recommend you do that unless you really need it for some reason.

SteveMacenski, 2021-03-11:
We can continue discussion, but closing since initial ticket resolved and now we're into other topic territory

closed 2021-03-11 by SteveMacenski

maxlein, 2021-03-12:
So to sum this up: For foxy the fix is to add a wait node after clearing.

maxlein, 2021-03-12:
Another bug I saw:  
If I place the goal inside the obstacle area, planner is planning endlessly.  
Navfn on the other side fails immediately with an error message.  

[image]

SteveMacenski, 2021-03-12:
Did you play with the settings for iterations on approach and tolerance? We do allow for invalid goals in SMAC so it will attempt to plan to a nearby node that is within tolerance if the literal goal itself is invalid. That's not a bug.

maxlein, 2021-03-12:
No I didn't look at them.
I just assumed there is some sort of error if I do this, and not just block the whole system without a timeout.

SteveMacenski, 2021-03-12:
There is a timeout set by the iterations on approach param

SteveMacenski, 2021-04-01:
@maxlein you mentioned that it was much slower with full footprint checking. Can you try out https://github.com/ros-planning/navigation2/pull/2286 which should speed you up? It only checks the full footprint if the cost of the cell is possibly inscribed.