ros-navigation/navigation2 #2218: Regulated Pure Pursuit not integrated with controller server

kamakshijain, 2021-03-04:
## Question/Bug Report
I am trying to run the example given  on `https://navigation.ros.org/configuration/packages/configuring-regulated-pp.html` and I am getting these errors. Planner Server is of type Smac Planner.
Do you think I missing out something or is regulated pure pursuit not ready for testing.
```
Failed to create controller. Exception: According to the loaded plugin descriptions the class nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController with base class type nav2_core::Controller does not exist. Declared types are  dwb_core::DWBLocalPlanner

[bt_navigator-8] [INFO] [1614884803.707910924] [bt_navigator]: Begin navigating from current location to (3.80, 2.89)
[controller_server-4] [INFO] [1614884803.729285364] [controller_server]: Received a goal, begin computing control effort.
[controller_server-4] [ERROR] [1614884803.729344205] [controller_server]: FollowPath called with controller name FollowPath, which does not exist. Available controllers are FollowPath .
[controller_server-4] [WARN] [1614884803.729356896] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[bt_navigator-8] [ERROR] [1614884803.748488984] [bt_navigator]: Navigation failed
```

**Required Info:**

- Operating System:
  -  Ubuntu 16.04 
- ROS2 Version:
  -  Foxy (inside Docker)
- Version or commit hash:
  -  
- DDS implementation:
  -  rmw_cyclonedds_cpp

Tagging you since you were the creator  @SteveMacenski

SteveMacenski, 2021-03-04:
Please do not tag maintainers.

https://github.com/ros-planning/navigation2/tree/foxy-devel RPP isn't even in foxy, nor do we offer support for 16.04.

kamakshijain, 2021-03-04:
I am extremely sorry. Will keep that in mind for future
Thank you

closed 2021-03-04 by kamakshijain