ros-navigation/navigation2 #2223: Give controllers pointer to goal checker

SteveMacenski, 2021-03-08:
Thus it can use that information locally as well if that state information is useful.

pabloinigoblasco, 2021-03-11:
I experienced cases where that is convenient. 
I have first to say that isolating the concept of goal checker looks correct to me (for the typical cases).

However, I think it is not always possible to decouple the controller and the goal checker. 

For example, the controller may contain semantic about how well it is important to follow the path. Maybe you have a path with a loop and pass twice on the same state. 

It looks to me like the simple_goal_checker is not able to identify that.

For example. In some of our examples we have a "backward-undo planner" that must exactly the same motion it did forward at the inverse direction. You must totally complete the path inversely. Sometimes the path contains loops with some state (x,y,theta) repeated.

We should not stop in that case when we reach the "goal state" first.

SteveMacenski, 2021-03-11:
> For some ways of motion, It looks to me like the simple_goal_checker is not able to identify that and handle this situation.

It is afterall the _simple_ goal checker, but agreed potentially that could be an addition. Though if in a loop, I'd hope that you didn't end up in the same XY at the acceptable theta range to trigger the `true` case - else hey - you didn't even need to do the loop at all, the initial one was good enough by your stated parameters :wink: 

>  Sometimes the path contains loops with some repeated state (x,y,theta). 

That... would be fine? Its only a problem if you **goal** state you go into with the same-ish (x,y,theta) as an intermediate point on the path. If you intersect the same (x,y,theta) as other points on the path, that doesn't matter.

pabloinigoblasco, 2021-03-11:
> In this ticket, I'm suggesting the controllers get a pointer to the current goal checker, not that the goal checker gets access to the current controller. 

I see. 

This case could be even more clear. Lets say, for example that the controller must predict trajectories to evaluate the quality of the trajectories, considering the final state and making sure that the goal is reached but not left (because of the dynamics).

> Though the goal checker doesn't have access to the path right now, that would be a PR I'd happily merge. With the path, current pose, and goal pose, you should trivially be able to detect and not return true on a loop without needing any information about the state of the controller at all in a new GoalChecker plugin.

I agree, passing the path is more convenient, it should not be hard to do. Let my have a try.

SteveMacenski, 2021-03-11:
> This case could be even more clear. Lets say, for example that the controller must predict trajectories to evaluate the quality of the trajectories, considering the final state and making sure that the goal is reached but not left (because of the dynamics).

I'm not sure I totally follow - is that an argument for or against this action?

Right now, none of the controllers I think would directly use the information, but it "feels" like something it would want to have access to. For predictive controllers, this could make alot of sense, I agree, if in the final approach  they wanted validation that it was going to "end well" versus the usual tracking logic

pabloinigoblasco, 2021-03-12:
It is an argument for this action.

In other words. The controller is the decision maker. To evaluate the best decision you may need to evaluate the trajectory and goal reaching. A good goal reaching may depend on the particular parametrization of the goal checker. That means the controller needs to have access to the goal checker to make these predictions about the ending.

SteveMacenski, 2021-03-12:
OK, adding to the galactic milestone so that its done before that initial release so itll be added to the next distro

SteveMacenski, 2021-03-16:
@pabloinigoblasco can you review #2247 It has enough of a change I'd feel better if some other eyes looked at this to make sure I was sane.

pabloinigoblasco, 2021-03-16:
Changes look to me. 
The feature is added seamlessly.

SteveMacenski, 2021-03-16:
Merge impending.

closed 2021-03-16 by SteveMacenski