ros-navigation/navigation2 #2225: nav2_lifecycle_manager::LifecycleManagerClient Pause/Resume Crash

bholewin, 2021-03-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - foxy
- Version or commit hash:
  - ros-foxy-navigation2	0.4.5-1focal.20210203.232630
- DDS implementation:
  - FastRTPS

#### Steps to reproduce issue
- Setup using the Getting Started guide here: https://navigation.ros.org/getting_started/index.html
- After step 3: ```ros2 launch nav2_bringup tb3_simulation_launch.py```
- Once Gazebo finishes loading, use the 2d Pose Estimate to initialize the robot's position. 
- Next click Waypoint mode in the Rviz Navigation 2 panel. 
- Click Navigation2 Goal and select a waypoint 
- Click Start Navigation from the Navigation 2 panel
- Click Pause from the Navigation 2 panel
- Click Resume from the Navigation 2 panel

#### Expected behavior
Robot would resume moving, Rviz and Gazebo would not crash
#### Actual behavior
Rviz and Gazebo crash
#### Additional information

Navigation 2 panel code that handles the pause button press:
https://github.com/ros-planning/navigation2/blob/main/nav2_rviz_plugins/src/nav2_panel.cpp#L316
code for the resume button press: 
https://github.com/ros-planning/navigation2/blob/main/nav2_rviz_plugins/src/nav2_panel.cpp#L331

Console output:
```
[bt_navigator-11] [ERROR] [1615235635.535703295] - :/tmp/binarydeb/ros-foxy-rclcpp-lifecycle-2.3.0/src/lifecycle_node_interface_impl.hpp:change_state:381: Unable to start transition 3 from current state active: Transition is not registered., at /tmp/binarydeb/ros-foxy-rcl-lifecycle-1.1.10/src/rcl_lifecycle.c:350
[lifecycle_manager-13] [INFO] [1615235635.536128951] - lifecycle_manager_navigation:/tmp/binarydeb/ros-foxy-nav2-lifecycle-manager-0.4.5/src/lifecycle_manager.cpp:message:261: Activating waypoint_follower
[waypoint_follower-12] [WARN] [1615235635.536369635] - rcl_lifecycle:/tmp/binarydeb/ros-foxy-rcl-lifecycle-1.1.10/src/rcl_lifecycle.c:rcl_lifecycle_get_transition_by_id:313: No transition matching 3 found for current state unconfigured
[waypoint_follower-12] [ERROR] [1615235635.536409160] - :/tmp/binarydeb/ros-foxy-rclcpp-lifecycle-2.3.0/src/lifecycle_node_interface_impl.hpp:change_state:381: Unable to start transition 3 from current state unconfigured: Transition is not registered., at /tmp/binarydeb/ros-foxy-rcl-lifecycle-1.1.10/src/rcl_lifecycle.c:350
[lifecycle_manager-13] [ERROR] [1615235635.536791293] - lifecycle_manager_navigation:/tmp/binarydeb/ros-foxy-nav2-lifecycle-manager-0.4.5/src/lifecycle_manager.cpp:changeStateForNode:145: Failed to change state for node: waypoint_follower
[lifecycle_manager-13] [ERROR] [1615235635.536829043] - lifecycle_manager_navigation:/tmp/binarydeb/ros-foxy-nav2-lifecycle-manager-0.4.5/src/lifecycle_manager.cpp:resume:243: Failed to resume nodes: aborting resume
[waypoint_follower-12] [ERROR] [1615235635.545342419] - waypoint_follower:/opt/ros/foxy/include/nav2_util/simple_action_server.hpp:error_msg:399: [FollowWaypoints] [ActionServer] Trying to publish feedback when the current goal handle is not active
[waypoint_follower-12] [ERROR] [1615235635.595347923] - waypoint_follower:/opt/ros/foxy/include/nav2_util/simple_action_server.hpp:error_msg:399: [FollowWaypoints] [ActionServer] Trying to publish feedback when the current goal handle is not active
[waypoint_follower-12] [ERROR] [1615235635.645353475] - waypoint_follower:/opt/ros/foxy/include/nav2_util/simple_action_server.hpp:error_msg:399: [FollowWaypoints] [ActionServer] Trying to publish feedback when the current goal handle is not active
[recoveries_server-10] [INFO] [1615235635.670288345] - recoveries_server:/tmp/binarydeb/ros-foxy-nav2-recoveries-0.4.5/include/nav2_recoveries/recovery.hpp:operator():171: spin running...
[waypoint_follower-12] [ERROR] [1615235635.695352677] - waypoint_follower:/opt/ros/foxy/include/nav2_util/simple_action_server.hpp:error_msg:399: [FollowWaypoints] [ActionServer] Trying to publish feedback when the current goal handle is not active
[waypoint_follower-12] [ERROR] [1615235635.745347673] - waypoint_follower:/opt/ros/foxy/include/nav2_util/simple_action_server.hpp:error_msg:399: [FollowWaypoints] [ActionServer] Trying to publish feedback when the current goal handle is not active
[waypoint_follower-12] [ERROR] [1615235635.795402201] - waypoint_follower:/opt/ros/foxy/include/nav2_util/simple_action_server.hpp:error_msg:399: [FollowWaypoints] [ActionServer] Trying to publish feedback when the current goal handle is not active
[ERROR] [rviz2-4]: process has died [pid 208779, exit code -11, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /opt/ros/foxy/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args -r __node:=rviz2'].
```

SteveMacenski, 2021-03-08:
Have you tried CycloneDDS?

bholewin, 2021-03-08:
I had not, but just installed using apt: ```sudo apt install ros-foxy-rmw-cyclonedds-cpp``` 
exported the following: ```export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp```
Then tried the above steps and am still seeing the same issue.

SteveMacenski, 2021-03-09:
```
Click Start Navigation from the Navigation 2 panel
Click Pause from the Navigation 2 panel
Click Resume from the Navigation 2 panel
```

Tested, I understand your issue. "Pause" isn't a "pause navigation" button, its a "deactivate the nav2 stack" button. It transitions the entire stack's lifecycle management down to the inactive state. You should **not** expect that hitting "pause" and then "resume" resumes the goal. Because its not pausing your navigation request, its resetting the entire state of the navigation system by bringing down the lifecycle nodes and transitioning them back up.

SteveMacenski, 2021-03-09:
https://github.com/ros-planning/navigation2/tree/rviz_state_machine_wp_updates The state machine of the RVIZ plugin for the waypoint mode was updated and I believe works properly, but for some reason, I'm not getting the "cancel" button to trigger the cancel transition (even though I think I perfectly mirrored the running state's cancellation sequence). I welcome your thoughts to help bring this fix in for your use.

SteveMacenski, 2021-03-09:
Update: Fixed, please play around with this update. I think you should find that it enables your cancellation feature but also removing invalid buttons in the active navigation state

bholewin, 2021-03-09:
> Update: Fixed, please play around with this update. I think you should find that it enables your cancellation feature but also removing invalid buttons in the active navigation state

Yeah that looks great to me, thank you! Closing out issue.

closed 2021-03-09 by bholewin