ros-navigation/navigation2 #2229: Route Server anologous to Planner Server

SteveMacenski, 2021-03-10:
As the outcome of https://github.com/ros-planning/navigation2/issues/2174, we should have an analog server using the work of #1595 to allow for "route planning" versus "freespace planning" that the planner server enables. 

For indoor applications like warehouse and service robot applications, you're in largely planar space. It makes alot of sense to work with an occupancy grid for planning in these situations since you're typically using 2D or 3D SLAM that can be easily formatted in this way. Then you want to do freespace planning from any A to any B. This is often required as you're going to really different places all of the time.

For larger scale outdoor applications, this is very different. A delivery robot in San Francisco doesn't want to work in cartesian coordinates of a map (navigate from 403432, 4302432 to 90655498, 430200), but GPS or a route graph made up of the city streets. "Freespace planning" like what the planner server via grid costmaps and search algorithms aren't the right way to go. Further, you're not working with 2D SLAM to have a nice representation. Most of the time, its going to be via GPS or VSLAM which don't provide you this occupancy information even if you wanted it. 

Instead, we should allow users to use their semantic representation of the environment (GPS, pre-defined route graphs, city route  graphs, etc) to plan a route for the controller server to follow. So rather than a plan built off of a searching algorithm on a gridmap, these will be routes built from the semantic navigation graph.

For the controller, this is no different, it still seeks to follow the same path/route. But now the route is computed via this semantic method versus a freespace map. 

This will enable VSLAM, 3D SLAM, and GPS systems to plan in all-terrain and urban environments and use the Nav2 system.

SteveMacenski, 2021-03-16:
Add service to find nearest pt on nav graph to a pose, thusly can create BT nodes for this service so that for a navigation task request, we have a bt that:
- finds nearest to start and goal
- plans from nav graph A->B
- appends B->goal
- prepends start->A
- sends to controller

---

- or parts out those in a subtree step by step

---

Demos:
- plan across warehouse large with graph set at start-end of aisles and important points of interest.
- plan across open hilly / construction multistory env to show how can use for outdoor nav without any freespace planning 
---

Roadmap for this work:

- [ ] Complete semantics work / basic MVP  demos with semantics library by itself so that people know how to use it / real-demos of it with practical use / what can be represented by it
- [ ] Create a new route_server which takes in the semantic information and is able to create a nav_msgs/Path based on the navigation graph requested to navigate from A->B or based on some ordered waypoint sequence to travel through. Ordered waypoints is easier, just append with some interpolation between them. For the navigation graph, we'd need to have some navigation-graph-A* to search for the shortest / lowest weight path.
- [ ] Create the action definition for this new route planner with a BT node / XML to utilize it in place of the planner server for semantic navigation
- [ ] Documentation / demos / etc. Most likely outdoors, since that would be the most compelling use-case, or a large networked building like a stupid-large warehouse where freespace planning is impractical.

abylikhsanov, 2023-06-23:
@SteveMacenski Is this still in progress or you need a contributor?

SteveMacenski, 2023-06-23:
This is in progress and is on a temporary pause to deal with some more pressing issues. Right now, most of the work required is integration work w.r.t. exposing the server to the BT, Python3 API, testing and tutorial. If that sounds interesting, I could use the help

abylikhsanov, 2023-06-23:
I would like to say that this could be very useful even for the indoor environments as many robots are shifting from 2D LiDAR SLAM to Visual based (as they are becoming more robust to dynamic changes and less expensive than 3D LiDARs). So creating 2D occupancy map is more of aligning to how the planners work instead of giving any benefits to the SLAM itself.
This is indeed an interesting issue, I will look into it

asasine, 2023-10-16:
Chiming in to say this would be immensely useful for our indoor environment. We represent our indoor environments with simplified adjacency graphs, similar to a road network, that has orders of magnitude less nodes and edges than an occupancy grid. We have an intermediate step in our navigation system which searches this graph using Dijkstra's or advanced arc routing / TSP implementations to efficiently identify paths and circuits that traverse certain aisles (an edge in our graph) and minimize deadheading. The ordered endpoints of these edges in the identified circuit/path are fed to the Nav stack in sequence, which performs costmap planning and local control.

Since it didn't exist in move_base (at the time we introduced it) or Nav2, we've ended up reimplementing some of this feature proposal and #1595 with custom solutions. If it were present in the Nav system at the time, we would have chosen it over a bespoke solution.

asasine, 2023-11-08:
@SteveMacenski based on your [roadmap from above](https://github.com/ros-planning/navigation2/issues/2229#issuecomment-800486328) (emphasis mine), I have a question about the downstream effects of the route server on local controllers.

> Create a new route_server which takes in the semantic information and **is able to create a nav_msgs/Path based on the navigation graph** requested to navigate from A->B or based on some ordered waypoint sequence to travel through.

The implementation of [`nav_msgs/Path`](https://github.com/ros2/common_interfaces/blob/ad69eab30975b967b8c840c690ab491b2d6234cd/nav_msgs/msg/Path.msg) is a vector of poses. I wonder if that constrains the benefit of the route server, which would plan in a semantic space as opposed to a costmap/grid, because it forces the planner to discretize its semantic plan into freespace.  What do you think of supporting planners and controllers that can communicate with more semantic representations?

One example I can think of is highway/street routing, where the plan is represented as a series of parametrized arcs of curvature and length and an Ackerman controller capable of controlling directly in this semantic space.

SteveMacenski, 2024-03-22:
+ demo with using this for grid navigation like used in amazon warehouses with QR codes. Drive to one, stop, rotate, go to next vector, etc. 

In addition to broader indoor route navigation on key junctions + outdoor demos (with semantics edge costs)

SteveMacenski, 2025-04-12:
Imminent to merge!

closed 2025-04-12 by SteveMacenski