ros-navigation/navigation2 #2240: Revert / fix SLAM launch file configuration pop/push failures

SteveMacenski, 2021-03-12:
CC @v-lopez

v-lopez, 2021-03-12:
Could you please share some details on what's the failure?

SteveMacenski, 2021-03-12:
I don't recall off hand, but it should be pretty obvious to you if using `slam:=True` option on the TB3 simulation launch file. The stack doesn't come up.

v-lopez, 2021-03-12:
Yes, I've seen now.

It seems that Pop doesn't pop the configuration, or some tasks are run concurrently and nodes such as the controller_server are started with the param file of `slam_toolbox`.

Looking into it, I apologise for the incovenience.

v-lopez, 2021-03-12:
I believe it's a bug in `launch`. Not sure if we can fix it right now.

The workaround I'd propose is to change the `params-file` argument name to something that doesn't clash, like `slam-toolbox-params-file` or `slam-params-file`.

But it's going to get hard to keep all LaunchConfiguration names unique with the biggest launch files.

What do you think about this?

SteveMacenski, 2021-03-12:
so a different param input file for slam vs everything else? Or just the variable name in a single launch file?

v-lopez, 2021-03-12:
Just the variable name in `online_sync_launch.py` and in `online_async_launch.py`.

So if the nav2_bringup param_file has  parameters for slam_toolbox, I will pass `params-file` from nav2_bringup as `slam-params-file` for slam_toolbox. If it doesn't have such parameters, I will not pass it, and it will not be picked up automatically (which is what happens now, and the reason why the Push, Unset, Pop) was needed.

SteveMacenski, 2021-03-12:
Oh, that would be totally fine

v-lopez, 2021-03-13:
Created PR on ros2/main and foxy branches.

Going to sleep now, I'll address any issues tomorrow if there are.

closed 2021-03-13 by SteveMacenski

SteveMacenski, 2021-03-16:
@v-lopez did you by chance file a ticket about these launch issues we ran into (or was that already reported)?

v-lopez, 2021-03-17:
> @v-lopez did you by chance file a ticket about these launch issues we ran into (or was that already reported)?
https://github.com/ros2/launch/issues/313

Here it is discussed. The current state as described by wjwood:
> We would like to get back to this with launch. Currently an included launch description has access to all configurations of the launch file that included it.

So I believe their intended approach is to:
1. Add a way to explicitely promote arguments from an included launch files to expose them at the current launch file.
2. Disable automatic promotion (current behavior, which is what is causing these issues)