ros-navigation/navigation2 #2251: Update BT nodes to use callback groups / executors to process subscriptions

SteveMacenski, 2021-03-16:
For example: https://github.com/ros-planning/navigation2/blob/main/nav2_util/include/nav2_util/service_client.hpp#L43

- [ ] Battery low
- [ ] Goal updater
- [ ] planner / controller / goal checker selectors

SteveMacenski, 2021-03-17:
It's come to my attention that the executors might be creating a new context remapping to a new DDS participant. This would not be good if we had each BT node creating a new participant to process separately. Some more research has to be done to figure out if this is workable before we jump into it.

Figure out if this does indeed add more participants to the network, if so, if there's a workable way around it so we can process callbacks uniquely in our BT nodes with little overhead.

If not, leave as is.

BriceRenaudeau, 2021-03-17:
Right now, BT nodes are using there parent node("_rclcpp_node") to create subscriber/service/action so there are not creating new context.
May be we can also pass a multi Threaded executor create and spin in the parent node through the blackboard to attach callback groups. 
It will still create a new context but only one for all the BT node and remove the client_node.

SteveMacenski, 2021-03-17:
> BT nodes are using there parent node("_rclcpp_node") to create subscriber/service/action so there are not creating new context

Correct @BriceRenaudeau , but this ticket was considering adding the same logic as you introduced in the simple service client. From the design document I linked to above and a little digging, the impression I'm under is that a new DDS participant is made in each new context. The constructor of the `executor` creates a new context object by default when none is given. 

Therefore, I think that we're actually creating a new network participant in the simple service client, which is probably worse for performance than before. It may be fixable, but I haven't had cycles to look into it.

This ticket is wanting to use that same design with a sub/pub/etc made from `node` but added to a local BT-node specific executor. I don't think we can do this due to the fact that it would create a new context, ergo a new DDS participant.

SteveMacenski, 2021-04-15:
I spoke with Chris from OR and he gave me the key to my understanding on this -- while a new context is constructed on a new executor, when none is given as an input, it uses the default context which stores a static variable to use https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/contexts/default_context.cpp#L23. Therefore, any new executor, node, etc will use that same context regardless, unless you give a context to the object on construction.

Thusly, we can continue with this work one the model of the `ServiceClient` where each BT node can have its own executor for spinning to only receive its isolated messages!

SteveMacenski, 2021-05-18:
merged!

closed 2021-05-18 by SteveMacenski