ros-navigation/navigation2 #2256: Ability to have several global costmap layer configuration in parallel

doisyg, 2021-03-17:
## Feature request

#### Feature description
Starting with a simplified use case description to make it clearer:
I would like to be able to choose at runtime between planning:

A. in a costmap with only a static layer
B. in a costmap with the same static layer + an obstacle layer

From what I understand it is possible to choose between different planner of the `planner_server` at runtime, but they all share the same global costmap configuration.

#### Implementation considerations
I guess I could already achieve what I ask for by using 2 separate `planner_server` instances, each one having a different global costmap configuration. However this would be inefficient : 2 `planner_server` instances instead of 1, and moreover the costmap layers would have to be duplicated (the impact is limited in the use case described above but just replace static layer by STVL to get a sense of the issue).

I have no idea about the implementation, but from a user perspective, I would ideally want to be able to select at the planner plugin level which costmap layers this specific planner plugin is using.

SteveMacenski, 2021-03-18:
It wouldn't really be any much more efficient to have them in the same server since the largest single overhead in the server is the costmap itself. We made the servers independent so that you could run multiple instances (and on other CPUs) if you liked for really particular situations / layer usages. So I think that is the right move.

If you wanted to have multiple costmaps in a given server, I'd be open to a discussion on it, but I don't have a clear vision how that would work in practical terms.

doisyg, 2021-03-23:
Closing as after trying it, it is actually simple to make a custom `planner_server` through inheritance supporting exotic use cases like mine (again, I really appreciate the effort in terms of architecture of nav2)

closed 2021-03-23 by SteveMacenski