ros-navigation/navigation2 #2262: Nav2 equivalent to move_base controller_patience

doisyg, 2021-03-21:
## Feature request

#### Feature description
More than a feature request this is an interogation. In ROS1, `move_base` has a parameter called `controller_patience` described as:

>  How long the controller will wait in seconds without receiving a valid control before space-clearing operations are performed

(i.e. before giving up)

This was further extended in MBF with `planner_max_retries`:
> How many times we will recall the planner in an attempt to find a valid plan before giving up

Is there an equivalent in nav2 ? And if not would a PR be of interest (I need a similar functionality so I will implement it regardless) ?

#### Implementation considerations
I see different way of implementing it

1. (personally preferred) Exposing this parameter in `nav2_controller` and try/catching the call to `computeVelocityCommands` https://github.com/ros-planning/navigation2/blob/3bf1f7a93d740c54b4c7ef7fc6dd914175bf46be/nav2_controller/src/nav2_controller.cpp#L388-L392
And publishing 0 velocities in case of an exception as long we are under `controller_patience`
2. Exposing it at the controller plugin level

SteveMacenski, 2021-03-21:
See the progress checker, if progress isn't made, then the local BT recoveries are triggered. The BT recovery nodes also have `num_of_retries` built into them for the number of attempts at failed planning/control/navigation before failing or triggering a secondary recovery wave.

doisyg, 2021-03-22:
Thanks for you answer,  using BT we can achieve a similar result (i.e. retrying the control action a specific number of times in case of failure).

However I feel that something is missing. I will try to be clear.
In case of the `nav2_core::Controller` plugin throwing a `nav2_core::PlannerException` when calling `computeVelocityCommands`, the `nav2_controller` "FollowPath" action terminates.
And in case the progress checker detects that there is no progress, the `nav2_controller` "FollowPath" action also terminates in a similar way.
Hence there is no way of distinguishing between a "FollowPath" failure due to no progress, and a failure due to the `nav2_core::Controller `plugin outputting invalid commands.

This would be an issue if we wanted to distinguish these 2 cases. This is where I am biased as I am trying to implement the following behavior:

> Make the `nav2_core::Controller` plugin follow a path and retry if needed for a period of time (or a number of time), BUT fail if no progress is detected.

SteveMacenski, 2021-03-22:
>  Hence there is no way of distinguishing between a "FollowPath" failure due to no progress, and a failure due to the nav2_core::Controller plugin outputting invalid commands.

We could potentially resolve this by making use of the `Result` field in `FollowPath` action file. We could have an enum for the reason of termination (success, failed to compute, failed to make progress, failed to find controller id, cancel, preempted, etc). 

Then though, we'd need a way within the BT to use that information if you wanted to make decisions based on it. That might be a simpler problem though?

doisyg, 2021-03-23:
I actually experimented with adding it to `nav2_controller` before you replied. Draft is here https://github.com/ros-planning/navigation2/pull/2264 to support the discussion.

> We could potentially resolve this by making use of the `Result` field in `FollowPath` action file. We could have an enum for the reason of termination (success, failed to compute, failed to make progress, failed to find controller id, cancel, preempted, etc).

I did not think of this solution at first but I like it. That would work well for my use case if the overhead of retrying the whole `FollowPath` action compared to just retrying the plugin is not significant.
Do we want to have a generic fail for the controller plugin (controller plugin computeVelocityCommands failed) or something more composite that could be piped to the `Result` field? The later however would require creating standardized ControllerExceptions or something similar.

> Then though, we'd need a way within the BT to use that information if you wanted to make decisions based on it. That might be a simpler problem though?
That's easy and that doesn't need to be supported/included by/in nav2

SteveMacenski, 2021-03-23:
I reviewed and added comments to your PR. I think we need to resolve some more specific use-case questions / philosophy to figure out what the best solution here is. Either way, we can do this more concisely in code, so I had a review comment that collapses the changes to ~15 lines for the same thing.

SteveMacenski, 2021-04-05:
Merged!

closed 2021-04-05 by SteveMacenski