ros-navigation/navigation2 #2265: Could not load library ibnav2_compute_path_to_pose_action_bt_node.so

ADI201998, 2021-03-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- ROS2 Version:
  - Eloquent binaries
- Version or commit hash:
  - 08f869f722f78b74bab9fa3f1905c78d3d3aa430
         apt binaries: ros-eloquent-navigation2	0.3.5-1bionic.20201207.184754

I launched the `nav2_tb3_simulation_launch.py` and I'm getting this error after clicking on Startup in RViz:
```
[lifecycle_manager-5] [INFO] [lifecycle_manager]: Configuring bt_navigator
[bt_navigator-11] [INFO] [bt_navigator]: Configuring
[bt_navigator-11] [ERROR] []: Caught exception in callback for transition 10
[bt_navigator-11] [ERROR] []: Original error: Could not load library: /home/adi99/ROS2/nav2_ws/install/nav2_behavior_tree/lib/libnav2_compute_path_to_pose_action_bt_node.so: undefined symbol: _ZTIN2BT14CoroActionNodeE
[bt_navigator-11] [WARN] []: Error occurred while doing error handling.
[bt_navigator-11] [FATAL] [bt_navigator]: Lifecycle node entered error state
[lifecycle_manager-5] [ERROR] [lifecycle_manager]: Failed to change state for node: bt_navigator
[lifecycle_manager-5] [ERROR] [lifecycle_manager]: Failed to bring up nodes: aborting bringup
```
The map also dosent load on RViz as `ros2 topic info /map` shows that there are no publishers.
Along with this, I keep getting this:
`[rviz2-3] [INFO] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 11.203 for reason 'Unknown'`

SteveMacenski, 2021-03-23:
Eloquent is EOL, we do not offer any support for it. Please ask on ROS Answers and a community member may be able to help

closed 2021-03-23 by SteveMacenski

Repp-cattt, 2021-03-29:
Hello, I have the same problem as you. Have you solved it?

ADI201998, 2021-03-29:
> Hello, I have the same problem as you. Have you solved it?

Yes I solved it. This [issue](https://github.com/ros-planning/navigation2/issues/1948) addresses the error. Building the navigation2 package and the BehaviourTree.CPP package from source solveed the problem for me.

Repp-cattt, 2021-03-29:
My ros2 is binary installed, do I need to re-use the source code to install ros2 and navigation2 packages?

ADI201998, 2021-03-29:
I dont know about that. Getting the latest version of BehaviourTree.CPP
library and building it, along with building the latest navigation2
eloquent package worked for me.

Repp-cattt, 2021-03-29:
Is your eloquent installed by binary system?

 
 
 &gt; My ros2 is binary installed, do I need to re-use the source code to
 &gt; install ros2 and navigation2 packages?
 &gt; ------------------&amp;nbsp;原始邮件&amp;nbsp;------------------
 &gt; 发件人: ***@***.***&amp;gt;;
 &gt; 收件人: ***@***.***&amp;gt;;
 &gt; 抄送: ***@***.***&amp;gt;; ***@***.***&amp;gt;;
 &gt; ibnav2_compute_path_to_pose_action_bt_node.so (#2265)
 &gt; Hello, I have the same problem as you. Have you solved it?
 &gt; Yes I solved it. This issue addresses the error. Building the navigation2
 &gt; package and the BehaviourTree.CPP package from source solveed the problem
 &gt; for me.

ADI201998, 2021-03-29:
No, I installed it from debian packages.

Repp-cattt, 2021-03-30:
Thank you for your reply. When I finish solving this problem and try to execute the routine, there will be a similar problem: "Failed to change state for node: planner_server" 
Have you ever encountered it? 
By the way, how does your robot navigate? 
As far as I know, ROS2 does not provide a feature pack for robot_pose_ekf.

 
 
 &gt; Is your eloquent installed by binary system?
 &gt; 发件人: "ros-planning/navigation2" ***@***.***&amp;gt;;
 &gt; 发送时间:&amp;nbsp;2021年3月29日(星期一) 晚上6:07
 &gt; ***@***.***&amp;gt;;
 &gt; ***@***.******@***.***&amp;gt;;
 &gt; 主题:&amp;nbsp;Re: [ros-planning/navigation2] Could not load library
 &gt; I dont know about that. Getting the latest version of BehaviourTree.CPP
 &gt; library and building it, along with building the latest navigation2
 &gt; eloquent package worked for me.
 &gt; On Mon, Mar 29, 2021 at 12:24 PM Repp-cattt ***@***.***&amp;gt;
 &gt; &amp;gt; My ros2 is binary installed, do I need to re-use the source code to
 &gt; &amp;gt; install ros2 and navigation2 packages?
 &gt; &amp;gt;
 &gt; &amp;gt;
 &gt; &amp;gt;
 &gt; &amp;gt;
 &gt; &amp;gt; ------------------&amp;amp;nbsp;原始邮件&amp;amp;nbsp;------------------
 &gt; &amp;gt; 发件人: ***@***.***&amp;amp;gt;;
 &gt; &amp;gt; 发送时间: 2021年3月29日(星期一) 下午2:46
 &gt; &amp;gt; 收件人: ***@***.***&amp;amp;gt;;
 &gt; &amp;gt; 抄送: ***@***.***&amp;amp;gt;; ***@***.***&amp;amp;gt;;
 &gt; &amp;gt; 主题: Re: [ros-planning/navigation2] Could not load library
 &gt; &amp;gt; ibnav2_compute_path_to_pose_action_bt_node.so (#2265)
 &gt; &amp;gt;
 &gt; &amp;gt;
 &gt; &amp;gt;
 &gt; &amp;gt;
 &gt; &amp;gt;
 &gt; &amp;gt;
 &gt; &amp;gt; Hello, I have the same problem as you. Have you solved it?
 &gt; &amp;gt;
 &gt; &amp;gt; Yes I solved it. This issue addresses the error. Building the
 &gt; navigation2
 &gt; &amp;gt; package and the BehaviourTree.CPP package from source solveed the
 &gt; problem
 &gt; &amp;gt; for me.
 &gt; &amp;gt;
 &gt; &amp;gt; —
 &gt; &amp;gt;
 &gt; &amp;gt; —
 &gt; &amp;gt; .

ADI201998, 2021-03-30:
Sorry, I didn't encounter any error like that.
I'm running the navigation stack on Gazebo. I'm using the turtlebot3 robot
and cloned the packages from the turtlebot3 and turtlebot3_simulation
repositories.