ros-navigation/navigation2 #2270: [Input Requested] Removal of launch configuration of default BT XML

SteveMacenski, 2021-03-26:
Currently, there are 2 ways of setting the default Behavior Tree XML file:
- Via the commandline-remapping in https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/bringup/launch/tb3_simulation_launch.py#L41 launch
- Via configuration files https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/bringup/params/nav2_params.yaml#L59

Up until now, we haven't had any default set, so the commandline / launch file method gave us a clear way to set a default using the shared file paths to the BT navigator package (since these must be complete filepaths). However, C++ now has this ability and I have a PR coming soon which will use it to set the default behavior tree XML files.

As such, I'm looking to simplify our launch files for readability and I'm not sure remapping in this way is even used by any of our users. I'm looking for information from folks if they're OK with me removing the launch configuration in the launch file and have the BT XML file set via the parameterization only. This parameterization would need to be the full path to the XML file (as would the commandline setting). However, the BT navigator would also have a reasonable default behavior tree XML set using the `ament_cpp_index` package to find the `nav2_bt_navigator` package and its behavior tree files.

joe28965, 2021-04-01:
> This parameterization would need to be the full path to the XML file

In the launch file, this is done using `get_package_share_directory` to get the completed path. I am not aware of a way to do this in the param file (please excuse me if I'm mistaken). Would this not cause issues for multiple people running the same code on their own computers?

SteveMacenski, 2021-04-01:
We do this in C++ in the BT node now (as of #2271) for the default trees to find the location. The parameter file requires a `full/path/to/bt.xml` as it did before as well.

But that has nothing to do with this request. That was non-remappable. That only worked for the default tree (and that logic was just shifted into the BT navigator node itself). The `get_package_share_directory` was never used for user remapped values of the default BT XML, there you also have to specify your own `full/path/to/bt.xml`.

joe28965, 2021-04-01:
I'm sorry, I am afraid I phrased it rather poorly.

What I meant was that it's possible to include the bringup.launch in a custom launch file, as done in the turtlebot example you gave https://github.com/ros-planning/navigation2/blob/53c2d6003db1fba52ba08af3ad730f588a599308/nav2_bringup/bringup/launch/tb3_simulation_launch.py#L181
and change the default xml to a different one in the custom launch file. There, in the custom launch file, one could use `get_package_share_directory` to easily get the path to their own BT (again, as done in the tb3 launch file). 

This way, if it would later run on a different system, the exact path up to the share directory would not matter (the change in username and workspace for instance). By removing the `default_bt_xml_filename` parameter from the launch file, it would not only impact those who used the command line to remap, but also those who change it by calling bringup.launch from their own launch files, or am I mistaken?

SteveMacenski, 2021-04-01:
Ah, because you're including this launch file in another that is passing these values, got it. Are you often changing those values or the paths to your behavior trees? If they're consistent across machines (installed in the same spot), then you could use the configuration yaml file instead with the absolute path. 

This is a reasonable thing then to keep supporting I think - though I'd really like to discuss alternatives so that these Bt XML files aren't a special case (especially since we're going to be adding now 2 different defaults)

joe28965, 2021-04-02:
> I wish within parameter yaml files we could do a `$(find bringup_pkg)/bts/my_bt.xml`

Yeah, that would be the dream.

> We also have the rewitten yaml tool in nav2_common which I think lets you inject / override a parameter in your yaml file. 

Yes, that should work. Now that you mention it, I might have seen that being used somewhere else too. In that case, you can ignore my comment regarding the launch configuration. I'll just update my code to use the override instead.

That was my only issue with removing the launch configuration. I can't think of another reason to keep it. I'm glad my worries were for nothing. Thanks for your quick responses and your insight, Steve

SteveMacenski, 2021-04-02:
Yeah, I think that rewritten yaml tool could find some general reuse in `launch_ros` itself.

 @jacobperron @wjwwood  I see that you two are the more recent developers on `launch_ros` (relatively speaking). We have some utilities we use for launch files in [nav2_common](https://github.com/ros-planning/navigation2/tree/main/nav2_common/nav2_common/launch). In particular, our [rewritten yaml](https://github.com/ros-planning/navigation2/blob/main/nav2_common/nav2_common/launch/rewritten_yaml.py) tool takes in a yaml file and can replace parameter keys by [absolute paths](https://github.com/ros-planning/navigation2/blob/main/nav2_system_tests/src/system/test_system_launch.py#L56-L67) or all [leaf keys](https://github.com/ros-planning/navigation2/blob/dcc1d4dafba6f6c7588cb5f7f6ca918b2a6c48ef/nav2_bringup/bringup/launch/navigation_launch.py#L53-L63). 

Is this something you think is sufficiently useful that I should submit a PR to contribute back to `launch` or `launch_ros` with these? The particular need that I think this fills for the general user is being able to override parameters involving `/absolute/path/to/a/file.extension` which may change by computer/username and can't just be put into a yaml on a larger scale. Something like supporting `$(find my_pkg)/files/file.extension` would also work too in yaml parsing, but obviously a different feature altogether

SteveMacenski, 2021-04-06:
https://github.com/ros-planning/navigation2/pull/2295 will complete @joe28965 can you verify for me this works for you? I don't want to break you

closed 2021-04-07 by SteveMacenski

joe28965, 2021-04-08:
Sorry, I didn't get notified of your comment until you closed this. Yes, it works for me, thanks for your advice with regards to the rewritten yaml tool.