ros-navigation/navigation2 #2272: Would not halt all actions while exception thrown during running BT

KavenYau, 2021-03-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - 41996f409fcd7b005cf779e0376273cee3de6d76

- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

- use `default navigate_w_replanning_and_recovery.xml` without any modifications 
- launch navigation2 according to the tutorials
- use rviz2 to send a goal
- then reproduce while any exception thrown (such as the send_goal request timeout exception)
```
[bt_navigator-4] [1608026552.800316163][bt_navigator][ERROR]: Action server failed while executing action callback: "send_goal failed"
```

If any exception thrown out during running the behavior tree, the 'navigate_to_pose' action will be aborted, but the ‘follow_path’ action will continue to follow the path which received before. 

My Question is why not catch the exceptions in `BehaviorTreeEngine::run()` like:
[image]

So it will call `bt_->haltAllActions(tree_.rootNode());` in `BtNavigator::navigateToPose()` to halt all bt nodes, and be treated as ‘Navigation failed’.

PS: If this is the expected behavior, could you tell me why?

SteveMacenski, 2021-03-30:
>  Action server failed while executing action callback: "send_goal failed"

We have had discussions at length https://github.com/ros2/ros2/issues/1074 about this issue which was started by a performance regression in ROS2 which I expected to be fixed. However, it doesn't look like that's going to happen in the immediate future, but you should look over the ticket to understand why this is happening.

https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L222-L224 we do halt the tree at the end of any `run` execution of the engine just to be sure. This is mostly done defensively since users can specify their own behavior tree nodes which may or may not be properly implemented.

Though you're right that we don't catch the exceptions here, instead the exception is caught in the action server work callbacks. Good timing, because I implemented something in https://github.com/ros-planning/navigation2/pull/2271 to actually notify halt the exceptions better in the navigator rather than in the action server https://github.com/ros-planning/navigation2/pull/2271/files#diff-0f2f78534d5bfd2912092929cbef78c18e670d07db52352e0a6eb0ecd19898c1R202 via a `try/catch` in the BT navigator main execution function. This lets us catch issues, return error, and then still halt things properly. 

I agree though that your internal `catch/return failed` logic may actually be a cleaner solution.

SteveMacenski, 2021-03-30:
I think you're right, its better where you had it then me, I've changed in the pending PR https://github.com/ros-planning/navigation2/pull/2271/commits/39ccd457653406bf31f0d59e8084f125ed72f04b

I'm closing this out since it is complete and will be merged in with #2271 -- you're welcome to continue discussion here and I can reopen if there's more

closed 2021-03-30 by SteveMacenski

nedo99, 2021-04-08:
@SteveMacenski Is there a timeline for merging this to foxy? Or at least a temp patch to fix the issue?

SteveMacenski, 2021-04-08:
I just did a foxy sync on Monday so it will be many weeks out, this was part of a major refactor so its unlikely to ever be synced into foxy. If you submitted a separate PR with just this patch to foxy, I'd probably merge it and it would be given out in the next sync

Serafadam, 2021-07-15:
Hi @SteveMacenski was there a patch to Foxy that addressed this issue? I'm getting this error pretty often in my own application, and I've been able to reproduce this error with this setup (although it's pretty random when it happens):

run `ros2 launch nav2_bringup tb3_simulation_launch.py slam:=True`

and this script https://gist.github.com/Serafadam/9da2f73673f4cacc656f13185615bdc3
Should I switch to some other version other than Foxy?

SteveMacenski, 2021-07-15:
I don't have a plan to migrate it to Foxy, no. It has been migrated to Galactic for you to use. I'd be happy to potentially field a PR where that work is migrated to Foxy, but when I attempted the cherry pick, the sheer number of conflicts made it impractical for me to handle it at the time. It won't be my intent to fix it in Foxy, but I'd review / merge a PR to do so.

roncapat, 2021-11-17:
I ran into the same problem on Foxy. Is the fix needed just the try/catch block in the first post? I may run a fresh gitted foxy branch in my workspace and try to fix if it is not a big task.

KavenYau, 2021-11-18:
@roncapat  After adding the try/catch block, you also need to add this code block to avoid crashing.

https://github.com/ros-planning/navigation2/blob/4dc756adb11f80c45755dde193b5a49e4085da21/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L305-L309

roncapat, 2021-11-18:
Will try on my machine those fixes to see if I can run a simple demo BT. If the patch works i open a PR on foxy-devel

roncapat, 2021-11-18:
After cloning foxy-devel in the workspace, I made a demo without touching the code. The BT works now without fixes. Don't know if it is still necessary to do a PR with the hereby described patches. If there is consensus I can still try and post them.

roncapat, 2021-11-18:
The reason lies in the fact that on Ubuntu the packages are of release 0.4.7 but foxy-devel is forward by [some commits](https://github.com/ros-planning/navigation2/compare/a947ab5656...foxy-devel)