ros-navigation/navigation2 #2273: Simple action server deadlock

KavenYau, 2021-03-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - 41996f4
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

Actually It's similar to this issue #1961

#### Expected behavior
 always accept goals 

#### Actual behavior
the server will not accept goals when the deadlock happens

#### Additional information
- here is the backtrace, https://gist.github.com/KavenYau/562d2180c69d22ee87bfbd5306ced44d

#### backtrace
- Thread 9: main thread
Frame# 6 [nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::handle_cancel](https://github.com/ros-planning/navigation2/blob/41996f409fcd7b005cf779e0376273cee3de6d76/nav2_util/include/nav2_util/simple_action_server.hpp#L102)
Frame# 13 [rclcpp_action::Server<nav2_msgs::action::ComputePathToPose>::call_handle_cancel_callback](https://github.com/ros2/rclcpp/blob/47f21dab3d2c01710c8ea04784e970eec4a92e15/rclcpp_action/include/rclcpp_action/server.hpp#L361)

- Thread 3612: working thread
Frame# 5 [rclcpp_action::Server<nav2_msgs::action::ComputePathToPose>::call_goal_accepted_callback](https://github.com/ros2/rclcpp/blob/47f21dab3d2c01710c8ea04784e970eec4a92e15/rclcpp_action/include/rclcpp_action/server.hpp#L393)
Frame# 9 [rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose>::succeed](https://github.com/ros-planning/navigation2/blob/41996f409fcd7b005cf779e0376273cee3de6d76/nav2_util/include/nav2_util/simple_action_server.hpp#L316)

#### deadlock
-  Thread 9: main thread
  1. [std::lock_guard\<std::mutex\> lock(goal_handles_mutex_);](https://github.com/ros2/rclcpp/blob/47f21dab3d2c01710c8ea04784e970eec4a92e15/rclcpp_action/include/rclcpp_action/server.hpp#L355)
  2. [std::lock_guard\<std::recursive_mutex\> lock(update_mutex_);](https://github.com/ros-planning/navigation2/blob/41996f409fcd7b005cf779e0376273cee3de6d76/nav2_util/include/nav2_util/simple_action_server.hpp#L102)

- Thread 3612: working thread
 1. [std::lock_guard\<std::recursive_mutex\> lock(update_mutex_);](https://github.com/ros-planning/navigation2/blob/41996f409fcd7b005cf779e0376273cee3de6d76/nav2_util/include/nav2_util/simple_action_server.hpp#L312)
 2. [std::lock_guard\<std::mutex\> lock(goal_handles_mutex_);](https://github.com/ros2/rclcpp/blob/47f21dab3d2c01710c8ea04784e970eec4a92e15/rclcpp_action/include/rclcpp_action/server.hpp#L393)

The lock sequences are different.

But if I commented the code [std::lock_guard\<std::recursive_mutex\> lock(update_mutex_);](https://github.com/ros-planning/navigation2/blob/41996f409fcd7b005cf779e0376273cee3de6d76/nav2_util/include/nav2_util/simple_action_server.hpp#L102), it would cause this rclcpp issue https://github.com/ros2/rclcpp/issues/1599

So if using another mutex or some atomic variables to mark the action state. I think it could be solved.
- If it is publishing succeeded state, so reject the cancel request. It returns succeeded state.
- If it is canceling, so abort publishing succeeded state. It returns canceled state.

SteveMacenski, 2021-03-30:
#1961  was fixed by https://github.com/ros2/rclcpp/issues/1285, have you tested after getting updates from that patch or via testing on ROS2 master + nav2 master? It seems like this has actually been resolved for you.

I don't see a reason why `handle_cancel` requires that mutex lock, so I think we could merge that patch if you submitted a PR (though it sounds like that introduces another issue). Do you have a working suggestion?  The vast majority of the state is maintained by the goal handles themselves, not by the simple action server.

SteveMacenski, 2021-04-06:
@KavenYau any update? It's unclear to me if there's something to be done here or if this has already been resolved in other PRs

KavenYau, 2021-04-06:
I am sorry for replying you after so long.  I'm a little busy recently.

> #1961 was fixed by ros2/rclcpp#1285

I think this probably fixed some lock problems of rclcpp itself.

> have you tested after getting updates from that patch or via testing on ROS2 master + nav2 master? It seems like this has actually been resolved for you.

Actually I haven't tested on ROS2 master + nav2 master. but I have seen the part of the source code. It's similar.

`goal_handles_mutex_` is a mutex of rclcpp using to protect `goal_handles_`.  It was acquired before calling user callback `handle_cancel`. and `current_handle_->succeed(result);` also need to acquire the mutex. So if something block on user callback 
 `handle_cancel`. The deadlock problem occurs.

To the rclcpp. It doesn't know where and when dose the user mutex(update_mutex) be acquired. So if it was acquired in `succeeded_current` function first, and the user callback `handle_cancel` was called, it was blocked.

> The vast majority of the state is maintained by the goal handles themselves, not by the simple action server.

I do agree with you. But exception thrown when doing transform from invalid state to any state（`SUCCEEDED` to `CANCELING`）. therefore I think the simple action server have to maintain some necessary state, such as `is_publishing_succeeded_state_` and `is_handling_cancel_request_`.

I think I will submit a PR in few days.

KavenYau, 2021-04-30:
With both ros2/rclcpp#1635 and ros2/rclcpp#1641 were merged, this issue could be closed.

closed 2021-05-01 by SteveMacenski