ros-navigation/navigation2 #2280: Look into ramifications of increasing server_timeout in BT

SteveMacenski, 2021-03-30:
CC https://github.com/ros2/ros2/issues/1074

See if we can increase this using async BT nodes without slowing the tick rate, or if we need to lower the tick rate proportionally to the time we can allot to this. 

18ms is the high number we've seen

SteveMacenski, 2021-03-31:
@KavenYau this is somewhat related to your other ticket. If you wanted to look into this, increasing this timeout will fix the `send_goal` failure mode. We just don't yet understand the ramifications of increasing this. The BT ticks at 100hz (so the 10ms number we have set) so if this takes longer, we need to profile that this is OK, make it OK, or slow down the BT ticking rate so that it is OK. 

I'd be more than happy to seed this to you if you had immediate need to fix it. It might take me a couple of weeks to get to this.

SteveMacenski, 2021-04-01:
This timeout should also be a parameter, right now its hardcoded at 10ms. This can be done if we also do the RUNNING state work as discussed in the working group meeting @naiveHobo if that is going to be part of your fix :smile:

SteveMacenski, 2021-04-15:
@naiveHobo did you make any progress on this?

KavenYau, 2021-04-18:
I increased `server_timeout` to 3000ms for temporary fix the issue. https://github.com/KavenYau/navigation2/commit/565b22af80641432dad1678e428aa93e214bbec.

With using some low performance mechine or high CPU load cases, I guess the number would be higher than 18ms, it's stochastic. This is probably a communication performance issue.

In my opinions, although the timeout is a parameter, it's difficult for me to determine a perfect timeout to set, maybe it's not crucial. I mean slowing down the BT ticking rate could only reduce the number of timeout occurring, 

How about trying to reduce the timeout impact on BT running? or adding threshold of timeout times to BT running?

KavenYau, 2021-04-18:
I increased `server_timeout` to 3000ms for temporarily fixing the issue. https://github.com/KavenYau/navigation2/commit/87e43c45a5b3fe730ca72c7cd3391c7d2ed596b7

With using some low performance mechine or high CPU load cases, I guess the number would be higher than 18ms, it's stochastic. This is probably a communication performance issue.

In my opinions, although the timeout is a parameter, it's difficult for me to determine a perfect timeout to set, maybe it's not crucial. I mean slowing down the BT ticking rate could only reduce the number of timeout occurring, 

How about trying to reduce the timeout impact on BT running? or adding threshold of timeout times to BT running?

SteveMacenski, 2021-04-20:
> This is probably a communication performance issue.

Yeah, we are aware of some regression in rclcpp causing this suddenly to be an issue. I don't think its necessarily crucial to know the exact right number, but the main question of this ticket is the ramifications that increasing that timeout has on the behavior tree system. The tree is being ticked at 100hz, so if this timeout is > 10ms, how does the BT action node respond? Should we be having 10ms-increments spinning until future is complete and then yielding the BT status as `RUNNING` to then spin again next  tick? Can we just naively increase that timeout without any other action, does that slow the BT tick rate or have other consequences?

naiveHobo, 2021-04-22:
> Can we just naively increase that timeout without any other action, does that slow the BT tick rate in those iterations or have other consequences?

Currently, increasing the server timeout only slows down the BT tick rate in those particular iterations when a BT action node sends a goal to an action server and waits for the response. **This does not have any consequences down the line**. For the iteration which took longer than the specified BT execution period, `rclcpp::WallRate` (that is used to maintain the tick rate for BT execution) does not sleep and returns false while also resetting its internal state to ensure that the succeeding iterations run at the specified rate.

SteveMacenski, 2021-04-26:
But what happens if the duration its waiting for the goal request is larger than the rate duration? (e.g. we iterate the tree at 100hz, what if it takes 100ms to return a goal) In that case, the tick rate of the tree would be lowered. Within the current implementation, that isn't a huge deal, but there may be other BT nodes in the future that rely on more regular timing.

I also think that this is kind of the purpose of the `RUNNING` state to be able to have longer-running things be revisited in future ticks. Wouldn't it be more "general" if we had it yield and spin again in a future `tick` up to the timeout duration rather than blocking the entire tree?

SteveMacenski, 2021-05-13:
Merging imminent, closing, thanks @naiveHobo !!

SteveMacenski, 2021-05-13:
And... that's it! I think that's the full V1 ticket list! I'll need to go around and see if anything else popped up we should add before releasing a V1, but this seems like it !!

closed 2021-05-13 by SteveMacenski