ros-navigation/navigation2 #2282: rosplanning/navigation2:main.release should install debs instead of building from source

ipa-hsd, 2021-03-31:
## Bug report
I am using the docker image `rosplanning/navigation2:main.release` and even though it has been tagged `release`, it seems that the `navigation2` packages are built from source. I am guessing [this](https://github.com/ruffsl/navigation2/blob/main/.dockerhub/distro.Dockerfile) is the `Dockerfile`? 

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy docker image `rosplanning/navigation2:main.release` 

#### Steps to reproduce issue

```
$ docker run -it --rm rosplanning/navigation2:main.release /bin/bash
# apt install ros-foxy-turtlebot3
```

#### Expected behavior
Just installs `turtlebot3` and few additional packages.
#### Actual behavior
it installs the `navigation2` debs again 
```
After this operation, 1602 MB of additional disk space will be used.
```

SteveMacenski, 2021-03-31:
The docker container is based on `ros2` master builds, **not** foxy, so you cannot simply apt install

ipa-hsd, 2021-03-31:
> The docker container is based on ros2 master builds

Ah should have realized by the tag name.