ros-navigation/navigation2 #2283: [tutorials] Python: Create basic python example usage of Nav2

SteveMacenski, 2021-03-31:
Base it off of the nav through poses / nav to pose tester nodes in nav2_system_tests
- Create object to take in goal poses to travel to, call action, and either block or poll for completion - outside API pythonic, no ROS types (commandline params?)
- On startup, wait for BT navigator to be up to know ready for requests
- On startup, optionally set the initial pose to automate the full startup sequence if this were launched in parallel with stack without any rviz operations / setting poses.

trunc8, 2021-05-08:
Hi @SteveMacenski, I'm keen to work on this task. But I have very little experience with ROS2 ([blog example](https://trunc8.github.io/2021/04/17/tut-ros2)), and I would be grateful for any guidance!
I have extensive experience with Docker. Would that be useful?
I am currently looking at the various options in [the build guide](https://navigation.ros.org/build_instructions/index.html). For this task, which build would be more convenient to work with?

SteveMacenski, 2021-05-10:
@wilcobonestroo mentioned he might be interested in this as well, so I want to give him a ping to find out. Maybe if he's working on the example for one language, you could work on the other? 

Probably the build manually from main, since the Navigate To Pose action definition changed slightly between what's available now in Foxy and what will be released in the coming weeks for Galactic. Docker would be useful if you want to install it that way, but I think this example usage demo is more about the code than the install / build method :smile: 

I'd super appreciate your assistance here! 

I think the tl;dr of this task is 
- Create a class in {python3, C++} to wrap the minimum needed capabilities to treat Navigation as a task (e.g. `initialize()` `goToPose()` `waitForNavigationInitialized()` `sendInitialPose()`, etc)
- Use that in a `main()` function to go a basic demo
- Write up a tutorial page on navigation.ros.org regarding it!

SteveMacenski, 2021-06-02:
Here's a basic navigator in python3 I made based on the system tests for the ros developers' day conference talk tutorial.

The only thing I think that really might want to be changed about this for this tutorial _might_ be to have it return and spin in the application (with a "getfeedback" or "checkstatus" function) so that it can be cancelled inline. Right now, this blocks until total completion.  

```
#! /usr/bin/env python3
# Copyright 2021 Samsung Research America
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

import time

from action_msgs.msg import GoalStatus
from geometry_msgs.msg import Pose
from geometry_msgs.msg import PoseWithCovarianceStamped
from lifecycle_msgs.srv import GetState
from nav2_msgs.action import NavigateThroughPoses, NavigateToPose

import rclpy

from rclpy.action import ActionClient
from rclpy.node import Node
from rclpy.qos import QoSDurabilityPolicy, QoSHistoryPolicy, QoSReliabilityPolicy
from rclpy.qos import QoSProfile

class BasicNavigator(Node):
    def __init__(self):
        super().__init__(node_name='basic_navigator')
        self.initial_pose = Pose()
        self.goal_handle = None
        self.result_future = None

        amcl_pose_qos = QoSProfile(
          durability=QoSDurabilityPolicy.TRANSIENT_LOCAL,
          reliability=QoSReliabilityPolicy.RELIABLE,
          history=QoSHistoryPolicy.KEEP_LAST,
          depth=1)

        self.initial_pose_received = False
        self.nav_through_poses_client = ActionClient(self,
                                                     NavigateThroughPoses,
                                                     'navigate_through_poses')
        self.nav_to_pose_client = ActionClient(self, NavigateToPose, 'navigate_to_pose')
        self.model_pose_sub = self.create_subscription(PoseWithCovarianceStamped,
                                                       'amcl_pose',
                                                       self._amclPoseCallback,
                                                       amcl_pose_qos)
        self.initial_pose_pub = self.create_publisher(PoseWithCovarianceStamped,
                                                      'initialpose',
                                                      10)

    def setInitialPose(self, initial_pose):
        self.initial_pose_received = False
        self.initial_pose = initial_pose
        self._setInitialPose()

    def goThroughPoses(self, poses):
        # Sends a `NavToPose` action request and waits for completion
        self.debug("Waiting for 'NavigateToPose' action server")
        while not self.nav_through_poses_client.wait_for_server(timeout_sec=1.0):
[... 147 lines cut ...]
    initial_pose.orientation.w = 0.0
    navigator.setInitialPose(initial_pose)

    # Wait for navigation to fully activate
    navigator.waitUntilNav2Active()

    # Go to our demos first goal pose
    goal_pose = PoseStamped()
    goal_pose.header.frame_id = 'map'
    goal_pose.header.stamp = navigator.get_clock().now().to_msg()
    goal_pose.pose.position.x = -2.0
    goal_pose.pose.position.y = -0.5
    goal_pose.pose.orientation.w = 1.0
    navigator.goToPose(goal_pose)

    ################################################
    #
    # Implement some code here for your application!
    #
    ################################################

    # cancel just to make sure nothing is still processing
    navigator.cancelNav()
    exit(0)
```

SteveMacenski, 2021-06-16:
https://github.com/ros-planning/navigation2/pull/2411 meets this need

closed 2021-06-16 by SteveMacenski

devvaibhav455, 2022-08-17:
Thanks Steve for the script. When I run it, I get
`[INFO] [1660761276.859200026] [basic_navigator]: amcl/get_state service not available, waiting...`
Any idea what to do for this?

HappySamuel, 2022-11-14:
Hi @devvaibhav455 
I also faced similar issue as you, when i used other localization method (instead of amcl), i get following:
`[INFO] amcl/get_state service not available, waiting...`
and nav2_simple_commander will no longer proceed to the next stage, is there a solution to solve this?

Best,
Samuel

devvaibhav455, 2022-11-14:
@HappySamuel : The cause of this error is that AMCL is not working. For my application, I didn't require AMCL/ no localization at all, so, I removed that part and just used that part through which I could command the robot to move to x,y. If your application requires you to run amcl, sorry, but I don't have any idea. All the best!

nakai-omer, 2023-05-07:
For anyone else who has this problem, but doesn't use `amcl`, figures `waitUntilNav2Active` is implicitly waiting for an `amcl` node. We are using slam_toolbox, and `waitUntilNav2Active` is expecting the localizer to expose a `get_state` service, which slam toolbox doesn't, so a dirty workaround would be to use like this: `waitUntilNav2Active(localizer="bt_navigator")`. But I guess it would just be better not to use it at all.

HuaYuXiao, 2023-12-22:
> localizer="bt_navigator"

```bash
tony@iqr-turtlebot4-121:~/EE211_ws/src/tb4_find_pickup_place$ /bin/python3 /home/tony/EE211_ws/src/tb4_find_pickup_place/navigation/pose_reach.py
[INFO] [1703211465.668163371] [basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1703211466.669899601] [basic_navigator]: bt_navigator/get_state service not available, waiting...
```

> For anyone else who has this problem, but doesn't use `amcl`, figures `waitUntilNav2Active` is implicitly waiting for an `amcl` node. We are using slam_toolbox, and `waitUntilNav2Active` is [...]

Sorry, but this actually won't work either.

```bash
tony@iqr-turtlebot4-121:~/EE211_ws/src/tb4_find_pickup_place$ /bin/python3 /home/tony/EE211_ws/src/tb4_find_pickup_place/navigation/pose_reach.py
[INFO] [1703211465.668163371] [basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1703211466.669899601] [basic_navigator]: bt_navigator/get_state service not available, waiting...
```