ros-navigation/navigation2 #2285: The robot cannot avoid obstacles and replan its route in the process of navigation.

miku54, 2021-04-01:
## Bug report
In the process of navigation, the robot can not avoid obstacles and replan the route, there are display lines of global planning path and local planning path, and the map is refreshed, but the route cannot be re-planned.
**Required Info:**

- Operating System:
ubuntu18.04 /ubuntu20.04
- ROS2 Version:
Eloquent /foxy
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```
launch navigation2
set 2D Pose Estimate A 
set goal B
set obstacle on A to B path
```

#### Expected behavior
ROS2 navigation can avoid obstacles while set goal from A to B.

#### Actual behavior
ROS2 navigation cannot avoid obstacles while set goal from A to B.
And stopped when he encountered an obstacle.
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2021-04-01:
Please file a question on ROS Answers with about 10x more details, configuration guides, log files, and information.

closed 2021-04-01 by SteveMacenski

miku54, 2021-04-02:
Thank you. I've taken care of it.