ros-navigation/navigation2 #2297: map_saver failed to write map during turtlebot3 SLAM simulation

matteoderose, 2021-04-06:
Hi, 
I'm running the SLAM simulation with turtlebot3 and trying to save the map in a custom folder (`/ros2_maps`). But it can't save it.

When I run: `$ ros2 run nav2_map_server map_saver_cli -f ~/ros2_maps/<my_map_name>`

**Required Info:**

- Operating System:
  - Linux Ubuntu 20.04 on a VM through VirtualBox
- ROS2 Version:
  - ROS2  Foxy binaries
- Version or commit hash:
  - ros-foxy-navigation2	0.4.5-1focal.20210226.004354
- DDS implementation:
  - Fast-RTPS (default)

#### Steps to reproduce issue
Steps reproduced in this video: https://www.youtube.com/watch?v=8w3xhG1GPdo&t=184s
It refers to turtlebot3 emanual of ROBOTIS (https://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#gazebo-simulation) with slight modifications for the installation. After installing gazebo11, nav2 and turtlebot3 packages I run:
Terminal 1:
```
$ export TURTLEBOT3_MODEL=burger
$ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```

Terminal 2:
```
$ export TURTLEBOT3_MODEL=burger
$ ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True
```

Terminal 3:
```
$ export TURTLEBOT3_MODEL=burger
$ ros2 run turtlebot3_teleop teleop_keyboard
```

After having navigated the whole map, create a new terminal and save the map in a folder in my home directory:
Terminal 4:
$ ros2 run nav2_map_server map_saver_cli -f ~/ros2_maps/<map_name>

#### Expected behavior
I expect to succesfully save the map.

#### Actual behavior
```
[INFO] [1617617193.971037644] [map_saver]: 
	map_saver lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1617617193.974981652] [map_saver]: Creating
[INFO] [1617617193.976031993] [map_saver]: Saving map from 'map' topic to '/home/<username>/ros2_maps/<my_map_name>' file
[WARN] [1617617193.976269143] [map_saver]: Free threshold unspecified. Setting it to default value: 0.250000
[WARN] [1617617193.976499318] [map_saver]: Occupied threshold unspecified. Setting it to default value: 0.650000
[WARN] [map_io]: Image format unspecified. Setting it to: pgm
[INFO] [map_io]: Received a 124 X 115 map @ 0.05 m/pix
[INFO] [map_io]: Writing map occupancy data to /home/<username>/ros2_maps/<my_map_name>.pgm
[ERROR] [map_io]: Failed to write map for reason: Magick: Unable to open file (/home/<username>/ros2_maps/<my_map_name>.pgm) reported by magick/blob.c:3058 (OpenBlob)
[ERROR] [1617617196.070815865] [map_saver]: Failed to save the map
[INFO] [1617617196.070913081] [map_saver]: Destroying
```

#### Additional information
I think It's a matter of permissions of the folder /ros2_maps. I tried to change them allowing to create and delete files to all users, but It keeps outputing the same error.

How to solve it? I'd like to save my map in a specific folder, rather than free in the home directory.

SteveMacenski, 2021-04-06:
Did you try with another given location? Are you trying overwrite a file? Does that directory even exist? Please give us the pertinent information. 

I just tested this locally with no issues with our recommended tutorial setup as well https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html.

> [ERROR] [map_io]: Failed to write map for reason: Magick: Unable to open file (/home/<username>/ros2_maps/<my_map_name>.pgm) reported by magick/blob.c:3058 (OpenBlob)

Also, `~` is not a valid operator for map server, as the path `/home/<username>/ros2_maps/<my_map_name>.pgm` clearly suggests, its try to access in the global space.

matteoderose, 2021-04-07:
I'm really sorry, I mispelled the name of the folder. Now it works. However, I run the simulation using cartographer package and not SLAM toolbox. I don't have a physical robot and maybe that's why the SLAM tutorial from the official doc you linked keeps giving me errors.

closed 2021-04-07 by SteveMacenski