ros-navigation/navigation2 #2299: Costmap2DROS crashes when adding a dummy vector on it

AlexeyMerzlyakov, 2021-04-07:
During the development the solution for #2051 it was revealed that adding some dummy `std::vector` (or `std::queue`) as a field of `Costmap2DROS` class causing crashes of process utilizing it.

## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- ROS2 Version:
Rolling ROS2 build from March 15 2021
- Version or commit hash:
ed43bed86253961f2723bb38389e51cabf6e7ebf

#### Steps to reproduce issue
1. Add a dummy vector as a `Costmap2DROS` class variable. It could be like follows:
```
diff --git a/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_ros.hpp b/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_ros.hpp
index 3d6011f4..8f6bbf54 100644
--- a/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_ros.hpp
+++ b/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_ros.hpp
@@ -310,6 +310,7 @@ protected:
   std::vector<std::string> default_types_;
   std::vector<std::string> plugin_names_;
   std::vector<std::string> plugin_types_;
+  std::vector<int> vector_;
   double resolution_{0};
   std::string robot_base_frame_;   ///< The frame_id of the robot base
   double robot_radius_;
```
2. Re-build nav2_costmap_2d and run navigation in a Gazebo simulation:
```
ros2 launch nav2_bringup tb3_simulation_launch.py 2>&1 | tee run.log
```
3. In opened RViz set the "Initial Pose" and observe a crash randomly appearing on configuring stage:

Case1:
```
[planner_server-9] [INFO] [1617811573.542616714] [planner_server]: Configuring
[planner_server-9] [INFO] [1617811573.542654889] [global_costmap.global_costmap]: Configuring
[planner_server-9] [INFO] [1617811573.553745652] [global_costmap.global_costmap]: Using plugin "static_layer"
[planner_server-9] [INFO] [1617811573.566978734] [global_costmap.global_costmap]: Subscribing to the map topic (/map) with transient local durability
[planner_server-9] [INFO] [1617811573.569086633] [global_costmap.global_costmap]: Initialized plugin "static_layer"
[planner_server-9] [INFO] [1617811573.569132773] [global_costmap.global_costmap]: Using plugin "obstacle_layer"
[planner_server-9] [INFO] [1617811573.576808407] [global_costmap.global_costmap]: Subscribed to Topics: scan
[planner_server-9] [INFO] [1617811573.578420594] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 384 X 384 at 0.050000 m/pix
[planner_server-9] [INFO] [1617811573.602704761] [global_costmap.global_costmap]: Initialized plugin "obstacle_layer"
[planner_server-9] [INFO] [1617811573.602752776] [global_costmap.global_costmap]: Using plugin "inflation_layer"
[planner_server-9] [INFO] [1617811573.611509289] [global_costmap.global_costmap]: Initialized plugin "inflation_layer"
[ERROR] [planner_server-9]: process has died [pid 42610, exit code -11, cmd '/home/leha/navigation2_ws/install/nav2_planner/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmp1rrl12q7 -r /tf:=tf -r /tf_static:=tf_static'].
```
Case2:
```
[planner_server-9] [INFO] [1617812348.380161031] [planner_server]: 
[planner_server-9] 	planner_server lifecycle node launched. 
[planner_server-9] 	Waiting on external lifecycle transitions to activate
[planner_server-9] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-9] [INFO] [1617812348.446100754] [planner_server]: Creating
[planner_server-9] [INFO] [1617812348.569635501] [global_costmap.global_costmap]: 
[planner_server-9] 	global_costmap lifecycle node launched. 
[planner_server-9] 	Waiting on external lifecycle transitions to activate
[planner_server-9] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-9] [INFO] [1617812348.580621740] [global_costmap.global_costmap]: Creating Costmap
...
[lifecycle_manager-13] [INFO] [1617812348.901554259] [lifecycle_manager_navigation]: Configuring planner_server
[ERROR] [planner_server-9]: process has died [pid 43506, exit code -11, cmd '/home/leha/navigation2_ws/install/nav2_planner/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmpkj8s1ig4 -r /tf:=tf -r /tf_static:=tf_static'].
```
Case3:
```
[INFO] [planner_server-9]: process started with pid [44345]
[planner_server-9] [INFO] [1617812503.458910979] [planner_server]: 
[planner_server-9] 	planner_server lifecycle node launched. 
[planner_server-9] 	Waiting on external lifecycle transitions to activate
[planner_server-9] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-9] [INFO] [1617812503.521983101] [planner_server]: Creating
[planner_server-9] [INFO] [1617812503.633003466] [global_costmap.global_costmap]: 
[planner_server-9] 	global_costmap lifecycle node launched. 
[planner_server-9] 	Waiting on external lifecycle transitions to activate
[planner_server-9] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-9] [INFO] [1617812503.637080292] [global_costmap.global_costmap]: Creating Costmap
[planner_server-9] corrupted size vs. prev_size
[ERROR] [planner_server-9]: process has died [pid 44345, exit code -6, cmd '/home/leha/navigation2_ws/install/nav2_planner/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server --params-file /tmp/tmpjly8p2fq -r /tf:=tf -r /tf_static:=tf_static'].
[lifecycle_manager-13] [INFO] [1617812503.913411011] [lifecycle_manager_navigation]: Configuring planner_server
```

#### Notes
- The problem was detected for both "planner_server" and "controller_server" under different run and patch conditions.
- Adding dummy `std::vector` to other classes in nav2_costmap_2d won't cause the problem.
- Confirmed: `std::queue` is also causes the same issue. `std::string` does not.

Investigation is in progress.

SteveMacenski, 2021-04-07:
I do not understand what you mean at all, can you provide a code example? I do not know where you are sticking an arbitrary container. Why would you think inserting an arbitrary object as an argument would _not_ cause problems?

AlexeyMerzlyakov, 2021-04-07:
> I do not understand what you mean at all, can you provide a code example?

I inserting an empty vector object (default std::vector constructor [makes it empty](http://www.cplusplus.com/reference/vector/vector/vector/)) as a `Costmap2DROS` class' variable. Then running navigation2 stack as usual, and planner/controller crashes.

> I do not know where you are sticking an arbitrary container.

In any place into  `Costmap2DROS` class, for example as shown in patch from the description.

> Why would you think inserting an arbitrary object as an argument would not cause problems?

This object is never used anywhere in nav2 stack. Just being declared. This should not cause no problems. However after this simple patch from above, planner/controller servers became to crash, which is completely unexpected behavior.

SteveMacenski, 2021-04-07:
Please provide the code you changed, I don't know what you're talking about.

SteveMacenski, 2021-04-08:
Did you try a clean rebuild? Maybe something in your local workspace cache is messed up

AlexeyMerzlyakov, 2021-04-12:
Yes, I did clean re-build for only "nav2_costmap_2d", but not for all packages. Completely removing build/install/log and re-building whole nav2 stack resolves the situation.
Thank you for the advice! That is explanation why the backtraces were looking very strange.
Closing the issue with "won't fix" label.

closed 2021-04-12 by AlexeyMerzlyakov