ros-navigation/navigation2 #2300: RewrittenYaml key_rewrites

mkolodziejczyk-piap, 2021-04-07:
Hi, RewrittenYaml has a handy feature to rewrite keys. AFAIK it's not used currently in the lauchfiles anywhere, but I tried to use it for example to choose controller
But probably because of no usage the function that actually handle keys rewriting is buggy:
https://github.com/ros-planning/navigation2/blob/aaf25612a9238e62db12b1ae3951993c2585da83/nav2_common/nav2_common/launch/rewritten_yaml.py#L135-L142

1. We can't delete the key in for loop on items() (this causes "RuntimeError: dictionary keys changed during iteration")
2. To go down to lower levels higher levels keys have to be included in key_rewrites dict even if they don't change.

In general I managed to make it work with the following code, but it's not very elegant and one still has to put dummy higher level keys in key_rewrites
```
  def substitute_keys(self, yaml, key_rewrites):
    if len(key_rewrites) != 0:
      for key in list(yaml.keys()):
        val = yaml[key]
        if isinstance(val, dict) and key in key_rewrites:
          new_key = key_rewrites[key]
          yaml[new_key] = yaml[key]
          if key!=new_key:
            del yaml[key]
            val = yaml[new_key]
          self.substitute_keys(val, key_rewrites)
```
I can make a PR for this unless someone has better fix?

Regards,

SteveMacenski, 2021-04-07:
It is being used  in the nav2 system tests for selecting an algorithm to run from an environmental variable. You shouldn't be using this to remove yaml, only overriding values, is that suggested blob just to remove a yaml entry?

mkolodziejczyk-piap, 2021-04-08:
I'm using it in the following way. Let's say I have one big nav2_param.yaml file with parameters for more than one controller
```
controller_plugins: ["FollowPath"]
# ...
TEBController:
# teb parameters
MPCController:
# mpc parameters

# No FollowPath entry!
```
Then in key_rewrites I choose the controller I'd like to run by replacing the name in nav2_params.yaml with FollowPath, e.g.
```
key_rewrites = {'MPCController': 'FollowPath'}
```
This is kind of hack, but I couldn't find easier way to do this.

Could you point me to the test where it is used? I couldn't find in search engine

Still I think that currently this del yaml[key] line will break things

SteveMacenski, 2021-04-14:
Any movement?

SteveMacenski, 2021-04-28:
@mkolodziejczyk-piap is this still a concern?

SteveMacenski, 2021-05-07:
Closing from inactivity, if there's an issue, we can reopen

closed 2021-05-07 by SteveMacenski