ros-navigation/navigation2 #2301: nav2_smac_planner: curvature term in smoothing

mkolodziejczyk-piap, 2021-04-07:
Hi, in curvature term in smoothing we want the turning radius to be close to max_curvature. 
https://github.com/ros-planning/navigation2/blob/aaf25612a9238e62db12b1ae3951993c2585da83/nav2_smac_planner/include/nav2_smac_planner/smoother_cost_function.hpp#L267-L276
Is this correct? I'd rather expect minimize curvature_params.turning_rad and huge penalty if curvature_params.turning_rad > _params.max_curvature?

Regards,

SteveMacenski, 2021-04-07:
Please read the Hybrid-A* paper from Thrun about this term - it penalizes when the radius is larger than the admissible turning radius

mkolodziejczyk-piap, 2021-04-08:
Thanks for this hint. I didn't know that this smoothing comes from original paper. If I still have some doubts, I'll come back

SteveMacenski, 2021-04-08:
OK, closing since reference was provided - in general though there's work to be done in the smoother to improve performance. Its on my TODO list, but its relatively low. It works, but some of the terms and the functional performance CPU time isn't where I'd like (it may even be better to do a more naive gradient descent than using Ceres, I believe that is providing alot of overhead). I'd love input from you to fix it up or another solution that still respects the costmap values in the smoothing process.

closed 2021-04-08 by SteveMacenski