ros-navigation/navigation2 #2303: SIGSEGV in bt_navigator

tzskp1, 2021-04-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- ROS2 Version:
  - Rolling Source
- Version or commit hash:
  - 8a7a4f261ccfa5db68d5add2fa493b32bacc6f33
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
I met SIGSEGV on an arm device.
backtrace is here.

```
#0  0x0000007fb6f5b6e4 in __GI___pthread_mutex_lock (mutex=0xa0) at pthread_mutex_lock.c:67
#1  0x0000007fb7edaa7c in __gthread_mutex_lock(pthread_mutex_t*) ()
   from /ros/install/behaviortree_cpp_v3/lib/libbehaviortree_cpp_v3.so
#2  0x0000007fb7ee0788 in std::mutex::lock() () from /ros/install/behaviortree_cpp_v3/lib/libbehaviortree_cpp_v3.so
#3  0x0000007fa30eadf8 in std::lock_guard<std::mutex>::lock_guard (__m=..., this=<synthetic pointer>)
    at /usr/include/c++/7/bits/std_mutex.h:162
#4  rclcpp_action::ClientGoalHandle<nav2_msgs::action::ComputePathThroughPoses>::get_status (this=<optimized out>)
    at /ros2_master/install/rclcpp_action/include/rclcpp_action/client_goal_handle_impl.hpp:101
#5  nav2_behavior_tree::BtActionNode<nav2_msgs::action::ComputePathThroughPoses>::should_cancel_goal (this=0x7f7c00a050)
    at /ros/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp:250
#6  nav2_behavior_tree::BtActionNode<nav2_msgs::action::ComputePathThroughPoses>::halt (this=0x7f7c00a050)
    at /ros/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp:223
#7  0x0000007fb7f05234 in std::function<void (BT::TreeNode*)>::operator()(BT::TreeNode*) const ()
   from /ros/install/behaviortree_cpp_v3/lib/libbehaviortree_cpp_v3.so
#8  0x0000007fb7f042e4 in BT::applyRecursiveVisitor(BT::TreeNode*, std::function<void (BT::TreeNode*)> const&) ()
   from /ros/install/behaviortree_cpp_v3/lib/libbehaviortree_cpp_v3.so
#9  0x0000007fb7f04378 in BT::applyRecursiveVisitor(BT::TreeNode*, std::function<void (BT::TreeNode*)> const&) ()
   from /ros/install/behaviortree_cpp_v3/lib/libbehaviortree_cpp_v3.so
#10 0x0000007fb7f04378 in BT::applyRecursiveVisitor(BT::TreeNode*, std::function<void (BT::TreeNode*)> const&) ()
   from /ros/install/behaviortree_cpp_v3/lib/libbehaviortree_cpp_v3.so
#11 0x0000007fb7f043d0 in BT::applyRecursiveVisitor(BT::TreeNode*, std::function<void (BT::TreeNode*)> const&) ()
   from /ros/install/behaviortree_cpp_v3/lib/libbehaviortree_cpp_v3.so
#12 0x0000007fb7f04378 in BT::applyRecursiveVisitor(BT::TreeNode*, std::function<void (BT::TreeNode*)> const&) ()
   from /ros/install/behaviortree_cpp_v3/lib/libbehaviortree_cpp_v3.so
#13 0x0000007fb7f04378 in BT::applyRecursiveVisitor(BT::TreeNode*, std::function<void (BT::TreeNode*)> const&) ()
   from /ros/install/behaviortree_cpp_v3/lib/libbehaviortree_cpp_v3.so
#14 0x0000007fb6c13424 in nav2_behavior_tree::BehaviorTreeEngine::haltAllActions (this=<optimized out>,
    root_node=0x7f7c009110) at /ros/src/navigation2/nav2_behavior_tree/src/behavior_tree_engine.cpp:120
#15 0x0000007fb73aa3f4 in nav2_behavior_tree::BtActionServer<nav2_msgs::action::NavigateThroughPoses>::executeCallback (
    this=0x55559f2cf0) at /ros/install/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp:224
#16 0x0000007fb73b800c in std::function<void ()>::operator()() const (this=0x5555c4a950)
```

I guess that the problem is null pointer access of `goal_handle_` in bt_action_node.

```
diff --git a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp b/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp
index a63ac790..e93e4f2a 100644
--- a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp
+++ b/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp
@@ -182,12 +182,14 @@ public:
       // user defined callback. May modify the value of "goal_updated_"
       on_wait_for_result();
-      auto goal_status = goal_handle_->get_status();
-      if (goal_updated_ && (goal_status == action_msgs::msg::GoalStatus::STATUS_EXECUTING ||
-        goal_status == action_msgs::msg::GoalStatus::STATUS_ACCEPTED))
-      {
-        goal_updated_ = false;
-        on_new_goal_received();
+      if (goal_handle_) {
+          auto goal_status = goal_handle_->get_status();
+          if (goal_updated_ && (goal_status == action_msgs::msg::GoalStatus::STATUS_EXECUTING ||
+                      goal_status == action_msgs::msg::GoalStatus::STATUS_ACCEPTED))
+          {
+              goal_updated_ = false;
+              on_new_goal_received();
+          }
       }
       rclcpp::spin_some(node_);
@@ -242,7 +244,7 @@ protected:
   bool should_cancel_goal()
   {
     // Shut the node down if it is currently running
-    if (status() != BT::NodeStatus::RUNNING) {
+    if (status() != BT::NodeStatus::RUNNING || !goal_handle_) {
       return false;
     }
@@ -266,7 +268,7 @@ protected:
         // TODO(#1652): a work around until rcl_action interface is updated
         // if goal ids are not matched, the older goal call this callback so ignore the result
         // if matched, it must be processed (including aborted)
-        if (this->goal_handle_->get_goal_id() == result.goal_id) {
+        if (this->goal_handle_ && (this->goal_handle_->get_goal_id() == result.goal_id)) {
           goal_result_available_ = true;
           result_ = result;
         }
```

This is my workaroud.

SteveMacenski, 2021-04-08:
That sounds reasonable to add checks if even valid - a PR would be appreciated.  I will need to look at this deeper once a PR is submitted if any of these conditional statements now in an `if` require any further work. Its possible that by not setting some of the flags when its `null` is creating other issues

KavenYau, 2021-04-10:
Is there a possibility of SIGSEGV occurs in other thread? Maybe I was wrong, I just kink of curious.

Some opinions might not related to this issue here.
The `goal_handle_` would be `nullptr` if it was running in multi-threads. Because so far I don't think `bt_action_node.hpp` is s thread safe. 

If this was the first time ticking. `status` had been set to `BT::NodeStatus::RUNNING`, but had not sent a goal request yet, or had sent a goal request but had not set `goal_handle_` yet. Then another thread call `halt()`. It would be an issue.
But the bt tree of navigation2 is running in one thread. https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L220. 

For solving multi-threads issue, i think just checking if `goal_handle_` is `nullptr` might be infeasible. Some case that one thread believe a goal was cancelled, but another thread continued to send the goal request. It was a strange behavior.

tzskp1, 2021-04-10:
Yes there is.
In strictly, I did not test on a minimum environment.
I just observed that the patch suppress SIGSEGV.
I don't have the environment now, so could you please wait until the end of the week?

SteveMacenski, 2021-04-12:
Sounds good, yes, please test before we can merge your PR

SteveMacenski, 2021-05-29:
merged and closing

closed 2021-05-29 by SteveMacenski