ros-navigation/navigation2 #2318: What is the topic of querying whether the state of the target point has been reached in ros2 navigation?

miku54, 2021-04-21:
## Bug report

**Required Info:**

- Operating System:
ubuntu20.04
- ROS2 Version:
foxy

#### Steps to reproduce issue
Hello, I would like to ask which topic is the topic of querying whether the target point status has been reached in ros2 navigation? 
in nav2_controller.cpp it send the signal is "Reached the go!"  , the navigation of ros1 gets the status of reaching the target point through / move_base/result topic, does ros2 also have this?

SteveMacenski, 2021-04-21:
That was the result action notification, nav2 is also based on the action server in ROS. See the waypoint follower for an example of how to use the action server interface in an application. You shouldn’t directly interact with the action server’s internal topics for a reliable workflow

miku54, 2021-04-21:
But I want to do more cruises and subscribe to the topic of result status.

SteveMacenski, 2021-04-21:
Thats not how you should be using the action server for reliable navigation / software development. 

Closing as answered

closed 2021-04-21 by SteveMacenski