ros-navigation/navigation2 #2323: Can't build nav2_bt_navigator

pductruong, 2021-04-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- ROS2 Version:
  - ROS2 Foxy (build from source)

I follow this instruction from [https://navigation.ros.org/build_instructions/index.html](url) for build from source nav2

Here is how it fails:

```
Starting >>> nav2_planner
--- stderr: nav2_bt_navigator                                                 
/home/tucuman/nav2_ws/install/nav2_behavior_tree/lib/libnav2_behavior_tree.so: undefined reference to `BT::PublisherZMQ::PublisherZMQ(BT::Tree const&, unsigned int, unsigned int, unsigned int)'
collect2: error: ld returned 1 exit status
make[2]: *** [bt_navigator] Error 1
make[1]: *** [CMakeFiles/bt_navigator.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< nav2_bt_navigator [13.4s, exited with code 2]

```
i did complie behaviortree_cpp_v3 from [https://github.com/BehaviorTree/BehaviorTree.CPP](url)

Any help is appreciated!

EtienneSchmitz, 2021-04-23:
Which branch do you try to compile ?

I test it on ROS 2 foxy build from sources, and it works for me (on branch foxy-devel)
Do you follow this [instructions](https://navigation.ros.org/build_instructions/index.html#build-nav2-for-released-distribution) ? (rosdistro is set to foxy for you)

pductruong, 2021-04-23:
> Which branch do you try to compile ?
> 
> I test it on ROS 2 foxy build from sources, and it works for me (on branch foxy-devel)
> Do you follow this [instructions](https://navigation.ros.org/build_instructions/index.html#build-nav2-for-released-distribution) ? (rosdistro is set to foxy for you)

Oh. I changed branch to foxy-devel and it worked.
Thank u

closed 2021-04-23 by pductruong