ros-navigation/navigation2 #2324: nav2_rviz_plugins foxy-devel fails to build on arm32

adamkrawczyk, 2021-04-25:
## Bug report

- Operating System:
  - ubuntu 20.04 arm32
- ROS2 Version:
  - Foxy source
- Version or commit hash:
  - 2590238de036513c96c6fd2c76884c7599b72a01
- DDS implementation:
  - FastDDS

#### Steps to reproduce the issue
```
1. Build ros2 in a separate workspace 
2. Build nav2 in another workspace
```

#### Expected behavior
nav2 compiles

#### Actual behavior
nav2_rviz_plugins fails to compile

#### Additional information                                                                                                  

Build with command `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --cmake-args -DBUILD_TESTING=OFF --merge-install`

--- stderr: nav2_rviz_plugins                                                                                                                                  
make[2]: *** No rule to make target '/home/user/ros2_foxy/install/rviz_ogre_vendor/opt/rviz_ogre_vendor/lib/libOgreOverlay.so', needed by 'libnav2_rviz_plugins.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/nav2_rviz_plugins.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< nav2_rviz_plugins [1.18s, exited with code 2]

Another sub-pkg-s compiles fine

```
Starting >>> nav2_map_server
Finished <<< nav2_lifecycle_manager [1.33s]                                                                                                            
Starting >>> nav2_amcl
Finished <<< nav2_behavior_tree [4.17s]                                                                                                                   
Starting >>> nav2_waypoint_follower
Finished <<< nav2_map_server [1.73s]                                                                                                                              
Starting >>> nav2_costmap_2d
Finished <<< nav2_waypoint_follower [1.32s]                                                                                                               
Starting >>> nav2_bt_navigator
Finished <<< nav2_amcl [1.98s]                                                                                                                                    
Starting >>> nav2_rviz_plugins
Finished <<< nav2_costmap_2d [2.63s]       
```

Best.

SteveMacenski, 2021-04-25:
This is pretty clear from the error you’re missing a dependency of ogre vendor for your platform. Please look into this or ask on ROS Answers

closed 2021-04-25 by SteveMacenski

adamkrawczyk, 2021-04-26:
Yeah, I thought so, but installing didn't change anything tried installing/reinstalling `libogre 1.12/1.9/2.1`.