ros-navigation/navigation2 #2326: Transform data too old when converting from odom to map

nedo99, 2021-04-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - `foxy-devel` `2590238de036513c96c6fd2c76884c7599b72a01`

#### Steps to reproduce issue

```
TURTLEBOT3_MODEL=waffle ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
ros2 launch slam_toolbox online_async_launch.py
ros2 launch nav2_bringup navigation_launch.py
rviz2
```

#### Expected behavior
I am able to set as many 2DNavGoals from `rviz2` as I want, even if I set the goal on an unreachable location. I would expect even after setting a goal to an unreachable location to get navigation failed, and afterward, when setting a reachable goal, navigation works as usual.

#### Actual behavior
After setting >5-6 goals (not at once, one after another one, after it reaches the goal), navigation goes to recovery mode and does spin, `wait running`, until it aborts. After that, it is not possible to reach even the near goals. It outputs this message most of the time:
```
[controller_server-6] [INFO] [1619168751.603520587] [controller_server]: Passing new path to controller.
[controller_server-6] [ERROR] [1619168751.603785501] [tf_help]: Transform data too old when converting from odom to map
[controller_server-6] [ERROR] [1619168751.603807308] [tf_help]: Data time: 233s 104000000ns, Transform time: 232s 56000000ns
[controller_server-6] [ERROR] [1619168751.603859730] [controller_server]: Unable to transform robot pose into global plan's frame
[controller_server-6] [WARN] [1619168751.603930573] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[controller_server-6] [INFO] [1619168751.618962554] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-7] [INFO] [1619168751.619290879] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[recoveries_server-8] [INFO] [1619507274.773342970] [recoveries_server]: Attempting spin
[recoveries_server-8] [INFO] [1619507274.778353503] [recoveries_server]: Turning 1.57 for spin recovery.
[recoveries_server-8] [INFO] [1619507275.780243732] [recoveries_server]: spin running...
```
and it goes to spin and waiting after this.

I increased the `transform_tolerance` to 0.7, but does not matter of the value the message is still there.

#### Additional information
I am aware of this issue [https://github.com/ros2/ros2/issues/1074](url) and this one [https://github.com/ros-planning/navigation2/issues/1652](url), and that is why added this patch [https://github.com/ros-planning/navigation2/compare/main...KavenYau:temporarily-fix-bt-action-node-endless-loop](url) and part of this one [https://github.com/ros-planning/navigation2/commit/39ccd457653406bf31f0d59e8084f125ed72f04b](url) (only `nav2_behavior_tree/src/behavior_tree_engine.cpp` part of the patch, since `nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp` is not available in `foxy-devel`).

Should I reduce controller frequency? I checked system performance and it is constant all the time, so lack of resources should not cause this issue. Did I apply the wrong patches for mentioned issues?

Looking forward to your replies.

Regards,
Nedim

SteveMacenski, 2021-04-27:
None of those links are relevant, you have a timing issue in your system and TF is not synchronized well. This is an issue I nor anyone can really help with very much. Your best bet is to start with a working known configuration (like the nav2_bringup turtlebot simulation launch files) and change things 1 thing at a time until you find where your issue lies. 

I don't think this is likely an issue in nav2 at all, I think it might just be an issue with settings / configurations / etc.

nedo99, 2021-04-27:
OK, but if there is a config issue, shouldn’t be reported right away and not after setting many goals?

SteveMacenski, 2021-04-27:
I can't answer you that, any TF timing issues are so user specific that I can't really help. If you launch our launch file and things work, then its something in your config

nedo99, 2021-04-28:
OK. Thanks!

closed 2021-04-28 by nedo99

a-strong-python, 2025-10-31:
> ## Bug report
> **Required Info:**
>
> * Operating System:
>
>   * Ubuntu 20.04
> * ROS2 Version:
>
>   * Foxy
> * Version or commit hash:
>
>   * `foxy-devel` `2590238de036513c96c6fd2c76884c7599b72a01`
>
> #### [...]

Have you solved this issue? I'm encountering the same problem. My map->odom->base_link frequency is all above 100Hz, but this issue still occurs.