ros-navigation/navigation2 #2336: python controller

mkolodziejczyk-piap, 2021-05-12:
Hi, I've got some experimental controllers written in PyTorch and JAX. I'd like to plug them as nav2 controllers. I haven't found any working example of plugging python code into nav2. What I did so far is I created a dummy controller which is basicly copy-paste of teb_local_planner core functionality and in the computeVelocityCommands() I'm executing a service call to my python node. The problems that I encountered:

1. It's not very clean solution as I have to run independently of nav2 stack my python node (the whole lifecycle concept isn't working)
2. After I execute service client async_send_request() I can't use spin_until_future_complete(). The only option I've found is to add callback similart to this example https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/services/add_two_ints_client_async.cpp, but this causes a problem for lifecyclenode (?) because for example I can't stop it executing with Ctrl+C

Any ideas how can I improve this python wrapping?

Regards,

soldierofhell, 2021-05-12:
Some links to ROS Answers to show I'm not the only one interested:
https://answers.ros.org/question/356655/advice-on-nav2-controller-plugin/
https://answers.ros.org/question/357237/writing-a-planner-plugin-for-ros2-using-python/

SteveMacenski, 2021-05-12:
There is no intention now or in the foreseeable future to support python controller algorithms. One of the benefits of the distributed action server method of Nav2 allows you to replace the controller server outright and replace with a fully python version if you like, but unless pluginlib starts supporting python, our hands are tied

SteveMacenski, 2021-05-13:
For reference of why python is probably not a good idea for a controller especially, I'm prototyping some path smoothers using gradient descent. The python version looks something like

```
    start_time = time.time()

    its = 0
    max_its = 1000
    tolerance = 1e-10
    new = [list(a) for a in zip(way_point_x, way_point_y)]
    path = list(zip(way_point_x_org, way_point_y_org))
    dims = 2
    change = tolerance
    data_w = 0.20 # 0.25
    smooth_w = 0.30 # 0.25 # 0.15-0.5
    
    while change >= tolerance:

        its = its + 1
        if its > max_its:
            print('Number of iterations has exceeded limit of ' + str(max_its) + '. Terminating.')
            break

        change = 0.0
        for i in range(1, len(new) - 1):
            for j in range(dims):
                x_i = path[i][j]
                y_i, y_prev, y_next = new[i][j], new[i - 1][j], new[i + 1][j]
                y_i_org =  y_i 

                if i > 2 and i < len(new) - 2:
                    # Smooth based on local 5 point neighborhood and original data locations
                    y_im2, y_ip2 = new[i - 2][j], new[i + 2][j]
                    y_i += data_w * (x_i - y_i) + smooth_w * (y_im2 + y_ip2 + y_next + y_prev - (4 * y_i))
                else:
                    # Smooth based on local 3 point neighborhood and original data locations
                    y_i += data_w * (x_i - y_i) + smooth_w * (y_next + y_prev - (2 * y_i))

                new[i][j] = y_i

                change += abs(y_i - y_i_org)

    new_x = [x[0] for x in new]
    new_y = [x[1] for x in new]

    # show results
    elapsed_time = (time.time() - start_time) * 1000
```

and the C++ version is 

```
    unsigned int its = 0;
    double data_w = 0.20;
    double smooth_w = 0.30;
    double change = tolerance_;
    unsigned int path_size = path.poses.size();

    double x_i, y_i, y_prev, y_next, y_i_org, y_im2, y_ip2;

    nav_msgs::msg::Path new_path = path;

    while (change >= tolerance_) {
      its += 1;
      change = 0.0;

      if (its >= max_its_) {
        RCLCPP_ERROR(
          rclcpp::get_logger("SmacPlannerSmoother"),
          "Number of iterations has exceeded limit of %i. Returning original path.", max_its_);
        return false;
      }

      if (dt >= max_time) {
        RCLCPP_ERROR(
          rclcpp::get_logger("SmacPlannerSmoother"),
          "Smoothing time exceeded allowed duration of %0.2f. Returning original path.", max_time);
        return false;
      }

      for (unsigned int i = 1; i != path_size - 1; i++) {
        for (unsigned int j = 0; j != 2; j++) {
          if (j == 0) {
            x_i = path.poses[i].pose.position.x;
            y_i = new_path.poses[i].pose.position.x;
            y_prev = new_path.poses[i - 1].pose.position.x;
            y_next = new_path.poses[i + 1].pose.position.x;
          } else {
            x_i = path.poses[i].pose.position.y;
            y_i = new_path.poses[i].pose.position.y;
            y_prev = new_path.poses[i - 1].pose.position.y;
            y_next = new_path.poses[i + 1].pose.position.y;
          }

          y_i_org =  y_i;

          if (i > 2 && i < path_size - 2) {
            // Smooth based on local 5 point neighborhood and original data locations
            if (j == 0) {
              y_im2 = new_path.poses[i - 2].pose.position.x;
              y_ip2 = new_path.poses[i + 2].pose.position.x;
            } else {
              y_im2 = new_path.poses[i - 2].pose.position.y;
              y_ip2 = new_path.poses[i + 2].pose.position.y;
            }

            y_i += data_w * (x_i - y_i) + smooth_w * (y_im2 + y_ip2 + y_next + y_prev - (4.0 * y_i));
          } else {
            // Smooth based on local 3 point neighborhood and original data locations
            y_i += data_w * (x_i - y_i) + smooth_w * (y_next + y_prev - (2.0 * y_i));  
          }

          if (j == 0) {
            new_path.poses[i].pose.position.x = y_i;
          } else {
            new_path.poses[i].pose.position.y = y_i;            
          }

          change += abs(y_i - y_i_org);
        }
      }
    }

```

The python version takes 77ms to compute what takes C++ 0.45ms which is **171x** faster. Since a controller runs very quickly, and many controller methods are based on gradient descent in one way or another, I think its reasonable to conclude its far too slow to rely upon. And this is for the absolute bare-bones basics!

mkolodziejczyk-piap, 2021-05-13:
Ok, but your python code is pure python, it's not even "vectorized" with numpy. By python I mean python wrappers to C++, CUDA, XLA "kernels". Check for example [JAX](https://jax.readthedocs.io/en/latest/index.html). I can rewrite your snippet in JAX and we can compare (also jitted or gpu versions).

SteveMacenski, 2021-05-13:
Again, this is more purely a case of issue with language support. Unless this python controller is being considered to be open-sourced, I can't easily justify engineering resources to make pluginlib language agnostic or try to create a python version of the planner and controller servers with pybind costmap support.

mikeferguson, 2021-05-13:
In theory, an end user should be able to write a c++ "planner" or "controller" plugin that uses Boost::Python to embed their Python functions (c++ calling python, rather than the more typical Python calling C++ - see https://www.boost.org/doc/libs/1_62_0/libs/python/doc/html/tutorial/tutorial/embedding.html). There would be a decent amount of work to translate the costmap, goals, and returned plan between C++ <-> Python, but it wouldn't require making a more generic python version of pluginlib or rewriting the planner or controller servers.

SteveMacenski, 2021-05-13:
I recall conversations with Shah when I was at Simbe around pluginlib and python / java. He mentioned to me that the same libraries that is based off of could also be manipulated to support other languages (with some effort). At the time, I remember looking a little into it and finding that it would be possible. I suppose I think of that direction being the most "general" such that we don't have to add boost or pybind APIs for specific packages that could change. Instead, doing it at the plugin level would essentially "autogenerate" that support which I viewed as lower friction than trying to make an entirely new controller/planner server in python and pybind costmap APIs to it.

Though, thinking about it another way, it perhaps might not be too much of a challenge (?) to have a C++ plugin shim for python plugins. For example, creating a `python_shim_controller` who's goal is just to be a plugin to shim between C++ and python and translate the inputs to the python version. It sounds do-able, but it would probably at best be a code template someone would have to fill in themselves with their custom code rather than being dynamically run-time loadable directly.

mikeferguson, 2021-05-14:
> Either way, probably not something I would spend time on myself to support unless there was an open-source plugin contribution to Nav2 to use it.

That's exactly why I suggested this approach - it's something a 3rd party can implement (basically, it's just another 3rd party controller at that point, no changes needed to nav2).

soldierofhell, 2021-05-14:
In the meantime I've found this recent PR that helps solve problems with service calls: https://github.com/ros-planning/navigation2/pull/2216

mkolodziejczyk-piap, 2021-05-14:
In the meantime I've found this recent PR that helps solve problems with service calls: #2216

SteveMacenski, 2021-05-14:
Got it - closing ticket. I'm not sure there's any action to be taken here at the moment.

closed 2021-05-14 by SteveMacenski