ros-navigation/navigation2 #2337: Move CI, developer instructions, and workflows from ROS2 master to Rolling

SteveMacenski, 2021-05-14:
- [ ] change dockerhub base image to rolling
- [ ] change CI to use rolling
- [ ] Have all dependencies in Rolling (including dependencies workspace) to allow for 1-stop-show distribution for Nav2 development without need for master ROS2 or underlay workspaces
- [ ] Update build / install instructions to use Rolling
- [ ] Update developer workflows to use Rolling and no longer master ROS2 builds + underlay/dependencies workspace
- [ ] Consider releasing Nav2 to rolling itself 

CC @ruffsl

SteveMacenski, 2021-08-05:
@ruffsl would I be correct in saying CI / all the docker images use rolling now? If so, I can update all of the documentation in Nav2 website to point to Rolling instead and then close this ticket out (maybe delete some of the tools in `tools`)

SteveMacenski, 2021-08-05:
Did a quick review, found https://navigation.ros.org/build_instructions/index.html as the only page going into install methods to master. I'll update this and control+F on the rest of it to update any other references for master.

ruffsl, 2021-08-06:
> would I be correct in saying CI / all the docker images use rolling now?

Yep, evening should be using the rolling release now.

SteveMacenski, 2021-08-24:
Done!

closed 2021-08-24 by SteveMacenski