ros-navigation/navigation2 #2340: BT Navigator got stuck because goal_status of bt_action_node was always STATUS_SUCCEEDED

KavenYau, 2021-05-15:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 20.04
- ROS2 Version:
  - Rolling Source
- Version or commit hash:
  - 8947c9fdca621624fddc49ff40af9d3f29693e81
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue

Keeping the `nav2` being long time running and navigating.

#### Expected behavior
Do not get stuck any time.

#### Additional information
```
[bt_navigator-7] [INFO] [1621000214.297475047] [bt_navigator]: Begin navigating from current location through 1 poses to (-1.10, 1.00)
[bt_navigator-7] [WARN] [1621000214.318210364] [bt_navigator_rclcpp_node]: compute_path_through_poses result_callback, goal_id: 45a153b81a2834a7ce487b7b9ae3d7
[bt_navigator-7] [WARN] [1621000214.318305716] [bt_navigator_rclcpp_node]: compute_path_through_poses this goal received result
[bt_navigator-7] [WARN] [1621000214.318474461] [bt_navigator_rclcpp_node]: compute_path_through_poses goal_handle_.reset() return status
[controller_server-4] [INFO] [1621000214.320552559] [controller_server]: Received a goal, begin computing control effort.
[bt_navigator-7] [WARN] [1621000217.338530423] [bt_navigator_rclcpp_node]: compute_path_through_poses result_callback, goal_id: 19e1b55ebf2d77f7ac65cada931a9af1
[bt_navigator-7] [WARN] [1621000217.338595321] [bt_navigator_rclcpp_node]: compute_path_through_poses this goal received result
[bt_navigator-7] [WARN] [1621000217.347825943] [bt_navigator_rclcpp_node]: compute_path_through_poses goal_handle_.reset() return status
[controller_server-4] [INFO] [1621000217.421377821] [controller_server]: Passing new path to controller.
[bt_navigator-7] [WARN] [1621000217.438086491] [bt_navigator_rclcpp_node]: follow_path result_callback, goal_id: 1c6db19114d5539f29a8b62218206733
[bt_navigator-7] [WARN] [1621000217.438143158] [bt_navigator_rclcpp_node]: follow_path previous goal received result
[controller_server-4] [INFO] [1621000219.871159409] [controller_server]: Reached the goal!
[bt_navigator-7] [WARN] [1621000219.898085410] [bt_navigator_rclcpp_node]: follow_path result_callback, goal_id: 817daec351be1eeb6f29c5a468a4674
[bt_navigator-7] [WARN] [1621000219.898160405] [bt_navigator_rclcpp_node]: follow_path this goal received result
[bt_navigator-7] [WARN] [1621000219.898279745] [bt_navigator_rclcpp_node]: BT::NodeStatus on_success follow_path
[bt_navigator-7] [WARN] [1621000219.898306641] [bt_navigator_rclcpp_node]: follow_path goal_handle_.reset() return status
[bt_navigator-7] [INFO] [1621000219.907802216] [bt_navigator]: Goal succeeded
[bt_navigator-7] [INFO] [1621000220.056756493] [bt_navigator]: Begin navigating from current location through 1 poses to (-2.00, 1.00)
```
variable value：
```
[Switching to Thread 0x7fafc17fa700 (LWP 817440)]
Thread 17 "bt_navigator" hit Breakpoint 1, nav2_behavior_tree::BtActionNode<nav2_msgs::action::ComputePathThroughPoses>::tick (this=0x56235ad745c0)
    at /home/kaven/code/ros2_master/nav2_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp:173
173     if (status() == BT::NodeStatus::IDLE) {
(gdb) p action_name_
$1 = "compute_path_through_poses"
(gdb) n
185     if (future_goal_handle_) {
(gdb) p future_goal_handle_
$2 = std::shared_ptr<std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::ComputePathThroughPoses> > >> (empty) = {get() = 0x0}
(gdb) n
203     if (rclcpp::ok() && !goal_result_available_) {
(gdb) n
nav2_behavior_tree::BtActionNode<nav2_msgs::action::ComputePathThroughPoses>::tick (
    this=0x56235ad745c0)
    at /home/kaven/code/ros2_master/nav2_ws/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp:203
203     if (rclcpp::ok() && !goal_result_available_) {
(gdb) p goal_result_available_
$3 = false
(gdb) n
205       on_wait_for_result();
(gdb) n
207       auto goal_status = goal_handle_->get_status();
(gdb) n
208       if (goal_updated_ && (goal_status == action_msgs::msg::GoalStatus::STATUS_EXECUTING ||
(gdb) p goal_updated_
$4 = false
(gdb) p goal_status
$5 = 4 '\004'
(gdb) 
```

Actually I observed that this issue had appeared in `foxy`, it was not a new issue caused by 8947c9fdca621624fddc49ff40af9d3f29693e81. 

I wonder know that if you observed this issue before?

SteveMacenski, 2021-05-15:
I don't understand, does this relate to #2303? I don't recognize those logs, is this a different branch than what's in nav2? 

Please start from fundamentals and describe the issue

KavenYau, 2021-05-15:
>  does this relate to #2303.

So far I don't think so, It seems that `compute_path_through_poses ` 's goal was succeeded, but `result_callback` not called, then the bt node would always return `RUNNING`, and it would never get a path forever.

>  I don't recognize those logs, is this a different branch than what's in nav2?

It's the branch `main` of `nav2`, I only added some logs to show steps,
Logs after removing the additional logs which I added: 
```
[bt_navigator-7] [INFO] [1621000214.297475047] [bt_navigator]: Begin navigating from current location through 1 poses to (-1.10, 1.00)
[controller_server-4] [INFO] [1621000214.320552559] [controller_server]: Received a goal, begin computing control effort.
[controller_server-4] [INFO] [1621000217.421377821] [controller_server]: Passing new path to controller.
[controller_server-4] [INFO] [1621000219.871159409] [controller_server]: Reached the goal!
[bt_navigator-7] [INFO] [1621000219.907802216] [bt_navigator]: Goal succeeded
[bt_navigator-7] [INFO] [1621000220.056756493] [bt_navigator]: Begin navigating from current location through 1 poses to (-2.00, 1.00)
```

SteveMacenski, 2021-05-18:
> So far I don't think so, It seems that compute_path_through_poses 's goal was succeeded, but result_callback not called, then the bt node would always return RUNNING, and it would never get a path forever.

Huh, ok. That context helps, but I think that code snippet / GDB session doesn't show the right thing. The `goal_updated_` and goal status markers are only for preempting the existing action with a new one. So if we're currently executing something and we have a new goal, that part of the code is saying to issue the new goal. 

I think what would be good to look at is `goal_result_available_` and `result_` to see if that's been updated (and/or print a log in the result callback to see if something is up in there). 

https://github.com/ros-planning/navigation2/pull/2334 changes each of the Bt nodes to use its own executor. It might be worth trying that branch to see if this issue still persists. If its an odd internal networking quirk, this might solve it implicitly. 

---

If the goal status was success there, that means either:
-  the result callback states weren't updated properly (mutex locking issue?)
-  the callback itself wasn't actually called
- the new goal overrided the `goal_handle_->get_goal_id()` too early so the new goal's `goal_handle_` replaced it causing the result callback's goal ID to not match the current goal handles. 

which seems like from the other logging is the direction you were looking already. Given I don't know where you put those log prints or the context of your experiments, can you shed light on which of these 3 you found might or aren't the issue?

KavenYau, 2021-05-18:
> I think what would be good to look at is goal_result_available_ and result_ to see if that's been updated (and/or print a log in the result callback to see if something is up in there).

That GDB session shows that `goal_result_available_ ` was `false`. and `future_goal_handle_` was not `nullptr`.

>  the callback itself wasn't actually called.

According to my experiments, the `result_callback` was not called while the issue appeared. Again, it's a race condition issue of `rclcpp_action` 😦 . I have made a PR for it. https://github.com/ros2/rclcpp/pull/1677

I think #2334 is nice, it might solve some issues implicitly. 👍

SteveMacenski, 2021-05-18:
OK, sounds like another `rclcpp_action` issue then :( Lets just make sure that your solution is also backported to Foxy and Galactic (did your other PRs get those as well?) so its not an issue for production users like yourself.

KavenYau, 2021-05-21:
Closing this issue since ros2/rclcpp#1677 merged.

closed 2021-05-21 by KavenYau

SteveMacenski, 2021-05-21:
Thanks for finding and fixing yet another rclcpp_actions locking issue!