ros-navigation/navigation2 #2341: Costmap2DPublisher  lock costmap mutex twice, is it Efficient for its update

hoppss, 2021-05-17:
this is not a bug, but one of my thinking. I want to know how to you think about it. pls ignore my bug issue, i will close it after some days. 

In Costmap2dPublisher class, publishCostmap()  function will publish raw costmap msg and update msg, it will lock twice  costmap2dRos object mutex to obatin data, is it efficient ?  

 I think it mignt be good to lock mutex once to otain data, and convert to costmap msg or occupancy data.
whether it is more efficient to publish msg in Costmap2dRos update thread ?
whether it is more efficient to make copy data in Costmap2dRos for other object?

SteveMacenski, 2021-05-17:
I don't see an issue with what's around right now, but I'd be open to a PR if you feel there's a way to improve it

hoppss, 2021-05-20:
there are some condition , costmap2dros object's mutex will be locked;

1.

closed 2021-05-20 by hoppss