ros-navigation/navigation2 #2345: Robot bumps into obstacles

saratpoluri, 2021-05-17:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- ROS2 Version:
  Foxy, installed using apt
- Version or commit hash:
  0.4.7-1focal.20210423.031543
- DDS implementation:
  Fast-RTPS
#### Steps to reproduce issue
This is my param file:
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

amcl_map_client:
  ros__parameters:
    use_sim_time: True

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: True

bt_navigator:
  ros__parameters:
    use_sim_time: True
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    default_bt_xml_filename: "navigate_w_replanning_and_recovery.xml"
    plugin_lib_names:
[... 317 lines cut ...]
    ld.add_action(declare_slam_cmd)
    ld.add_action(declare_use_visual_slam_cmd)
    
    ld.add_action(declare_map_yaml_cmd)
    ld.add_action(declare_use_sim_time_cmd)
    ld.add_action(declare_params_file_cmd)
    ld.add_action(declare_bt_xml_cmd)
    ld.add_action(declare_autostart_cmd)

    # Add the actions to launch all of the navigation nodes
    ld.add_action(prokit_cmd)
    ld.add_action(bringup_cmd)

    return ld

#### Expected behavior
The robot avoids obstacles and navigates to goal pose.

#### Actual behavior
Robot avoids obstacles and navigates to goal pose

The navigation package is planning a path that is violation of the ObstacleFootprintCritic. 
[image]

If you look at the image, it sees the obstacles, marks the occupancy grid, it knows what the robots footprint is and yet the planned path has the center of the robot passing through inflation radius and the boundaries of the robot ramming into the obstacle identified on the costmap.

SteveMacenski, 2021-05-17:
Please ask for configuration assistance on ROS Answers (also be sure to properly format the question and code segments).

closed 2021-05-17 by SteveMacenski

saratpoluri, 2021-05-17:
I don't think it is an issue with configuration. It looks like a bug in navigation2.

How is possible for robot's footprint boundaries to be touching the occupied grid in the costmap? That is something that shouldn't be happening.

SteveMacenski, 2021-05-17:
Issues like that are almost exclusively caused by misconfiguration, especially of the controller. Occam's razor states that these kinds of issues are far more likely to be caused by those issues than actual issues in the software which haven't been reported by anyone else. In my experience and from the information provided, this isn't likely a bug in the software, but an issue with your setup. I'd be happy to reverse course if evidence was provided that made me think there was an issue here to fix. 

For configuration help, I suggest reading the navigation.ros.org configuration sections and search ROS Answers for other questions on this same topic.

saratpoluri, 2021-05-18:
I have thoroughly went through all the settings for the controller_server and could not identify an issue. Obviously, I am missing something. I have, however, verified that the issue is indeed with the controller_server because the smac_planner does the path planning correctly. The motion planner is just not able to follow the trajectory. 

I see from your announcement post [here](https://discourse.ros.org/t/nav2-smacplanner-hybrid-a-2d-a-now-available-reminder-meeting-oct-15-cancelled/16759) that someone has verified smac_planner on ackermann drive robots. 

I would really appreciate if any one of the param files could be shared. There are no sample parameter files available online for ackermann drive robots or for that matter any drive other than for turtlebot.