ros-navigation/navigation2 #2358: map_saver_cli arguments are not forwarded

charlielito, 2021-05-21:
## Bug report

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - 2e18dd1cae31807f68746eb1a2c702526a34f4cc
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
```
ros2 run nav2_map_server map_saver_cli -f test_map -t test_topic --free 0.2 --occ 0.7
```

The logs are:

```
[INFO][1621641458.320967395][map_saver]: 
        map_saver lifecycle node launched. 
        Waiting on external lifecycle transitions to activate
        See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO][1621641458.321036351][map_saver]: Creating
[INFO][1621641458.321142243][map_saver]: Saving map from 'test_topic' topic to 'test_map' file
[WARN][1621641458.321156177][map_saver]: Free threshold unspecified. Setting it to default value: 0.250000
[WARN][1621641458.321164777][map_saver]: Occupied threshold unspecified. Setting it to default value: 0.650000
```

The usage is:
```
Usage:
  map_saver_cli [arguments] [--ros-args ROS remapping args]

Arguments:
  -h/--help
  -t <map_topic>
  -f <mapname>
  --occ <threshold_occupied>
  --free <threshold_free>
  --fmt <image_format>
  --mode trinary(default)/scale/raw

NOTE: --ros-args should be passed at the end of command line
```

#### Expected behavior
The generated yaml file should have `occupied_thresh: 0.7` and `free_thresh: 0.2`.

#### Actual behavior
As the logs says, the default values are used instead, i.e.  `occupied_thresh: 0.65` and `free_thresh: 0.25`

SteveMacenski, 2021-05-24:
A PR would be appreciated

charlielito, 2021-05-24:
> A PR would be appreciated, I see where the param is read in
> 
> https://github.com/ros-planning/navigation2/blob/30b405c58e6d53ba8c96381416bc4679d35a1483/nav2_map_server/src/map_saver/map_saver.cpp#L53
> 
> both as a node and as an input
> https://github.com/ros-planning/navigation2/blob/30b405c58e6d53ba8c96381416bc4679d35a1483/nav2_map_server/src/map_saver/main_cli.cpp#L114
> 
> Is the input one being overridden by the declare?

The error was the use of `atoi` function to parse a floating number, that returns 0 in that case. Just submitted a PR

PR #2362 (2021-05-24): Fix map_saver_cli arguments not parsed well
Fixes #2358

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | #2358  |
| Primary OS tested on | Ubuntu 20.04 |
| Robotic platform tested on | NA |

---

## Description of contribution in a few bullet points

* Fixes `map_saver_cli` when parsing `--free` and `--occ` arguments.

#### For Maintainers: 
- [ ] Check that any new parameters added are updated in navigation.ros.org
- [ ] Check that any significant change is added to the migration guide
- [ ] Check that any new functions have D [... cut]
commit 3d66437b (2021-05-24): Fix map_saver_cli arguments not parsed well (#2362)

Fixes #2358
commit 4b3391ee (2021-05-24): Galactic sync 1 : All commits before May 24 (#2363)

* reset all goal checkers on cleanup state (#2354)

* Fix map_saver_cli arguments not parsed well (#2362)

Fixes #2358

* bumping to 1.0.2 for re-release with updates
closed 2021-05-24 by SteveMacenski