ros-navigation/navigation2 #2368: Moving fixes from PR #2083 into foxy-devel

nkalupahana, 2021-05-26:
## Bug report

**Required Info:**

- Operating System:
  - macOS 11
- ROS2 Version:
  - Foxy, from source
- Version or commit hash:
  - latest on `foxy-devel`
- DDS implementation:
  - Fast-DDS

Building on macOS was fixed by PR #2083 -- but those changes haven't gotten added to the foxy-devel branch, which is the branch people building for Foxy are expected to build off of. Is there a reason those changes haven't been cherry-picked over? I'd be happy to do it otherwise.

SteveMacenski, 2021-05-26:
It was not migrated because it changed function signatures in `ObservationBuffer`. Other than that, all of the other changes could be made and I'd be happy to merge a PR into Foxy to resolve that for you as long as it doesn't modify any function signatures

SteveMacenski, 2021-06-02:
@nkalupahana any progress? Currently blocked on the unique pointer, as far as I can tell from looking the move is still required by best practices to give up the ownership rights on the unique pointer to the middleware to reduce jitter

nkalupahana, 2021-06-02:
I'm a bit tied up with work at the moment, but I should get some spare time in the next week to figure out what's going on and finish off the PR.

SteveMacenski, 2021-06-02:
OK, thanks for the update

SteveMacenski, 2021-06-08:
Merging imminent

PR #2375 (2021-06-08): Moves changes from PR #2083 to foxy
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | Closes #2368 |
| Primary OS tested on | macOS |
| Robotic platform tested on | N/A |

---

## Description of contribution in a few bullet points

- Foxy branch now builds on macOS, based on fixes from PR #2083! Proof of build on CI: https://github.com/nkalupahana/ros2-foxy-macos/runs/2698988683?check_suite_focus=true

---

#### For Maintainers: 
- [ ] Check that any new parameters added are updated in navigation.ros.org
- [ ] Check that any significa [... cut]
closed 2021-06-08 by SteveMacenski