ros-navigation/navigation2 #2372: library from newer version required on ros foxy

locmark, 2021-05-28:
## Bug report

- Operating System: Ubuntu 20.04
- ROS2 Version: ros2-foxy (installed via apt-get)
- Version or commit hash: 0.4.7-1focal.20210423.031543
#### Steps to reproduce issue
copy-paste attached archive to your workspace src directory. 
[my-nav-package.zip](https://github.com/ros-planning/navigation2/files/6561629/my-nav-package.zip)

colcon build --packages-select nav
. install/setup.bash
ros2 launch nav nav.launch.py map:=/home/k/agv/agv-ros/maps/map1.yaml map_type:=occupancy

#### Expected behavior
** works **

#### Actual behavior
[lifecycle_manager-10] [INFO] [1622211994.561657286] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-8] [INFO] [1622211994.562754576] [bt_navigator]: Configuring
[bt_navigator-8] [ERROR] [1622211994.680234193] []: Caught exception in callback for transition 10
[bt_navigator-8] [ERROR] [1622211994.680263029] []: Original error: Could not load library: libnav2_compute_path_through_poses_action_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-8] [WARN] [1622211994.680296104] []: Error occurred while doing error handling.
[bt_navigator-8] [FATAL] [1622211994.680311486] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-10] [ERROR] [1622211994.681249996] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-10] [ERROR] [1622211994.681296984] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

##### entire log below: [log.txt](https://github.com/ros-planning/navigation2/files/6561660/log.txt)

SteveMacenski, 2021-05-28:
Foxy clearly does not contain that capability https://github.com/ros-planning/navigation2/tree/foxy-devel/nav2_behavior_tree/plugins/action, did you add that BT node to your yaml files or pull files from the wrong branches? 

https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bringup/bringup/params/nav2_params.yaml foxy's default param files do not contain this either. This appears to be user error unless there's somewhere that the compute path through poses BT node was accidentally added to the foxy branch, which I do not see currently

locmark, 2021-05-29:
> oxy's default param files do not contain this either. This appears to be user error unless there's somewhere that the compute path through poses BT node was accidentally added to the foxy branch, which I do not see currently

Indeed, it is my mistake.. I used yaml config from branch main instead of foxy-devel...

closed 2021-05-29 by locmark