ros-navigation/navigation2 #2373: Controller/planner server segfault on clear local/global costmap

charlielito, 2021-05-28:
## Bug report

- Operating System:
  - Docker + Ubuntu 20.04, 10th gen i7
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  - Galactic branch: 4b3391ee01993e97cc8a566363581108d1f291da
- DDS implementation:
  - cyclone 

#### Steps to reproduce issue
I am running gazebo with a custom map and custom robot. I am able to navigate, and while doing it, sometimes randomly the controller server dies when requested to clear the local costmap. Also sometimes the planner server dies when requested to clear the global costmap. I have the same set up as in foxy, just changed a little the `nav2_params.yaml` to meet the new API. I am also using the same default BT file as in foxy branch `navigate_w_replanning_and_recovery.xml`. In foxy everything runs well and it never dies.

<details>
  <summary>nav2_params.yaml</summary>

```yaml
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.25
    alpha2: 0.2
    alpha3: 0.15
    alpha4: 0.2 
    alpha5: 0.2 # for omnidirectional robot_model_type only
    base_frame_id: "chassis"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 40.0
    laser_min_range: -1.0 # -1 laser’s reported minimum
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

amcl_map_client:
  ros__parameters:
    use_sim_time: True

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: True

bt_navigator:
  ros__parameters:
    use_sim_time: True
[... 301 lines cut ...]

waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"   
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200
```  

</details>

#### Expected behavior
Do not crash any node

#### Actual behavior
Sometimes controller server crashes, sometimes planner server

#### Additional information
Here are some backtraces when the nodes die

[controller_server_segfault.log](https://github.com/ros-planning/navigation2/files/6562375/controller_server_segfault.log)
[planner_server_segfaul.log](https://github.com/ros-planning/navigation2/files/6562376/planner_server_segfaul.log)

SteveMacenski, 2021-05-28:
I'm not sure, it would definitely be worth more investigation. Nothing has changed anytime recently that I'm aware of, but the only thing that's really changed since  the initial port from ROS1 was https://github.com/ros-planning/navigation2/commit/13f518ac0a021f65d2dc6c0dfac95f3a13467b46#diff-15e7b290bebe57765609a46208aa6f20ba2b06ac10a3237f06259265a086cacc -- so that's where I'd start investigating. 

What's your compute platform?

charlielito, 2021-05-28:
My teesting compute platform is 10th gen i7 laptop

AlexeyMerzlyakov, 2021-06-15:
In the presented above `nav2_params.yaml` file the main changes are related to voxel_layer PCL2 observation sources and custom robot parameters. From the backtraces above the problem most likely is related to `VoxelLayer` plugin. So, I've added PCL to voxel_layer' observation sources and enabled this layer for both: local and global planners. Then I've tried to reproduce the problem with:
1. Given above configuration using OSRF Galactic Docker container while running standard TB3 waffle [Gazebo simulation](https://navigation.ros.org/getting_started/index.html#running-the-example) in there.
2. Latest ROS2 Rolling and nav2 stack versions (both built from sources) while running standard and [AWS small warehouse](https://github.com/aws-robotics/aws-robomaker-small-warehouse-world) TB3 waffle Gazebo simulation in there.

In parallel infinite loop I've sent `ClearEntireCostmap` service calls to local and global planners. Unfortunately during 15~20 minutes of continuous calls in each configuration case the crash was not appeared never.

@charlielito, did you ever observe the crash on any Gazebo and TB3 simulation?
Since voxel grid is related to PCL obtained from specific source, that could be helpful if you could give some Gazebo simulation clues or record a small repro-case ROS2Bag, reproducing which one we could obtain the same PCL Grid and crash (more or less stably).

charlielito, 2021-06-16:
Hey thanks for taking the time to look into this! Indeed the crash happens with our Gazebo environment with a custom robot. It also happened in our real hardware/robot. I'll try the TB3 waffle demo to see if it is crashing there and also come back with some ros2bag for you

AlexeyMerzlyakov, 2021-06-18:
Yes, it would be really appreciated, if you could provide a context to catch the problem.

SteveMacenski, 2021-06-18:
In thread 14, it looks like the usual update map cycle: `updateMap()` -> `updateBounds()` -> `raytraceFreespace()` -> `clearVoxelLineInMap()` -> `raytraceLine()` -> `bresenham3D()` -> clearing.

Are you running SLAM at the same time? That shouldn't make anything break, just looking for clues about why it is that some data location wouldn't be available. None of this has changed since ROS1. 

Do you see this happen with the 2D obstacle layer at all (if you were willing to try that for us)? The voxel layer uses a ton of the same mechanics as the obstacle layer so if you saw it there, that would help narrow it down. Are you going "off the map" at all? What sensor are you using to get this error

charlielito, 2021-06-21:
@AlexeyMerzlyakov I am encountering this error in Gazebo but with a custom robot. At the end i didn't try the TB3 waffle because the problem seems to be when using PCL. I tested my environment without PCL sources and never saw any crashes. Just when I add a PCL source to the VoxelLayer, the problems begin.

Also tried your strategy of calling the clearing services both to local and global costmaps, but couldn't make it crash :disappointed: I am only seeing this crash whilst navigating to a goal. Here you can find a zip rosbag attached with all topics. The robot got stuck and I was setting a new goal when the planner server died in that case. 

https://drive.google.com/file/d/1j54RIVWt83MbNRpKmendjb7bOVCf7LzN/view?usp=sharing

@SteveMacenski I am not running SLAM. I am running the stack with AMCL with a static map. I also tried only using 2D obstacle layer with a PCL source but it didn't crash at all. The sensor that generates the PCL is a gazebo stereocamera.

SteveMacenski, 2021-06-21:
Huh, did you build a different PCL version or something? I dont think there’s any action we can take then. Should we close this ticket?

charlielito, 2021-06-21:
What? Just updated my comment. With PCL I meant pointcloud data, nothing to do with the PCL library.

SteveMacenski, 2021-06-21:
I wonder if this is related to nans or infs or something similar.

AlexeyMerzlyakov, 2021-06-24:
@charlielito, thank you very much for carefully prepared dataset. This did helped me: I've reproduced the same issue on `planner_server` executable. Some debugging on it shown that there is incorrect `offset` parameter passed _inside_ `bresenham3D()` inline routine. Moreover, _outside_ `bresenham3D()` (in `raytraceLine()` routine) it is being set correctly. Also, simple calculations from full backtrace you've attached here (`planner_server_segfaul.log`) confirms the situation: `offset` is equal to `4294966748` but it should be `11278` as simply calculated by `x0,x1,y0,y1,min_len,offset_dy`.

This might be stack corruption, GCC bug or something I have no idea yet (e.g. C++ standard deprecation when passing by reference into inline function). Anyway, it looks like the following patch cured the problem for me:
```
diff --git a/nav2_voxel_grid/include/nav2_voxel_grid/voxel_grid.hpp b/nav2_voxel_grid/include/nav2_voxel_grid/voxel_grid.hpp
index de3e3071..42efda7b 100644
--- a/nav2_voxel_grid/include/nav2_voxel_grid/voxel_grid.hpp
+++ b/nav2_voxel_grid/include/nav2_voxel_grid/voxel_grid.hpp
@@ -300,7 +318,7 @@ private:
   inline void bresenham3D(
     ActionType at, OffA off_a, OffB off_b, OffC off_c,
     unsigned int abs_da, unsigned int abs_db, unsigned int abs_dc,
-    int error_b, int error_c, int offset_a, int offset_b, int offset_c, unsigned int & offset,
+    int error_b, int error_c, int offset_a, int offset_b, int offset_c, unsigned int offset,
     unsigned int & z_mask, unsigned int max_length = UINT_MAX)
   {
     unsigned int end = std::min(max_length, abs_da);
```

After one week of break (I'll be on vacation) I will return to this problem and try to figure out more detailed explanation of what it happens here.

SteveMacenski, 2021-06-28:
That was a brand new feature added recently to have the minimum / maximum raytracing and marking ranges. So I don't think its out of the question that its a legit bug. I'd make sure to check the `unsigned` versus signed `int`s, if you don't explicitly cast between them, I've seen that be a problem before. Given the signcature of that fuction is all `int` and only that 1 is an `unsigned int`, that makes me immediately suspicious

AlexeyMerzlyakov, 2021-07-10:
It looks like I've finally found what is happening in `bresenham3D()`. My initial workaround about fixing offset was wrong. The reference for `offset` in `bresenham3D()` signature is made _intentionally_ for use dynamically changing nature of `offset` value. The essence is in the `OffA off_a, OffB off_b, OffC off_c` objects which are of `GridOffset` type and having the reference to the same `offset` inside. By calling `operator(offset_val)` for `GridOffset` objects we are shifting the `offset` to a given in operator value.

More debugging shown that we are running out of the space of VoxelGrid `data_` array (which is equal to `size_x_ * size_y_`). The exceeding of array boundaries might be as in positive direction, but as in negative direction falling behind `0`: `-1` will be treated as `4billons-millons...` of unsigned int type. The direction depends on step (`dx/dy/dz`) sign. Both of directions cases were observed during the debug.

So, regarding the root of the problem, it should be related to this code in `raytraceLine()` routine:
```
    double min_x0 = x0 + dx / dist * min_length;
    double min_y0 = y0 + dy / dist * min_length;
    double min_z0 = z0 + dz / dist * min_length;
```
When we setting minimal `x0/y0` not to correspond to VixelGrid array boundaries, but somewhere in the middle of it, we finally will run out of its boundary in the end as shown in the picture below (sorry, I am not a good artist but hope you will get the situation :smile: ):
[image]

The bug was introduced with 13f518ac0a021f65d2dc6c0dfac95f3a13467b46 as @SteveMacenski initially mentioned, that is why it did not observed on Foxy. This need more technical analysis about how it should be correctly fixed. One of the option - is to add boundary checks to `bresenham3D()` when changing the `offset`. Another option - is to fix array boundaries iterators on `min_length` value. I'll try to figure out the right solution a few days later.

AlexeyMerzlyakov, 2021-07-19:
Update: The out of array bounds accessing situations were appeared due to incorrectly calculated `min_x0`/`min_y0`/`min_z0` values due to precision loss in `double`->`int`->`double` conversions. This leaded to incorrectly calculated `offset` value in some cases causing to go out-of-ranges. The fix for the problem could be to change `dx/dy/dz` values from `int` to `double` type which forces to all other dependent parameters to be calculated well.

Additionally, the shifts for `min_length` were compensated, so this was not a problem cause. However, during the code analysis I've found that components shifts were all compensated by absolute `min_length` value instead of OX,OY,OZ components this `min_length`. This won't cause an initial segfault problem, but actually a root of wrong behavior causing calculations of smaller than necessary length areas. Correcting it within the same PR.

SteveMacenski, 2021-07-20:
@charlielito Can you verify this goes away using https://github.com/ros-planning/navigation2/pull/2460

@AlexeyMerzlyakov can you verify that you tested this went away and is working properly?

AlexeyMerzlyakov, 2021-07-20:
> @AlexeyMerzlyakov can you verify that you tested this went away and is working properly?
I think I need to add some regression TC here for the cases which fixed in this issue.

charlielito, 2021-07-20:
I made the changes proposed in #2460 to my galactic branch and tested it. It __seems__ that this solved the planner server segfault since I run several times the stack and never encountered that segfault. Nonetheless, the controller server still crashes.

Here is the traceback of the controller server crashing:
[traceback.log](https://github.com/ros-planning/navigation2/files/6851363/traceback.log)

I also recorded another rosbag when the controller crashed:
https://drive.google.com/file/d/1Xjj5YTf9GGkWISiIFbqcyO6ZDHFU0dyJ/view?usp=sharing

More over. I tested this also with TB3 in simulation and this also is happening there after adding to the yaml file the realsense pointcloud source. Tested with this yaml file (has .txt extension because github does not support .yaml files):
[nav2_params.yaml.txt](https://github.com/ros-planning/navigation2/files/6851441/nav2_params.yaml.txt)

AlexeyMerzlyakov, 2021-07-21:
According to the traceback, the cause of the crash is still out-of-voxel-data array accessing in `bresenham3D()` routine. From the calculations in the given traceback, all offsets and length should be set correctly. However, as problem still appears, it probably could be that I've missed something or here we have to deal with another hidden problem.
Need more analysis.

SteveMacenski, 2021-08-24:
@charlielito, Alexey has come up with a new solution, can you test that out and verify if it fixed the underlying issue in all cases?

charlielito, 2021-08-24:
Just tested #2460 locally and it seems it is working, it hasn't crashed any time. Thanks!

SteveMacenski, 2021-08-24:
Sweet! Anything further you'd want to see before we merge / close the ticket?

closed 2021-08-25 by SteveMacenski