ros-navigation/navigation2 #2376: BT Action Nodes crash `bt_navigator` node upon action server call timeout

paucarre, 2021-05-29:
## Bug report

- Operating System: Ubuntu 20.04
- ROS2 Version: foxy
- Version or commit hash: 2e18dd1cae31807f68746eb1a2c702526a34f4cc
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue
- Use any Behavior Tree Action Node ( BT::ActionNodeBase )
- Call an action with lower-than-needed timeout.
- The whole `bt_navigator` node crashes (it would likely happen with any other BT enigne node)
- No logs besides "send_goal failed",
- Even the `ros daemon` needs restarting as it is no longer able to see the node that crashed

#### Expected behavior
- The node itself should return a  `BT::NodeStatus::FAILURE`.  In any case, because of a timeout or some other problem, the whole `bt_navigator` should never crash, specially with close to no feedback information.

#### Actual behavior
- The whole BT engine node crashes ( `bt_navigator` )
- No meaningful logs provided ( only "send_goal failed" )
- The node doesn't fail, just the whole `ros2` node suddenly dies and daemon is healthy.

#### Additional information
- This issue also takes place in other situations when the `ActionNodeBase` does checks on action results. Generally speaking, this is the way I think Nodes in BT should deal with errors:
  - Runtime exceptions generally speaking should be highly discouraged, and only admissible in extreme cases very well reasoned. 
  - If a framework has means of notifying errors in a pure way ( like BT with status or functional ways with Result[T] or Either[T] ), exceptions should be discouraged and instead, errors shall be logged and use the error notification mechanism that the framework provides.

## Feature request
- I will get a PR for this in a few minutes.

SteveMacenski, 2021-06-01:
I think a better solution is to handle the exception, not remove support for exceptions. Other Bt nodes will throw so we should instead make sure that there isn't crashing when exceptions are thrown in the behavior tree.

paucarre, 2021-06-01:
Thanks for feedback @SteveMacenski 

I totally agree with you that preventing exceptions from causing major damage is something necessary. 

I'd say though that throwing exceptions, *specifically for BTNodes* , is to me not the most suitable mechanism to notify errors.
These are the reasons why:
 - First and foremost, allowing exceptions as a mechanism of error notification leads existing nodes in the [BehaviorTree.CPP](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/controls/fallback_node.cpp) to misbehave. Specifically, there is no expectation of exceptions being thrown (no try-catch) and **no fallback mechanism is triggered upon exception**.  This means that `navigation2` will not be able to use `BehaviorTree.CPP` as the nodes within the framework do not expect exceptions and thus fallback and other recovery mechanisms are triggered. See [`executeTick()` method implementation does **not** account for exceptions](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/725eba7e0abbc704284dec393706e5404f1472e3/src/tree_node.cpp#L33)
 - Second, and at the same level of the first point, currently nodes in the master branch of `navigation2` do call [`tick()` method without making any check for exceptions](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/control/recovery_node.cpp). Thus, accepting exceptions in nodes a a mechanism of error notification will also break the contract in `navigation2` nodes.
 - Errors in `BT nodes` are expected and acceptable, and their mechanism of notification is setting `NodeStatus::Failure` as their status. Notifying `Failure` guarantees code execution is predictable ( [like Either/Result in FP frameworks](https://github.com/Dobiasd/FunctionalPlus/blob/master/test/result_test.cpp#L70) ) .
 - If we have both `Failure `and throwing exceptions as an accepted mechanism for error notification, then we'll need to increase the code wrapping with try-catch clauses and check both exceptions and Failure state. I think this can very potentially make code much less readable and highly convoluted for no good reason. Besides, developers will wonder whether there are situations where `Exceptions` are the right way to notify problems, and others where `NodeStatus::Failure` is. I see it leading to confusing for the end developers as well. 
 - `BT Nodes` are stateful. As I commented in [Future work that may be required](https://github.com/ros-planning/navigation2/pull/2377) PR, the status of a node can be `Failure` or not depending on the state it is in (e.g. it has the right inputs). If we throw exceptions when we find it to "fail" in the constructor, then we don't allow nodes to become ready. If we enforce more the idea to use `Failure` state as much as possible, these cases will be drastically reduced.

SteveMacenski, 2021-06-01:
There are certain types of failures that shouldn't just be hidden and should probably throw a notice to the behavior tree application that something has clearly gone wrong. If an invalid status / code comes by, that would be a good example of evidence that something is acting up in a way that it shouldn't and we should throw something. I agree that the robot should be able to recover from transient networking outages though -- the ability to have recovery actions or retries for network timeouts seem reasonable. 

The BT has exception handling here: https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/src/behavior_tree_engine.cpp#L47. To make a statement that BT.CPP is somehow "incorrect" to use exceptions is invalid. This is just C++ code, you can throw as long as you have a mechanism to catch, which we do. Throwing is an important tool in our developer's toolchest and we will _not_ outright remove it. There are failure conditions where throwing is the most reasonable thing so the behavior tree navigator or an application can handle a critical failure. 

---

@naiveHobo how do you feel about this? I think this brings up a good point that a networking failure should probably `fail` the BT node, not throw an exception. When an exception is thrown, its caught by the BT engine (either navigator or application using BT) for handling, exiting the existing task. For many cases, I think its the right answer (e.g. if the node state is invalid or node couldn't be locked signaling a serious issue or mis-using API). Though for the networking failures, I think that should probably return a failure code so that it may be retried, I don't think that should probably kill the entire application task. Thoughts?

Here are all of the places exceptions are used in the BT:

```
plugins/control/round_robin_node.cpp:          throw BT::LogicError("Invalid status return from BT node");
plugins/control/pipeline_sequence.cpp:        throw std::runtime_error(error_msg.str());
plugins/control/recovery_node.cpp:    throw BT::BehaviorTreeException("Recovery Node '" + name() + "' must only have 2 children.");
plugins/control/recovery_node.cpp:            throw BT::LogicError("A child node must never return IDLE");
plugins/control/recovery_node.cpp:            throw BT::LogicError("A child node must never return IDLE");
plugins/decorator/speed_controller.cpp:    throw BT::BehaviorTreeException(err_msg);
include/nav2_behavior_tree/bt_conversions.hpp:    throw std::runtime_error("invalid number of fields for point attribute)");
include/nav2_behavior_tree/bt_conversions.hpp:    throw std::runtime_error("invalid number of fields for orientation attribute)");
include/nav2_behavior_tree/bt_conversions.hpp:    throw std::runtime_error("invalid number of fields for PoseStamped attribute)");
include/nav2_behavior_tree/bt_action_node.hpp:        throw std::logic_error("BtActionNode::Tick: invalid status value");
include/nav2_behavior_tree/bt_action_node.hpp:      throw std::runtime_error("send_goal failed");
include/nav2_behavior_tree/bt_action_node.hpp:      throw std::runtime_error("Goal was rejected by the action server");
include/nav2_behavior_tree/bt_action_server_impl.hpp:    throw std::runtime_error{"Failed to lock node"};
```

The only 2 I have potential objections to are the send goal failure and rejected in bt_action_node, the others look OK to me.

SteveMacenski, 2021-06-18:
Any movement here?

philison, 2021-08-12:
Were the changes that were part of the merge into the foxy-devel branch reverted? I would love to see the changes in galactic as well, since a simple rejection by an action server will cause the entire BT to fail. There is no other clean way to fix this, right?

SteveMacenski, 2021-08-12:
This was never merged into any branch, the author of the PR has not been responsive

SteveMacenski, 2021-08-25:
Actions for this ticket
- Make getting ports outside of the constructors of the Bt nodes
- Evaluate the exceptions thrown in the BT and see which should be caught in the BT-node-level to return a certain type of code (FAILURE) rather than bring down the entire BT task. 

@philison I believe these 2 below are good candidates for catching the exception in the node itself and returning FAILURE of the task. I'd merge a PR which implemented that. The other exceptions I highlighted in the comment above seem like real BT-terminating error cases that shouldn't be hidden from the application and should probably be caught and end the tree execution. What do you think?

```
include/nav2_behavior_tree/bt_action_node.hpp:      throw std::runtime_error("send_goal failed");
include/nav2_behavior_tree/bt_action_node.hpp:      throw std::runtime_error("Goal was rejected by the action server");
```

closed 2021-10-18 by SteveMacenski