ros-navigation/navigation2 #2381: Smac Planner's path between frames is unstable,and easy to jump

JackRen88, 2021-06-02:
Hello,everyone!!

Smac Planner is set as the global planner in navigation,the planner’s planning frequency is 1hz。When there is an obstacle on the road, the planned path between frames is prone to be unstable.As the picture shows: plan the path to the left of the obstacle at time t, and plan the path to the right of the obstacle at time t+1, which will easily cause unstable control.
[image]

SteveMacenski, 2021-06-02:
Please ask on ROS answers, but in general, there's not really a solution to that other than not to replan or only replan when absolutely required (e.g. only when last path has become invalid). #2374 addresses this in a general sense that could use some attention if you'd be open to contributing to the project!

But this isn't in general a specific issue with SMAC / NavFn, its just an issue with replanning. As the robot moves or new environmental information is put into the costmap, slight differences in costs can cause 2 very-similarly costed plans to oscillate.

SteveMacenski, 2021-06-02:
https://github.com/ros-planning/navigation2/issues/1570 also addresses a similar concern. Both or either of which we'd be happy to chat on those tickets themselves about potential implementation details if you'd be open to contributing! I think both are relatively straight forward and could solve your issue

closed 2021-06-02 by SteveMacenski