ros-navigation/navigation2 #2382: bt_navigator crashes with segmentation fault

tal-grossman, 2021-06-02:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- ROS2 Version: Galactic (built from src)
- Version or commit hash: 1.0.3, c73ae815c9d71c89bc90cb8ce11deb3c22c07627
- DDS implementation: Cyclone

#### Steps to reproduce issue

- Navigate around obstacles to get to recovery and failure repetedly.
(I dont know why and if crash happens necceraly on failure but this is how I reproduce it).

Crash happens here for action nodes: https://github.com/ros-planning/navigation2/blob/b63b11f8a638faf389d6b99911d95899abe1580b/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L326
Due to `this->goal_handle_` empty ptr.
**gdb backtrace output: (action: nav2_msgs::action::FollowPath)** 
```
Thread 132 "bt_navigator" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7faa1fce90 (LWP 30305)]
memcmp () at ../sysdeps/aarch64/memcmp.S:52
52	../sysdeps/aarch64/memcmp.S: No such file or directory.
(gdb) bt
#0  memcmp () at ../sysdeps/aarch64/memcmp.S:52
#1  0x0000007fb2f21344 in nav2_behavior_tree::BtActionNode<nav2_msgs::action::FollowPath>::send_new_goal()::{lambda(rclcpp_action::ClientGoalHandle<nav2_msgs::action::FollowPath>::WrappedResult const&)#1}::operator()(rclcpp_action::ClientGoalHandle<nav2_msgs::action::FollowPath>::WrappedResult const&) const ()
   from /home/temi/robox2/ws/install/lib/libnav2_follow_path_action_bt_node.so
#2  0x0000007fb2f24ffc in rclcpp_action::ClientGoalHandle<nav2_msgs::action::FollowPath>::set_result(rclcpp_action::ClientGoalHandle<nav2_msgs::action::FollowPath>::WrappedResult const&) ()
   from /home/temi/robox2/ws/install/lib/libnav2_follow_path_action_bt_node.so
#3  0x0000007fb2f25150 in std::_Function_handler<void (std::shared_ptr<void>), rclcpp_action::Client<nav2_msgs::action::FollowPath>::make_result_aware(std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::FollowPath> >)::{lambda(std::shared_ptr<void>)#1}>::_M_invoke(std::_Any_data const&, std::shared_ptr<void>&&) ()
   from /home/temi/robox2/ws/install/lib/libnav2_follow_path_action_bt_node.so
#4  0x0000007fb7262b00 in rclcpp_action::ClientBase::handle_result_response(rmw_request_id_t const&, std::shared_ptr<void>) ()
   from /opt/ros/galactic/lib/librclcpp_action.so
```

**gdb backtrace output: (action: nav2_msgs::action::ComputePathToPose)**
```
Thread 17 "bt_navigator" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fa9ffce90 (LWP 30730)]
memcmp () at ../sysdeps/aarch64/memcmp.S:52
52	../sysdeps/aarch64/memcmp.S: No such file or directory.
(gdb) bt
#0  memcmp () at ../sysdeps/aarch64/memcmp.S:52
#1  0x0000007fb2fc1614 in nav2_behavior_tree::BtActionNode<nav2_msgs::action::ComputePathToPose>::send_new_goal()::{lambda(rclcpp_action::ClientGoalHandle<nav2_msgs::action::ComputePathToPose>::WrappedResult const&)#1}::operator()(rclcpp_action::ClientGoalHandle<nav2_msgs::action::ComputePathToPose>::WrappedResult const&) const ()
   from /home/temi/robox2/ws/install/lib/libnav2_compute_path_to_pose_action_bt_node.so
#2  0x0000007fb2fc5734 in rclcpp_action::ClientGoalHandle<nav2_msgs::action::ComputePathToPose>::set_result(rclcpp_action::ClientGoalHandle<nav2_msgs::action::ComputePathToPose>::WrappedResult const&) ()
   from /home/temi/robox2/ws/install/lib/libnav2_compute_path_to_pose_action_bt_node.so
#3  0x0000007fb2fc58ec in std::_Function_handler<void (std::shared_ptr<void>), rclcpp_action::Client<nav2_msgs::action::ComputePathToPose>::make_result_aware(std::shared_ptr<rclcpp_action::ClientGoalHandle<nav2_msgs::action::ComputePathToPose> >)::{lambda(std::shared_ptr<void>)#1}>::_M_invoke(std::_Any_data const&, std::shared_ptr<void>&&) ()
   from /home/temi/robox2/ws/install/lib/libnav2_compute_path_to_pose_action_bt_node.so
#4  0x0000007fb7262b00 in rclcpp_action::ClientBase::handle_result_response(rmw_request_id_t const&, std::shared_ptr<void>) ()
   from /opt/ros/galactic/lib/librclcpp_action.so

```

#### Expected behavior
goal will fail after defined number of recovreis has been exhausted - with msg: 
`bt_navigator][ERROR]: Goal failed`
#### Actual behavior
sometimes bt_navigator crashes with segmentation fault.
#### Additional information
**Suggested Fix:**
check goal_handle empy first.
`if (!goal_handle_ || this->goal_handle_->get_goal_id() == result.goal_id) `

tal-grossman, 2021-06-02:
closed due to https://github.com/ros-planning/navigation2/pull/2356 which on the main branch

closed 2021-06-02 by tal-grossman

SteveMacenski, 2021-06-02:
@tal-grossman I'll cherry pick that over (but still may be an issue with the binaries in the meantime)

tal-grossman, 2021-06-03:
> @tal-grossman I'll cherry pick that over (but still may be an issue with the binaries in the meantime)

@SteveMacenski I think you should consider not releasing the binearis if that is case. It is very disturbing.