ros-navigation/navigation2 #2383: Rviz - Send goal call failed

tal-grossman, 2021-06-02:
## Bug report

**Required Info:**

Operating System: Ubuntu 20.04
ROS2 Version: Galactic (built from src)
Version or commit hash: 1.0.3, c73ae81
DDS implementation: Cyclone

#### Steps to reproduce issue
 - navgiate to pose using nav2_rviz panl with `Nav2 Goal` botton

#### Expected behavior
Goal is sent and rviz acknowledged of goal recieved. user can Cancel if desired. 
#### Actual behavior
```
 [ERROR] [1622640405.128305326] [rviz2]: Send goal call failed
```

#### Additional information

- Navigation nodes are running on aarch64 architecture phisical robot. Rviz is running on PC. robot and PC are connected via Wifi.
- seems like navigation_client's `server_timout` here: https://github.com/ros-planning/navigation2/blob/b63b11f8a638faf389d6b99911d95899abe1580b/nav2_rviz_plugins/src/nav2_panel.cpp#L41
is exeeded. When Increasing it, is working ok according to expected behvior. 

**Suggested Fix:**
- increase timeout or have it as a configurable parameter on launch file

SteveMacenski, 2021-06-02:
I think you'd be safe just pushing that up to 100ms given its a demo plugin and shouldn't be used in production. I was wanting to avoid parameters for the rviz panel since its often launched separately so the defaults really need to do the job well. 

I'd be open to a PR to either expose that parameter and increase the timeout, or just increase the timeout outright.

tal-grossman, 2021-06-03:
> Were you doing the same thing and did you see this issue before the Galactic upgrade? If not, that's something I'm concerned about if there's further performance regressions.

Actually I'm not sure. We have difinely seen `"send_goal failed"` as reported back in https://github.com/ros2/ros2/issues/1074. But it wasn't from Rviz side. I don't think I have seen it on Rviz. 

I do think though timout should be increased as default because - what is the worst cast with big timeout?

SteveMacenski, 2021-06-03:
Yeah, we should have largely resolved that with https://github.com/ros-planning/navigation2/pull/2320 reliably and not messing with the bt navigator's tick rate.

The bad thing is it could lock up the rviz panel which would be not looking very good. I'm not entirely sure if those sends are in separate threads off hand so that they wouldn't be. If they are in separate threads, I think we could safely go up to 1s.

SteveMacenski, 2021-06-16:
@tal-grossman any movement?

SteveMacenski, 2021-06-18:
https://github.com/ros-planning/navigation2/pull/2416 expands the time out to 100ms. If we find that we need a higher value, we can reopen this ticket and we should parameterize many more things in the nav2 panel. Closing as the intent is complete but we might revisit in the future if the use of the panel no longer is for demos/testing but becomes more important for folks to use. In that case, we should parameterize many more things in it at the same time so it can be adjusted per use case.

closed 2021-06-18 by SteveMacenski