ros-navigation/navigation2 #2388: Docker `not found: "/opt/overlay_ws/install/nav2_util/share/nav2_util/local_setup.bash"`

vinnnyr, 2021-06-03:
## Bug report

**Required Info:**

- Operating System:
  - Host: Ubuntu 20.04
  - Docker
- ROS2 Version:
 -  Docker
- Version or commit hash:
  - Docker: docker.io/rosplanning/navigation2:main.release (SHA sha256:928a3335aa11d2bede6c73d01b4856e2b34b93996b4a8f846ec35835668ff32c) 
- DDS implementation:
  - N/A

#### Steps to reproduce issue

```
docker pull rosplanning/navigation2:main.release
docker run -it rosplanning/navigation2:main.release
```
#### Actual behavior
```
not found: "/opt/overlay_ws/install/nav2_util/share/nav2_util/local_setup.bash"
root@2d4ebed6c3bc:/opt/overlay_ws# 
```
#### Expected behavior
No `not_found` error

#### Additional information
I am not sure if this is a real issue / will effect the operation of nav2, or not, but I just wanted to bring this issue up to see what everyone's thoughts are.

SteveMacenski, 2021-06-03:
Don't use the docker container right now, they're in a transient state due to rolling issues. The fact that your print out says `rolling` means this is an old container, we're back on ros2 nightlies.

vinnnyr, 2021-06-03:
ok good to know. For those who are reading, looks like the `main` tag works as of time of writing

(`Digest: sha256:f60e9af8d32a897362c7ba869906ffb4531f269c1f25d9222dfd896a77a193ed`)
```
docker pull rosplanning/navigation2:main
docker run -it rosplanning/navigation2:main
```

SteveMacenski, 2021-06-04:
@ruffsl is that sensible?

ruffsl, 2021-06-04:
1. I wouldn't suggest using the docker images from the `rosplanning/navigation2` repo for anything other than CI at the moment, as that is currently there only purpose. If you'd like to run nav2 from a docker container for production, I'd recommend building a child image that builds FROM `ros:<distro>` and installs nav2 from apt.
2. The `main.release` and `main.debug` tags have been deprecated with https://github.com/ros-planning/navigation2/pull/2348 and will not be rebuilt. I've just purged these old tags from the docker hub registry to avoid future confusion on what tags are active. The new tag `rosplanning/navigation2:main` only builds up to the `builder` multistage to save on image size, and thus does not come with the overlay pre-built.
3. If you'd like to run nav2 from source for development, you can still use the root level Dockerfile in this repo. You may just want to pull and use the `rosplanning/navigation2:main` image tag as a build cache to avoid having to build and install the underlay on your own workstation and save some time:

```
docker pull rosplanning/navigation2:main
git clone https://github.com/ros-planning/navigation2.git
cd navigation2
docker build \
    --cache-from rosplanning/navigation2:main \
    --tag my/nav2:tag .
```

SteveMacenski, 2021-06-04:
Should this be closed? I"m uncertain if this is an actual issue or just something we're seeing because of the odd issues (and outdated docker containers). Are you seeing this with the suggestions Ruffin gave?

vinnnyr, 2021-06-04:
Ok, I can understand. Given Ruffin's suggestions now, is it time to edit or remove this portion of the docs? https://navigation.ros.org/build_instructions/index.html#docker

SteveMacenski, 2021-06-04:
Oh that would be appreciated, thanks for pointing that out!

SteveMacenski, 2021-06-08:
Merged

closed 2021-06-08 by SteveMacenski