ros-navigation/navigation2 #2389: Proper use of Gazebo / model spawner for more modular model / world use

SteveMacenski, 2021-06-04:
Right now, we have the model and the world all in 1 world file, rather than having a world file that we spawn a gazebo model into.

There's a `spawn_entity.py` and a multirobot version in Nav2 that we should be using instead to spawn robots within a world so that these things are not tied together. This would let someone change the robot or world more easily to spawn (different robots, in different places in the world, or in different worlds).

I think the gazebo gzserver needs to be the ros factory so file instead, and any other things required (separate URDF file with gazebo plugins from the .world file)

sathak93, 2021-07-16:
hi steve

i have manually input world file and spawned robot model sdf file and changed gazebo state to factory.
[image]
i debugged it  to  some extent.
the only difference between the launch file with spawn entity and combined model + world is 
the layered  costmaps  are not published 
the if block in 
https://github.com/ros-planning/navigation2/blob/5a7027536068183602f10b419bd711fafdad6784/nav2_costmap_2d/src/costmap_2d_ros.cpp#L422 is always becomes false as the current time is always zero
[image]
[image]

can you comment on  why auto current_time = now ()  is not updated or anything i did wrong?

sathak93, 2021-07-16:
making the if block  true publishes global and local costmaps

SteveMacenski, 2021-07-16:
I'm not sure, did it work before you changed the model spawning / gazebo? These seem unrelated but clearly must be somehow... maybe the gazebo .so file isn't publishing the clock in the same way?

sathak93, 2021-07-17:
@SteveMacenski  i found  the problem it seems that libgazebo_ros_factory.so  should be used in addition to libgazebo_ros_init.so instead of replacing it. 
this also need to be changed to fix multi robot simulation not working https://github.com/ros-planning/navigation2/blob/5a7027536068183602f10b419bd711fafdad6784/nav2_bringup/bringup/launch/multi_tb3_simulation_launch.py#L105
it may also solves issues #2446

SteveMacenski, 2021-07-17:
Oh really— interesting find! Thanks!!

SteveMacenski, 2021-07-22:
What's the status of this?

sathak93, 2021-07-23:
sorry i was busy 
i will create PR today

SteveMacenski, 2021-08-18:
Merging imminent

closed 2021-08-18 by SteveMacenski