ros-navigation/navigation2 #2398: Cannot install galactic navigation2 from apt

tim-fan, 2021-06-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- ROS2 Version:
  -  Galactic, installed via deb packages https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html
  - same behavior occurs in ros:galactic docker image (see reproduction steps below) 
- Version or commit hash:
  - n.a.
- DDS implementation:
  -  galactic default

#### Steps to reproduce issue

```
docker run -it ros:galactic bash -c "apt update && apt install ros-galactic-navigation2"
```

#### Expected behavior
navigation2 is installed

#### Actual behavior
Error printed:
E: Unable to locate package ros-galactic-navigation2

#### Additional information
Apologies if this is just user error, or better suited for ROS answers.

I'm trying to install nav2 on galactic as per instructions:
https://navigation.ros.org/build_instructions/index.html#install

However the apt install is failing to find the package, as reported above.
I note that in contrast the foxy version is found
```
docker run -it ros:galactic bash -c "apt update && apt install ros-foxy-navigation2"
```
^ that succeeds

Looking into packages.ros.org, I notice there are no .deb files stored for galactic navigation2:
http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-galactic-navigation2/
In contrast the foxy version does have .deb files:
http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-navigation2/
Perhaps the missing .debs relate to why I can't install? (I'm not clear on how releases / distribution via apt usually works).

SteveMacenski, 2021-06-08:
It is not yet released, please wait until the first galactic sync

closed 2021-06-08 by SteveMacenski

tim-fan, 2021-06-08:
Ok thank you for clarifying, I guess I was confused by the[ ROS package index](https://index.ros.org/p/navigation2/github-ros-planning-navigation2/#galactic) showing status=released for this. I suppose the release process has a few more steps that I am not aware of.