ros-navigation/navigation2 #2406: ROS Galactic groot monitoring

kribe48, 2021-06-14:
## Bug report
In ROS galactic, an additional action server has been added (navigate through poses). However, both servers use the same groot_zmq_publisher/server_port parameter values. This makes it impossible for me to monitor the behavior trees in groot. Am I using it wrong, or is it a bug that needs to be updated?

- Operating System:
ubuntu 20.04
- ROS2 Version:
Galactic
- Version or commit hash:
From source

#### Expected behavior
Able to monitor both navigate through poses and navigate to pose in groot

#### Actual behavior
Not possible to monitor either of them

SteveMacenski, 2021-06-14:
This is a known issue with Groot, that it only allows a single instance per process. I'd welcome a PR to help resolve this, but as I alluded to [this ticket](https://github.com/ros-planning/navigation2/issues/2386), if this isn't resolved relatively soon, my tentative plan is to strip support for groot out of Nav2 since it seems to be causing issues with critically needed features. 

Destroying and replacing Groot objects every time the navigator changes turns out to not be as straight forward as I thought it would be, so you'd be correct that right now it only works with NavigateToPose actions, not NavigateThroughPoses.

SteveMacenski, 2021-06-18:
I believe #2409 would resolve these concerns @kribe48. Do you agree?

kribe48, 2021-06-21:
Yes, I agree!