ros-navigation/navigation2 #2414: Release AWS warehouse world to foxy/rolling/galactic + readd dep for nav2-simple-commander

SteveMacenski, 2021-06-17:


joe28965, 2021-06-21:
question about the first part of this issue. There IS a AWS warehouse foxy version. I know, I created it, and made a pull request to the repo for it. So it would only be galactic/rolling right? Also, did that branch break for galactic/rolling?

SteveMacenski, 2021-06-21:
There is a foxy branch that works fine across the board (because model files dont really change), but they're not released so there are no binaries and its not rosdep installable, so we can't release Nav2 to a new distro until that is released because now we have a dependency on it

joe28965, 2021-06-21:
oh, you're talking about a binary release. My bad, I misunderstood. I was also curious what could change in a model file that would break it for other branches. Makes sense that it still works.

SteveMacenski, 2021-08-05:
They are going to release it themselves now, closing here.

closed 2021-08-05 by SteveMacenski