ros-navigation/navigation2 #2415: Additional features for the Python3 Simple Navigator

SteveMacenski, 2021-06-18:
- [ ] calling the controller 
- [ ] Allowing the nav requests to specify behavior trees
- [ ] Allowing the wait for navigation to bring up to have a configurable set of nodes to wait for (and default to the ones currently present) 
- [ ]  Call common recoveries
- [ ] TF transform getter for common transforms (get a pose in odom/map)
- [ ] Constructor to take in optional namespaces/topics to use for creating the clients and subscribers
- [ ] Get or change parameters in a given node

jacobwaller, 2021-08-02:
To add on to 

> Allowing the wait for navigation to bring up to have a configurable set of nodes to wait

as far as I can see, simple navigator does not work if SLAM-Toolbox is doing the localization because of its reliance on the nav2_amcl node

SteveMacenski, 2021-08-02:
Yeah, that's part of what that should do, have options for the nodes to wait on and not assume which are in place if the user can provide better estimates.

oldtopos, 2021-12-15:
Do not understand "calling the controller" and "Call common recoveries"

wait for navigation refers to function "waitUntilNav2Active" ?

ComputePathToPose and ComputePathThroughPoses take planner_id and use_start, should these be given overrides similar to bt override ?

SteveMacenski, 2021-12-16:
> Do not understand "calling the controller" and "Call common recoveries"

Its the analog to [getPlan](https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L207-L233) by calling the planner server, but for local trajectory planning via the controller server. See the `FollowPath.action` in `nav2_msgs`: https://github.com/ros-planning/navigation2/tree/main/nav2_msgs/action

Call common recoveries is the same, have `wait()` `spin()` `backup()` APIs calling their appropriate action servers (see the `nav2_msgs` link above).

With planning/control/recoveries, a Python application can have essentially full control over Nav2 without the Behavior Tree to create a simple application if they like or call servers to compute particular values for debug / application needs.

tonynajjar, 2022-01-23:
I'm working on adding wait, spin and backup. Spin and backup make sense to me but I'm not sure how wait is expected to work in a python script. As far as I understand in a BT, the wait node would stall the BT. In a python script, wait behave like a time.sleep()?

SteveMacenski, 2022-01-24:
Essentially, yes. I agree it might not make too much sense actually to include in its own right. You could simply sleep in the python script. I think its OK to skip unless you feel it would be good to add to be complete

SteveMacenski, 2022-02-02:
For setting parameters, I adapted this class to handle it. This can be massaged into the simple commander later 

```
class SetExternalParam(Node):

    def __init__(self, server_name):
        super().__init__('nav2_simple_commander_parameter_setter')

        self.cli = self.create_client(SetParameters, '/' + server_name + '/set_parameters')
        while not self.cli.wait_for_service(timeout_sec=1.0):
            self.get_logger().info('service not available, waiting again...')
        self.req = SetParameters.Request()

    def send_request(self, param_name, param_value):
        if isinstance(param_value, float):
            val = ParameterValue(double_value=param_value, type=ParameterType.PARAMETER_DOUBLE)
        elif isinstance(param_value, int):
            val = ParameterValue(integer_value=param_value, type=ParameterType.PARAMETER_INT)
        elif isinstance(param_value, str):
            val = ParameterValue(string_value=param_value, type=ParameterType.PARAMETER_STRING)
        elif isinstance(param_value, bool):
            val = ParameterValue(bool_value=param_value, type=ParameterType.PARAMETER_BOOL)
        self.req.parameters = [Parameter(name=param_name, value=val)]
        self.future = self.cli.call_async(self.req)
        while rclpy.ok():
            rclpy.spin_once(self)
            if self.future.done():
                try:
                    response = self.future.result()
                    if response[0].successful:
                        return True
                except Exception as e:
                    pass
                return False
```

tonynajjar, 2022-02-27:
Did you add the SetExternalParam class somewhere or that's just a draft? Can't find it

SteveMacenski, 2022-02-27:
No, that's just a draft I put together while working on another project and needed the capability in short order.

closed 2022-03-01 by SteveMacenski

VineetTambe, 2023-03-26:
Has the development on the following issue item been completed?
- [ ] ``` Constructor to take in optional namespaces/topics to use for creating the clients and subscribers ```
 
I am working on multi-robot stuff; having the nav2 simple commander API use namespaces would be really helpful.
If it is yet to be done, I was planning to make the changes and raise a pull request.

SteveMacenski, 2023-03-26:
I do not believe it is done (yet)

VineetTambe, 2023-03-26:
Cool! I'll raise a PR in a few days for this then!

VineetTambe, 2023-03-26:
Update: I have the initial name space thing working on a single robot with namespace argument set as "" (This is for sanity check to ensure the normal functionality of simple commander is not affected). 

But I am having issues testing the multirobot functionality of nav2.

I am checking out to the humble branch and then running:
``` ros2 launch nav2_bringup multi_tb3_simulation_launch.py use_sim_time:=True```

Then in another terminal I am running the following [test-script](https://github.com/VineetTambe/nav2-test-scripts/blob/main/test.py). The DDS that I am running is cycloneDDS.

However, when I run the above script I get data on the ```/robot1/initialpose``` topic but then the scipt gets stuck at [this line](https://github.com/VineetTambe/navigation2/blob/81d7056f73235573407cb2ea833dbbec275fe986/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L560) with the error: ```Result of get_state: unconfigured```

Can someone help me figure out where I am going wrong?? Am I missing a step?